introlab/rtabmap_ros #344: A question about odometry fusion

JoeyLeeNPU, 2019-08-30:
Hi,

I find that rtab-map can input vision, laser and extend odometry, so I want to confirm that if I input three parts including an extent VIO (such as RS or ZED visual odometry), a RGB image and a 2Dlaser scan. How does it work in odomtry module? Will it compute visual and laser odomtry, and then fuse the extent VIO to obtain the final odometry?

Thanks for help!

matlabbe, 2019-09-01:
Hi,

If you have external odometry, you don't need to start `rgbd_odometry`, `stereo_odometry` or `icp_odometry` nodes, just remap the odometry topic or `odom_frame_id` parameter of `rtabmap` node. The graph will be created with poses directly from your odometry.

To have better loop closure detection performance, providing also a registered depth image to RGB image will help.

cheers,
Mathieu

JoeyLeeNPU, 2019-09-02:
Thanks for your help！
In paper JFR 2019 for RTAB-map introduction，I found that it has two methods that loop closure detection for vision and proximity detection for 2D laser to optimize global trajectory. 
Some days ago, I use an external odomtry (ZED's /odom which is a close source VIO) and ZED's rgd/depth image as the input in my rtab. It works better for SLAM.
I add a long-range 2d laser recently in my rtab frame because I found a 2D laser with WheelIMU (here it is replaced by zed's odomtry) performs best in the paper.
So if like this (both have vision and laser sensor as input), will visual loop closure detection (only vision) and laser-based proximity detection (only laser without vision) **work together** (or other mode?) for graph optimization? 
Another question is that **a 2D laser with visual-inertial odomtry (VIO)** will be a better choice on a UAV platform?

matlabbe, 2019-09-02:
Long-range 2D lidar would be used in rtabmap only in 3DoF mode. In that paper, we assume a robot moving on a plane (x,y,theta). For 6DoF SLAM, I would not use the 2D lidar, because as soon the drone will tilt, the scan matching will likely fail. Note that like in this video: https://www.youtube.com/watch?v=IMSozUpFFkU&t=31s, as we know the pitch/roll of the UAV, scan matching could be done in 3DoF, while the Z value would be taken from VIO. This is not however supported in rtabmap.

We may create a node using VIO and lidar as inputs, and use 2D scan transformed in 3D with roll/pitch, and try to correct x,y,theta values of VIO with the previous scan or the accumulated previous scans in a local map. For the later, it seems similar to use icp_odometry with VIO as input odometry guess and the scan converted in 3D (PointCloud2) in stabilized frame of the drone, with `Reg/Force3DoF=true`. Well, it is just some thoughts, though if you read some papers about drones using 2D lidar for 3D navigation, you may find a more complete and robust solution.
 
cheers,
Mathieu

JoeyLeeNPU, 2019-09-03:
Thank you so much for help.
In these days, I don't set Reg/Force3DoF=true when I use both VIO and 2D laser in rtabmap, but I also obtain a pose (x, y, z, theta). 
According to your reply, 
(1) I think (x, y, theta) is optimized by 2D scan, and the height (z) is the source result of VIO?  Because I found the **pitch/roll of the UAV, scan matching could be done in 3DoF, while the Z value would be taken from VIO. This is not however supported in rtabmap.** I doubt if my rtabmap pose is incorrect.
(2) Or in this condition (don't set Reg/Force3DoF=true), 2D laser scan **isn't work** for rtabmap?
(3) If when using both vio and 2D laser, I set the **Reg/Force3DoF=true**. The pose result is only include (x, y, theta)? If like this, I will use downward one-point lidar to obtain my altitude for UAV.

@matlabbe thanks for your help again and hoping for you reply O(∩_∩)O

matlabbe, 2019-09-06:
If you have a rosbag (of VIO odom, rgb+depth+camera info, tf, tf_static and scan) with the UAV moving in the environment, and a launch file that is used for setting up rtabmap, we may better reproduce the problem here and see better what you mean.

JoeyLeeNPU, 2019-09-07:
Thanks for your help!
I will do it as soon as possible, and hoping for the discuss in more details.

closed 2019-10-16 by JoeyLeeNPU