introlab/rtabmap_ros #345: Some questions about "shifting" and pointcloud

jerryt3204, 2019-08-30:
Hi, i'm using RTABMAP_ROS on TX2 recently.
OS: Ubuntu 16.04
ROS: Kinetic

I've encountered a problem like this picture:
[image]
I didn't move my car or anything yet, but it "seemed like" it was moving (according to the blue line, i just call it "shifting"). So my pointcloud looks strange.
But sometimes when i reboot my TX2 and launch again, it works fine(no moving).
I don't know what can cause this problem...

And the second question is, i usually use RVIZ for visualization and rtabmapviz for loop closure.
But sometimes RVIZ's "old" pointcloud disappeared, in other words, RVIZ only shows current pointcloud, the older pointcloud disappeared, so i can't see my whole map in RVIZ.

I'm new in RTABMAP_ROS and SLAM area, any suggestion is appreciated !!

matlabbe, 2019-09-01:
Hi,

For the odometry drifting, did you move objects in from of the camera? Currently you are using visual odometry, so if some visual features are moving, the robot will think it has moved. Note that if your car has odometry (wheel encoders), consider to use it to save computation ressources by avoiding to do visual odometry (that would fix your problem of moving when standing still and will help for the other comment below).

For parts of the map disappearing, you may have memory management enabled. You can disable it to see the difference `Rtabmap/TimeThr=0`. Doing visual odometry on TX2 may use a full core at 100%, so there is not as much remaining free time for the mapping thread. If you can use wheel odometry, I suggest to do so. Then as you are using a lidar, I would enable  `RGBD/RefineNeighborLinks` parameter to true to correct the wheel odometry errors with the lidar.

jerryt3204, 2019-09-04:
hi @matlabbe , thanks for your quick reply !

I've checked, i didn't move objects in front of the camera, but this problem still occured like the picture below.
[image]

I followed your advice, i set "visual_odometry" to "false", and my wheel encoders publish /odom, but i encountered this:
[image]

It seems like it doesn't receive any data, so i use "rostopic list" to check. I'm sure they're publishing. So i wanna know if i have to do another settings or somewhat?

here is how i launch my rtabmap_jerry.launch:
 ```
 <include file="$(find jerry)/launch/rtabmap_jerry.launch">
    <arg name="rtabmap_args"            value="--delete_db_on_start"/>
    <arg name="depth_topic"             value="/camera/aligned_depth_to_color/image_raw"/>
    <arg name="rgb_topic"               value="/camera/color/image_raw"/>
    <arg name="camera_info_topic"       value="/camera/color/camera_info"/>

    <arg name="approx_sync"             value="true"/>  <!--it makes everything works, but dont know why-->
    <arg name="rtabmapviz"              value="true"/>
    <arg name="rviz"                    value="false"/>
    <arg name="subscribe_scan"          value="true"/>
    <arg name="frame_id"                value="base_link"/>           <!-- construct relationship between "odom" and "what you type" -->

    <arg name="tag_linear_variance"     value="0.0001" />
    <arg name="tag_angular_variance"    value="0.0005" />             <!-- >=9999 means ignore rotation in optimization, when rotation estimation of the tag is not reliable -->

    <arg name="visual_odometry"         value="false"/>
    <arg name="icp_odometry"            value="false"/>
    <arg name="queue_size"              value="20"/>   <!-- tuned -->
  </include>
```
Any suggestion is appreciated !!

matlabbe, 2019-09-06:
To use your odometry (assuming it is published on `/odom` topic), add [odom_topic](https://github.com/introlab/rtabmap_ros/blob/986ed79d35b0fb6e77eb2a1e2723954d75f27941/launch/rtabmap.launch#L87) argument:
```xml
<arg name="odom_topic"               value="/odom"/>
```
or TF can be also used:
```xml
<arg name="odom_frame_id"            value="odom"/> 
<arg name="odom_tf_angular_variance" value="0.0005"/> 
<arg name="odom_tf_linear_variance"  value="0.0001"/>
```

It is very strange the visual odometry is drifting as much though. If you can record a small rosbag, it could help to debug more:
`$ rosbag record /tf /tf_static /camera/color/image_raw /camera/color/camera_info /camera/aligned_depth_to_color/image_raw /scan`

cheers,
Mathieu

jerryt3204, 2019-09-09:
Thanks for your quick reply again.

After i followed your steps, i eventually disabled VO and use my /odom instead.
I'm also curious about the drifting, so i will check my system, any updates will be uploaded.

Thanks!!

mszuyx, 2020-09-24:
@jerryt3204  @matlabbe  Hi, I also have similar issue. If I have only RGB-D camera mapping and VO enable, everything works fine. Once I set arg name="subscribe_scan" or arg name="subscribe_scan_cloud" to true, the full map starts to **spin**....  Did you mange to resolve the issue?

My hardware setup is:
Clearpath Jackal
Jetson TX2 with Jetpack 3.3
ZED camera
Velodyne 16

My launch file looks like this:
<launch>
  

  
           
  
     
        
  <arg name="stereo"                    default="false"/>
  <arg     if="$(arg stereo)" name="depth"  default="false"/>
  <arg unless="$(arg stereo)" name="depth"  default="true"/>
 
  
  <arg name="rtabmapviz"              default="true" /> 
  <arg name="rviz"                    default="false" />
  
  
  <arg name="localization"            default="false"/>
  
  
  <arg name="use_sim_time"            default="false"/>
  <param if="$(arg use_sim_time)" name="use_sim_time" value="true"/>

  
  <arg name="cfg"                     default="" /> 
  <arg name="gui_cfg"                 default="~/.ros/rtabmap_gui.ini" />
  <arg name="rviz_cfg"                default="$(find rtabmap_ros)/launch/config/rgbd.rviz" />
  
  <arg name="frame_id"                default="base_link"/>  
  <arg name="odom_frame_id"           default="odom"/>                
  <arg name="map_frame_id"            default="map"/>
  <arg name="ground_truth_frame_id"   default=""/>     
  <arg name="ground_truth_base_frame_id" default=""/>  
  <arg name="namespace"               default="rtabmap"/>
  <arg name="database_path"           default="~/.ros/rtabmap.db"/>
  <arg name="queue_size"              default="500"/>
  <arg name="wait_for_transform"      default="0.2"/>
  <arg name="args"                    default="--delete_db_on_start"/>              
  <arg name="rtabmap_args"            default="$(arg args)"/>   
  <arg name="launch_prefix"           default=""/>              
  <arg name="output"                  default="screen"/>        
  <arg name="publish_tf_map"          default="true"/>

  
  <arg     if="$(arg stereo)" name="approx_sync"  default="true"/>
  <arg unless="$(arg stereo)" name="approx_sync"  default="$(arg depth)"/>         
   
  
  <arg name="rgb_topic"               default="/zed_node/rgb/image_rect_color" />
  <arg name="depth_topic"             default="/zed_node/depth/depth_registered" />
  <arg name="camera_info_topic"       default="/zed_node/rgb/camera_info" />
  <arg name="depth_camera_info_topic" default="$(arg camera_info_topic)" />
  
  
  <arg name="stereo_namespace"        default="/zed_node"/>
  <arg name="left_image_topic"        default="$(arg stereo_namespace)/left/image_rect_color" />
  <arg name="right_image_topic"       default="$(arg stereo_namespace)/right/image_rect_color" />      
  <arg name="left_camera_info_topic"  default="$(arg stereo_namespace)/left/camera_info" />
  <arg name="right_camera_info_topic" default="$(arg stereo_namespace)/right/camera_info" />
  
  
[... 311 lines cut ...]
      <remap from="scan_descriptor"        to="$(arg scan_descriptor_topic)"/>
      <remap from="odom"                   to="$(arg odom_topic)"/>
    </node>
  
  </group>
  
  
  <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(arg rviz_cfg)"/>
  <node if="$(arg rviz)" pkg="nodelet" type="nodelet" name="points_xyzrgb" args="standalone rtabmap_ros/point_cloud_xyzrgb" output="$(arg output)">
    <remap if="$(arg stereo)" from="left/image"        to="$(arg left_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="right/image"       to="$(arg right_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="left/camera_info"  to="$(arg left_camera_info_topic)"/>
    <remap if="$(arg stereo)" from="right/camera_info" to="$(arg right_camera_info_topic)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/image"         to="$(arg rgb_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="depth/image"       to="$(arg depth_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/camera_info"   to="$(arg camera_info_topic)"/>
    <remap from="rgbd_image"        to="$(arg rgbd_topic_relay)"/>
    <remap from="cloud"             to="voxel_cloud" />

    <param name="decimation"  type="double" value="4"/>
    <param name="voxel_size"  type="double" value="0.0"/>
    <param name="approx_sync" type="bool"   value="$(arg approx_sync)"/>
  </node>

</launch>

matlabbe, 2020-09-27:
Make sure you don't have multiple odometry publishing on same TF or same topics. Just adding subscribe_scan or subscribe_scan_cloud would have an effect only on the grid map if icp registration is no set.

mszuyx, 2020-09-29:
@matlabbe Thank you for getting back to me so quick! I modified the "demo_robot_mapping" launch file to enable both Lidar and stereo mapping. Now everything works fine! (I was trying to come up with my own launch file base on "rtabmap.launch") Like what you mentioned, it could be I launch multiple odom over the base_link, or I messed up some hyper parameters in the launch file. 
It seems like modifying one of the demo launch files will be the easiest way to approach it.