introlab/rtabmap_ros #346: Tuning paramaters TX2 Zed RPlidarA2

bj-neilson, 2019-09-05:
Hi Matthieu,

I have RTABMap Ros running reasonably well.  I am trying to map outdoors in an orchard of tree rows.  I find when I turn at the end of a tree row the robot never sees a 180deg turn.  Maybe 150deg or so.  So there is a V shaped 2D map rather than a map of both sides of a tree row.

I am using the Zed Odom and ICP with the RPLidarA2.  I have been trying to tune by changing parameters in .launch file.  Some parameters change, some do not.  I'm a bit confused.  For example I was unable to change the value of "RGBD/OptimizeMaxError" is 3.000000.  I tried in launch file to change to 0.1, but it still runs at 3.00000.

Kindly see below...

<launch>
  <group ns="rtabmap">

    
    
    <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="standalone rtabmap_ros/rgbd_sync" output="screen">
      <remap from="rgb/image"       to="/zed/zed_node/rgb/image_rect_color"/>
      <remap from="depth/image"     to="/zed/zed_node/depth/depth_registered"/>
      <remap from="rgb/camera_info" to="/zed/zed_node/rgb/camera_info"/>
      <remap from="rgbd_image"      to="rgbd_image"/> 
      
      
      <param name="approx_sync"       value="false"/> 
    </node>

    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start">
          <param name="frame_id" type="string" value="base_link"/>

          <param name="subscribe_depth" type="bool" value="false"/>
          <param name="subscribe_rgbd"  type="bool" value="true"/>
          <param name="subscribe_scan"  type="bool" value="true"/>
	  <param name="subscribe_rgb"   type="bool" value="false"/>

          <remap from="path" to="/zed/zed_node/path"/>
	  <remap from="odom" to="/zed/zed_node/odom"/>
          <remap from="scan" to="/scan"/>
          <remap from="rgbd_image" to="rgbd_image"/>

          
    
	  <param name="queue_size" 	  	  type="int"    value="15"/>
	  <param name="RGBD/NeighborLinkRefining" type="string" value="true"/>
          <param name="RGBD/ProximityBySpace"     type="string" value="true"/>
          <param name="RGBD/AngularUpdate"        type="string" value="0.01"/>
          <param name="RGBD/LinearUpdate"         type="string" value="0.01"/>
          <param name="RGBD/OptimizeFromGraphEnd" type="string" value="true"/>
          <param name="Grid/FromDepth"            type="string" value="false"/> 
          <param name="Reg/Force3DoF"             type="string" value="true"/>
          <param name="Reg/Strategy"              type="string" value="1"/> 
 	  <param name="RGBD/OptimisemaxError"	  type="string" value="0.1"/>  
	  <param name="RGBD/ProximityPathMaxNeighbors" type="string" value="15"/>
	  <param name="wait_for_transfom"         type="bool"   value="true"/>
	  <param name="wait_for_transform_duration"    type="double" value="0.2"/>

          
          <param name="Icp/VoxelSize"                 type="string" value="0.05"/>
          <param name="Icp/MaxCorrespondenceDistance" type="string" value="0.1"/>
          <param name="wait_for_transfom"             type="bool"   value="true"/>
	  <param name="wait_for_transform_duration"   type="double" value="0.2"/>
     </node>
  </group>
</launch>

ubuntu@ubuntu-tx2:~/catkin_ws$ roslaunch zed_rtabmap_example mower.launch
... logging to /home/ubuntu/.ros/log/1f753e18-cf87-11e9-85fb-00044ba839fa/roslaunch-ubuntu-tx2-27728.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu-tx2:39513/

SUMMARY
[... 409 lines cut ...]
[ INFO] [1567654955.095622426]: rtabmap (46): Rate=1.00s, Limit=0.000s, RTAB-Map=0.5099s, Maps update=0.0003s pub=0.0000s (local map=11, WM=11)
[ INFO] [1567654955.727460997]: rtabmap (47): Rate=1.00s, Limit=0.000s, RTAB-Map=0.1219s, Maps update=0.0003s pub=0.0000s (local map=11, WM=11)
[ INFO] [1567654956.761058529]: rtabmap (48): Rate=1.00s, Limit=0.000s, RTAB-Map=0.1189s, Maps update=0.0003s pub=0.0000s (local map=11, WM=11)
[ INFO] [1567654957.797129380]: rtabmap (49): Rate=1.00s, Limit=0.000s, RTAB-Map=0.1237s, Maps update=0.0003s pub=0.0000s (local map=11, WM=11)
[ INFO] [1567654958.882469487]: rtabmap (50): Rate=1.00s, Limit=0.000s, RTAB-Map=0.1200s, Maps update=0.0003s pub=0.0000s (local map=11, WM=11)
[ INFO] [1567654959.864383779]: rtabmap (51): Rate=1.00s, Limit=0.000s, RTAB-Map=0.1263s, Maps update=0.0004s pub=0.0000s (local map=11, WM=11)
[ WARN] (2019-09-05 13:42:41.235) OptimizerG2O.cpp:1061::optimize() Computing marginals: vertex 52 has negative hessian index (-1). Cannot compute last pose covariance.
[ INFO] [1567654961.242580440]: rtabmap (52): Rate=1.00s, Limit=0.000s, RTAB-Map=0.4247s, Maps update=0.0004s pub=0.0000s (local map=12, WM=12)
[ WARN] (2019-09-05 13:42:42.216) OptimizerG2O.cpp:1061::optimize() Computing marginals: vertex 53 has negative hessian index (-1). Cannot compute last pose covariance.
[ WARN] (2019-09-05 13:42:42.216) Rtabmap.cpp:2888::process() Rejecting all added loop closures (1, first is 53 <-> 34) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 3.093660 (edge 1->46, type=1, abs error=0.053685 m, stddev=0.017353). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2019-09-05 13:42:42.216) Rtabmap.cpp:2922::process() Loop closure 53->34 rejected!
[ INFO] [1567654962.222566434]: rtabmap (53): Rate=1.00s, Limit=0.000s, RTAB-Map=0.3932s, Maps update=0.0003s pub=0.0000s (local map=13, WM=13)
[ WARN] (2019-09-05 13:42:43.246) OptimizerG2O.cpp:1061::optimize() Computing marginals: vertex 54 has negative hessian index (-1). Cannot compute last pose covariance.
[ WARN] (2019-09-05 13:42:43.246) Rtabmap.cpp:2888::process() Rejecting all added loop closures (1, first is 54 <-> 35) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 10.129725 (edge 35->54, type=1, abs error=0.159222 m, stddev=0.015718). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2019-09-05 13:42:43.246) Rtabmap.cpp:2922::process() Loop closure 54->35 rejected!
[ INFO] [1567654963.252893902]: rtabmap (54): Rate=1.00s, Limit=0.000s, RTAB-Map=0.3836s, Maps update=0.0004s pub=0.0000s (local map=14, WM=14)
[ WARN] (2019-09-05 13:42:44.401) OptimizerG2O.cpp:1061::optimize() Computing marginals: vertex 38 has negative hessian index (-1). Cannot compute last pose covariance.
[ WARN] (2019-09-05 13:42:44.464) OptimizerG2O.cpp:1061::optimize() Computing marginals: vertex 55 has negative hessian index (-1). Cannot compute last pose covariance.
[ INFO] [1567654964.472400477]: rtabmap (55): Rate=1.00s, Limit=0.000s, RTAB-Map=0.5741s, Maps update=0.0003s pub=0.0000s (local map=15, WM=15)
 
Sorry for the big cut and paste.  I'd appreciate your thoughts on anything from the launch file/parameters.  Then I'll try again outside and record some fresh data to show.

Thank you,

Ben

matlabbe, 2019-09-06:
Hi, if you disable the scan (`subscribe_scan`=false), is it better? If you have again a V shape, it would be more related to zed odometry.

You have written "RGBD/OptimisemaxError" instead of "RGBD/OptimizeMaxError" in the launch file. I don't really recommend to set it below 3, otherwise a lot of good loop closures may be rejected.

cheers,
Mathieu

bj-neilson, 2019-09-09:
Hi Mathieu,
Thanks for the help...I found the spelling error. That helped a bit!  I'll set that back to 3.0 and try.
Will also try the 2 odometry strategies separately and try to determine why I'm getting a V shape.  The weekend got in the way of some testing.
Cheers,
Ben

JoeyLeeNPU, 2019-10-16:
I want to ask whether your lidar is working better in this launch file? I want to refer this to write my launch file.

bj-neilson, 2019-10-16:
Hi @JoeyLeeNPU, 
My odometer is not really working better from the lidar.  Maybe the RPLidarA2 is not quite providing the best solution in my environment.  I'm spending more time trying to improve the camera setup for better visual odometry.  Not quite solved yet.

JoeyLeeNPU, 2019-10-16:
I feed hokuyo 2d laser into rtabmap, maybe also not work well. I think if I provide both laser and camera for odometry, the rtabmap will be better? Because I read paper of rtabmap and find this maybe a good solution for uav slam.

JoeyLeeNPU, 2019-10-16:
@bj-neilson Could we discuss more about this? In email or other software. Need for your help.