introlab/rtabmap_ros #348: Instability of trajectory points in the process of two-sensor modeling

TomLuosy, 2019-09-11:
[image]
[image]
Hi,I encountered problems with ROS modeling.I modeled it with two sensors rotating.It can be seen that the trajectory point is very stable without rotation of the sensor.But when I started rotating the sensors, the trajectory points began to float, even if they were just rotating in one place.
I want to know what determines the trajectory point when modeling with two sensors. How can I improve this situation?Thanks.

TomLuosy, 2019-09-13:
> What I see is that when you rotate the sensor, the camera lacks of visual features, creating very large drifts. ultimately it could get lost ([red screen](https://github.com/introlab/rtabmap/wiki/Kinect-mapping#lostodometry)). What are the sensors used? You can use [static_transform_publisher](http://wiki.ros.org/tf#static_transform_publisher) to create a transform between them. What is your tf tree (`rosrun tf view_frames`)?

[image]
Hi,thanks your reply!
I found the location set in the launch file.The setting of the last four parameters is very good, which also explains that the modeling effect is very good without rotation. How do I set the first three parameters of translation to match the motion of the sensor?

matlabbe, 2019-09-14:
You should try to see what /camera1_link and /camera2_link frames are referring to. For example, if the camera link is the base of the camera and the cameras are 10 cm apart, centred on y axis from base_link frame, one transform would be 0 -5 0 and the other  0 5 0. Note that you can also set the angle in euler format (yaw pitch roll) instead of quaternion for convenience. Do you have a photo of your setup?

TomLuosy, 2019-09-17:
I spent a few days finding the cause of the problem. The sensor I am using is intel's d435. In its official documentation I found that when using dual sensor modeling, the origin of the reference coordinate system is set on the left sensor. As shown, in the modeling, I want the origin between the two sensors. What I did before was only to make the images on both sides parallel, but in reality they are more like a parallelogram. This also explains the very good results when modeling without rotating the sensor.
So the problem I need to solve now becomes how to set the origin of the reference coordinate system between the two sensors.
[image]
[image]

matlabbe, 2019-09-22:
I would put a base_link frame in the middle base of the robot, then from there, a frame for each camera: base_link-> camera1_link and base_link -> camera2_link. You can show in RVIZ the TF tree to see if the frames / point clouds match the current setup.