introlab/rtabmap_ros #350: rtab-map pose covariance

JoeyLeeNPU, 2019-09-21:
Hi~

We want to use rtab-map as a method for our drone to obtain external state information. 
But in PX4 board, its EKF2 can perform better if we input an pose estimation with the covariance.
Hence, I want to know if rtab-map can compute a pose with covariance. If can, could u tell me how to find this message in ROS package.

Thanks,
Joey

matlabbe, 2019-09-22:
Do you mean odometry pose or localization pose? For odometry, rtabmap's odometry publishes the odometry topic (`/rtabmap/odom`) containing a covariance. On localization only, rtabmap publishes `/rtabmap/localization_pose` topic with a covariance too.

JoeyLeeNPU, 2019-09-23:
Thanks for your help
I have found that Odom and localization mode both have the covariance, but my Odom is an external Odom (vio) and I use rtab for mapping mode rather than localization only.
So any suggestions?

Joey

matlabbe, 2019-09-23:
`localization_pose` is published only on loop closures in mapping mode (when `map->odom` frame changes). In those cases, the covariance represents the marginal (g2o, gtsam) after graph optimization. It won't be published if no loop closures happen.

JoeyLeeNPU, 2019-09-23:
So could I use localization_pose in mapping mode for my px4 external pose estimation? 
That is, when no loop closures happen it only pub a pose, while happened,  a pose with covariance Is published?

matlabbe, 2019-09-23:
No pose msgs (with or without covariance) are published when there are no loop closures. 

The problem is that on loop closures, the robot may jump many meters (if the loop is large). I don't think you should feed the map correction to px4 controller. I would feed only your vio messages to px4's ekf2. On rtabmap side, I would set `RGBD/OptimizeFromGraphEnd` to true, so that the map is transformed instead of jumping the pose of the robot.

cheers,
Mathieu

JoeyLeeNPU, 2019-09-24:
I have a try that feed only vio messages to px4's ekf2, and it do better for flights, but I want to use a more accurate position information for UAV to localize other objects (some markers), hence, I want to try the rtab's pose. Some days ago, I use the /tf to listen the pose topic and send it to ekf2, it can also do better for flights. But when I learn more about px4's ekf2, I notice that a pose with covariance is helpful for state estimation. 
So, if I want to use the **rtab's pose** optimized by loop closures detection, **which topic should I choose for px4 to subscribe**? Because I see that you say "localization_pose" will not publish pose when there are no loop closures.  Or I only can use /tf to obtain the transformation (pose without covariance)?

Thanks so much!  hah^_^
Joey

matlabbe, 2019-09-24:
If you want the pose to be continuously published to EKF2, you would have to use TF without covariance.

closed 2019-10-15 by JoeyLeeNPU

rakshith95, 2022-09-27:
@matlabbe does rtabmap update the tf  only when there's loop closure as well?

matlabbe, 2022-09-27:
Map->odom tf is published at 20 Hz, but its transform would change only on loop closures. For example, if you start a new map, map->odom will be Identity until a loop closure is detected.

rakshith95, 2022-09-29:
I see, thank you!