introlab/rtabmap_ros #352: catkin_make fail - No rule to make target '/usr/lib/aarch64-linux-gnu/libGL.so'

bj-neilson, 2019-09-24:
Hi,
I'm trying to install rtabmap_ros from source on a TX2 flashed with new JP4.2

I ensured that I used Step 7 in the Tegra instructions. ie
"To avoid libGL undefined errors:

$ cd /usr/lib/aarch64-linux-gnu/
$ sudo ln -sf tegra/libGL.so libGL.so"

Installation keeps failing as per:

ubuntu@tx2:~/catkin_ws$ catkin_make -j4 --pkg rtabmap_ros
Base path: /home/ubuntu/catkin_ws
Source space: /home/ubuntu/catkin_ws/src
Build space: /home/ubuntu/catkin_ws/build
Devel space: /home/ubuntu/catkin_ws/devel
Install space: /home/ubuntu/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/catkin_ws/build"
####
####
#### Running command: "make -j4" in "/home/ubuntu/catkin_ws/build/rtabmap_ros"
####
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_PublishMap
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Point2f
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_RGBDImage
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_GetPlan
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_OdomInfo
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Link
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Info
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Point3f
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_SetLabel
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_ListLabels
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_KeyPoint
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_UserData
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_MapGraph
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_ResetPose
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_SetGoal
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_MapData
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_GPS
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_NodeData
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Path
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_GetMap
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Goal
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target rtabmap_ros_gencfg
[ 11%] Built target rtabmap_ros_generate_messages_py
[ 23%] Built target rtabmap_ros_generate_messages_eus
[ 33%] Built target rtabmap_ros_generate_messages_lisp
[ 43%] Built target rtabmap_ros_generate_messages_cpp
[ 44%] Built target cv_bridge
[ 46%] Built target image_geometry
[ 56%] Built target rtabmap_ros_generate_messages_nodejs
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libGL.so', needed by '/home/ubuntu/catkin_ws/devel/lib/rtabmap_ros/rgbd_sync'.  Stop.
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libGL.so', needed by '/home/ubuntu/catkin_ws/devel/lib/rtabmap_ros/icp_odometry'.  Stop.
CMakeFiles/Makefile2:4504: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_rgbd_sync.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_rgbd_sync.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:4542: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_icp_odometry.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_icp_odometry.dir/all] Error 2
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libGL.so', needed by '/home/ubuntu/catkin_ws/devel/lib/rtabmap_ros/stereo_odometry'.  Stop.
CMakeFiles/Makefile2:4580: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_stereo_odometry.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_stereo_odometry.dir/all] Error 2
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libGL.so', needed by '/home/ubuntu/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry'.  Stop.
CMakeFiles/Makefile2:4618: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_rgbd_odometry.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_rgbd_odometry.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4" failed

I tried deleting the link and doing it again.  No change.  Any thoughts?

Thanks,
Ben

matlabbe, 2019-09-24:
It is not the same error than this post: https://devtalk.nvidia.com/default/topic/1007290/jetson-tx2/building-opencv-with-opengl-support-/post/5141945/#5141945

It is like it doesn't exist on your system. Does `/usr/lib/aarch64-linux-gnu/tegra/libGL.so` exist? If you are using a recent Jetpack, Ubuntu 18.04 may be installed. The steps described were with Ubuntu 16.04 (Jetpack 3).

bj-neilson, 2019-09-25:
Thanks Matthieu,
Yes you’re right. As I kept searching this morning it appears I’m missing Nvidia drivers. Something a bit odd with the newest JetPack perhaps. I’ll go back a version and try again.
Thanks
Ben

bj-neilson, 2019-09-28:
Seem to have found a solution...

ln -sf /usr/lib/aarch64-linux-gnu/libGL.so.1.0.0 libGL.so

instead of  

sudo ln -sf tegra/libGL.so libGL.so

matlabbe, 2019-11-01:
Just to make it clearer for other people having the same problem:
```
$ cd /usr/lib/aarch64-linux-gnu/
$ sudo ln -sf libGL.so.1.0.0 libGL.so
```
Thx!

closed 2019-11-01 by matlabbe

248msc, 2020-07-16:
In my case (ubuntu 16.04, nvidia-430), libGL.so and the original binary libGL.so.1.7.0 is only found in /usr/lib/nvidia-430 .
Doing the following solved the problem.
`cd /usr/lib/x86_64-linux-gnu
sudo cp /usr/lib/nvidia-430/libGL.so.1.7.0 libGL.so.1.7.0
sudo ln -sf libGL.so.1.7.0 libGL.so`

Somehow, I have to do the copy step (not linking the binary in /usr/lib/nvidia-430 directly) to make it work.
Hope this help others with similar issue.