introlab/rtabmap_ros #353: Frame ID match and localization tf is positioned at wrong locations....

dilipdhapse, 2019-09-24:
Hello, I am facing a problem in frame mating in Rtabmap in localization mode.

I have created a simulated gazebo outdoor environment in that, I have launched turtlebot3 Wafflepi robot with a simulated stereo camera. It will give me a stereo left and right images with camera info. This stereo images and camera info is given to stereo image processing node where it will rectify image, create depth image, point cloud, and monochrome images. 

Rtabmap with default settings with stereo=true and odom=/gazebo_odom.
from that, I had built a map and relaunched in localization mode. but in localization mode it not giving a perfect match. so tf after matching a lot different than actual pose of robot.

What parameters should I consider.

any help will be fine.

Thanks

matlabbe, 2019-09-24:
Is the robot localized? Do you mean the localization error is high? By default rtabmap restarts at the last saved location in the map. If the robot is moved before restarting, the robot will not be at the right place, then a localization should happen to relocalize the robot. In simulation, I don't know your environment, but maybe there a re not enough visual features to relocalize.

dilipdhapse, 2019-09-25:
Thank you very much for the reply sir.

It is localizing in the position where the match ID position is found.
suppose my current robot position in the env is (xyz rpy 1.5 2 0 0 0 0).
from here I am getting one frame with ID 560. this frame matches with ID
205 which was recorded at (xyz rpy 2 2.1 0 0 0 0.23) in database.

so I don't have recorded frame at  (xyz rpy 1.5 2 0 0 0 0) while slamming.

so there is a lot different in actual frame pose and math frame.

how to tackle this problem.

matlabbe, 2019-09-27:
I think the loop closure detection may be good, but the transformation estimation may not. Referring to your post here: http://official-rtab-map-forum.67519.x6.nabble.com/localization-accuracy-of-rtabmap-ros-package-in-simulated-environment-td6286.html

The problem is that features used for loop closure detection (like the house and tree) are somewhat outliers (because they are too few) during motion estimation because there are many more features on the repetitive grass. If 90% of the features are on the grass, you will likely have wrong localizations like that. In SLAM mode, this kind of loop closure would be rejected because the graph will deform too much, but in localization mode, there is no such check (unless `RGBD/MaxOdomCacheSize` is not zero).

I would suggest to change the grass texture by a plain green color without texture. Or set the ROI as in post [here](http://official-rtab-map-forum.67519.x6.nabble.com/localization-accuracy-of-rtabmap-ros-package-in-simulated-environment-tp6286p6287.html) to avoid extracting features on the ground.

cheers,
Mathieu

dilipdhapse, 2019-09-27:
Thanks Introlab/Rtabmap_ros,

I will try these modifications.