introlab/rtabmap_ros #354: rtabmap_ros with gazebo sometimes has too large of map

nathangeorge1, 2019-09-25:
Hello,

I created a simple Gazebo world and a launch for Rtabmap and about 80% of the time, it gives:

```
[ERROR] (2019-09-25 14:46:50.399) OccupancyGrid.cpp:1091::update() Large map size!! map min=(-3.244275, -3.805648) max=(588270292926873203486941511680.000000,1857818376606259870670258176.000000). There's maybe an error with the poses provided! The map will not be created!
[N=/rtabmap/rtabmap,F=publishMaps-1451]: Grid map is empty! (local maps=1)
```

Strange this is sometimes, everything works perfectly.

Thanks,
Nathan

matlabbe, 2019-09-25:
Check odometry poses when this happens, maybe there is something wrong there. You can also share the database to see better the data in it.

nathangeorge1, 2019-09-26:
Rtabmap is now working in Gazebo, but that's if I use the default world `file="$(find gazebo_ros)/launch/empty_world.launch"`

My odom seems normal; so it seems the source of the problem is when I used my world, attached here (added .txt so it would upload): [trial.world.txt](https://github.com/introlab/rtabmap_ros/files/3659047/trial.world.txt)

There must be something abnormal about my .world file, but I can't figure it out. Any ideas?

nathangeorge1, 2019-09-26:
It's working!

It turns out all I need to do is remove the number of items in my custom Gazebo world and it's working.

Do you know why removing about 20 items from the Gazebo world would affect map size so greatly? This is important because I want my system to work with larger maps.

The values `max=(588270292926873203486941511680.000000,1857818376606259870670258176.000000` appear too great for the change I made; so maybe there's a division by 0 happening?

matlabbe, 2019-09-27:
I am more interested of the odometry values **when** you get "588270292926873203486941511680.000000,1857818376606259870670258176.000000" error. Can you give all commands you used to reproduce the problem? I could try the same environment on my computer to see if I can reproduce the error.

nathangeorge1, 2019-09-27:
This should be everything you need:

- download [Navigation.launch.txt](https://github.com/introlab/rtabmap_ros/files/3663757/Navigation.launch.txt)
- download [Gazebo.launch.txt](https://github.com/introlab/rtabmap_ros/files/3663758/Gazebo.launch.txt)
- download [MoveBase.launch.txt](https://github.com/introlab/rtabmap_ros/files/3663762/MoveBase.launch.txt)
- download [Rtabmap.launch.txt](https://github.com/introlab/rtabmap_ros/files/3663761/Rtabmap.launch.txt)
- remove .txt from the end of everything
- all files have "find" syntax for automatic pathing; change them to your packages
- do `roslaunch pkg Navigation.launch`
- let me know if this errors and you need something

These output the following continuously:
```
[ERROR] (2019-09-27 12:04:55.542) OccupancyGrid.cpp:1091::update() Large map size!! map min=(-3.228411, -3.927270) max=(8937997447269975223959552.000000,41826001596101110530048.000000). There's maybe an error with the poses provided! The map will not be created!
[N=/rtabmap/rtabmap,F=publishMaps-1451]: Grid map is empty! (local maps=1)
```

matlabbe, 2019-09-27:
`nm_navigation` package is missing. It seems there are some configuration files in that package.

nathangeorge1, 2019-09-27:
Yes "nm_navigation" is my package but I attached all relevant docs above. It's what I meant by 

`all files have "find" syntax for automatic pathing; change them to your packages`

Basically change `nm_navigation` to your testing package

matlabbe, 2019-09-27:
I don't have the following stuff:
```
$(find nm_navigation)/conf/turtlebot3_$(arg model).urdf.xacro
$(find nm_navigation)/conf
$(find nm_navigation)/launch/default_params/smoother.yaml
$(find nm_navigation)/launch/default_params/costmap_common_params.yaml
		<rosparam file="$(find nm_navigation)/launch/default_params/costmap_common_params.yaml" command="load" ns="global_costmap" />
		<rosparam file="$(find nm_navigation)/launch/default_params/costmap_common_params.yaml" command="load" ns="local_costmap" />
		<rosparam file="$(find nm_navigation)/launch/default_params/local_costmap_params.yaml" command="load" />
		<rosparam file="$(find nm_navigation)/launch/default_params/global_costmap_params.yaml" command="load" />
		<rosparam file="$(find nm_navigation)/launch/default_params/dwa_local_planner_params.yaml" command="load" />
		<rosparam file="$(find nm_navigation)/launch/default_params/move_base_params.yaml" command="load" />
		<rosparam file="$(find nm_navigation)/launch/default_params/global_planner_params.yaml" command="load" />
		<rosparam file="$(find nm_navigation)/launch/default_params/navfn_global_planner_params.yaml" command="load" />

```
Note that I can disable move_base, but for the robot description, you don't seem to use the default turtlebot3 as there are zed topics...

nathangeorge1, 2019-09-27:
Here you go: [matlabbe_files.tar.gz](https://github.com/introlab/rtabmap_ros/files/3664294/matlabbe_files.tar.gz)

I think move_base is necessary to activate the bug. There is ZED stuff in there, but you should be able to ignore it. I can't actually give nm_navigation yet because that's my product but I can give you what you need. Eventually a free version with all this will be on my website; actually soon after this bug is fixed! Also just realized I can just upload a tar.gz

matlabbe, 2019-09-28:
I did move around 10 minutes in the turtlebot3 kitchen world and no errors... Which version of rtabmap are you using? Can you also reproduce the bug with the [official turtlebot3+rtabmap navigation demo](http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot#Turtlebot3_On_Melodic)?

nathangeorge1, 2019-09-30:
I am using rtabmap version 0.19.4 and rtabmap_ros version 0.19.3, installed from source. This error seems to only happen with the newer versions, when using lidar.

I can try to reproduce with that demo, but will need some time. If it's not imperative, I can come back to this later.

matlabbe, 2019-09-30:
If you have a database to share, I could also take a look.

nathangeorge1, 2019-09-30:
Here you go: [bug.db.tar.gz](https://github.com/introlab/rtabmap_ros/files/3672380/bug.db.tar.gz)

matlabbe, 2019-09-30:
The laser scan seems having very large values. A quick fix is to set `Grid/RangeMax` to 10 for example to filter them. This fixed the error on my side.

nathangeorge1, 2019-09-30:
That did it! That's all it was...the simulated lidar was creating too much data. Thanks for all the help!

closed 2019-09-30 by nathangeorge1

Ekanshh, 2020-03-18:
> The laser scan seems having very large values. A quick fix is to set `Grid/RangeMax` to 10 for example to filter them. This fixed the error on my side.

That solved my problem too. Thanks :)