introlab/rtabmap_ros #355: Dense Point Cloud

caiofis, 2019-09-26:
I am trying to get a more dense point cloud at the output, I already changed all the parameters to make it denser, but with no results. It seems that is have some kind of voxel filtering inside even if I set it to minimal. 
Is there any way of remove it to get the denser point cloud possible?

cheers,
Caio

matlabbe, 2019-09-27:
Do you mean `cloud_map` output? If so, it is indeed downsampled to 5 cm voxel, with parameter `Grid/VoxelSize`. 

Note that producing **online** more denser clouds may be not optimal. Why you want a dense cloud **online**? You can export a full resolution cloud **offline** using exporting tool of rtabmap. Just open the database in rtabmap standalone app, then do "File->Export 3D clouds...", check "Regenerate clouds", and set the density you want.

cheers,
Mathieu

caiofis, 2019-09-27:
Yeah, I mean `cloud_map`. I need to perform some 3D segmentation, for now I am using the raw data from Kinect, which is pretty noisy. My idea was to use the RTABMap as a registration tool, maybe using the passthrough filter to publish only a fraction of the dense point cloud. Do you think I should use other tool since I just need a reconstruction with local consistency? 
I tried to perform ICP with the measurements from Kinect, but my output is kinda of noisy, how do you remove the outliers to merge the point clouds with ICP? I am diving into your code here, but it is quite big.
;)

matlabbe, 2019-09-28:
You can decrease `Grid/VoxelSize` to 1 cm voxel for example to make it a little denser. But as you need only local dense clouds, you may subscribe to odometry and the point cloud of the kinect, then assemble them yourself at the resolution you want. You can also look at the [point_cloud_assembler](https://github.com/introlab/rtabmap_ros/blob/master/src/nodelets/point_cloud_assembler.cpp) nodelet of rtabmap_ros or the one from http://wiki.ros.org/laser_assembler that do such stuff.

By default, rtabmap does visual registration, not icp with the kinect-like data.