introlab/rtabmap_ros #358: low fps

jungladicitta, 2019-10-09:
I have tested rtabmap on my jetson tx2 with Intel Realsense d435 camera and got very low fps and freezes.
I tried to run rs_rgbd.launch separately and it works fine.
Here is rtabmap parameters: [rtabmap.launch](https://github.com/jungladicitta/turtlebot3_slam_3d/blob/only_camera_2/turtlebot3_slam_3d/launch/rtabmap.launch)

What can I do to speedup rtabmap?

matlabbe, 2019-10-09:
Can you show the cpu usage and terminal output of rtabmap launch file? 
`$ top`
That rtabmap configuration would be quite low in cpu, with map refresh rate at 1 Hz.

jungladicitta, 2019-10-10:
@matlabbe 
I run turtlbot3_slam_3d.launch file. Included launch files you can view in the same [directory](https://github.com/jungladicitta/turtlebot3_slam_3d/tree/only_camera_2/turtlebot3_slam_3d/launch)

[image]

```
nvidia@tegra-ubuntu:~$ roslaunch turtlebot3_slam_3d turtlebot3_slam_3d.launch
... logging to /home/nvidia/.ros/log/b26cde06-eb20-11e9-8806-00044bc771c7/roslaunch-tegra-ubuntu-28643.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
 
started roslaunch server http://JETSON_HOSTNAME:36666/
 
SUMMARY
========
 
PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 15
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
[... 220 lines cut ...]
[ INFO] [1570686240.366913271]: rtabmap (36): Rate=1.00s, Limit=0.700s, RTAB-Map=0.2103s, Maps update=0.0330s pub=0.0000s (local map=1, WM=1)
[ INFO] [1570686242.588266209]: rtabmap (37): Rate=1.00s, Limit=0.700s, RTAB-Map=0.1546s, Maps update=0.0294s pub=0.0001s (local map=1, WM=1)
[ INFO] [1570686244.557585882]: rtabmap (38): Rate=1.00s, Limit=0.700s, RTAB-Map=0.1928s, Maps update=0.0341s pub=0.0001s (local map=1, WM=1)
[ INFO] [1570686246.655089502]: rtabmap (39): Rate=1.00s, Limit=0.700s, RTAB-Map=0.1788s, Maps update=0.0316s pub=0.0001s (local map=1, WM=1)
[ INFO] [1570686247.656808347]: rtabmap (40): Rate=1.00s, Limit=0.700s, RTAB-Map=0.1428s, Maps update=0.0433s pub=0.0001s (local map=1, WM=1)
[ INFO] [1570686249.497560847]: rtabmap (41): Rate=1.00s, Limit=0.700s, RTAB-Map=0.1746s, Maps update=0.0500s pub=0.0001s (local map=1, WM=1)
[ INFO] [1570686251.566788767]: rtabmap (42): Rate=1.00s, Limit=0.700s, RTAB-Map=0.1861s, Maps update=0.0289s pub=0.0001s (local map=1, WM=1)
^C[rviz-9] killing on exit
[rtabmap/rtabmap-8] killing on exit
[camera/points_xyzrgb_hw_registered-7] killing on exit
[camera/color_rectify_color-6] killing on exit
[robot_state_publisher-3] killing on exit
[camera/realsense2_camera-5] killing on exit
[camera/realsense2_camera_manager-4] killing on exit
[odomtry_publisher-2] killing on exit
[pointcloud_to_laserscan-1] killing on exit
terminate called without an active exception
rtabmap: Saving database/long-term memory... (located at /home/nvidia/.ros/rtabmap.db)
rtabmap: 2D occupancy grid map saved.
rtabmap: Saving database/long-term memory...done! (located at /home/nvidia/.ros/rtabmap.db, 10 MB)
shutting down processing monitor...
... shutting down processing monitor complete
done

```

jungladicitta, 2019-10-11:
@matlabbe 
> That rtabmap configuration would be quite low in cpu, with map refresh rate at 1 Hz.

why? what is the problem to refresh every second? which parameters can I change?

nacorti, 2019-10-13:
@jungladicitta use the `/rtabmap/rtabmap/Rtabmap/DetectionRate: 1.0` param in your launch file to set the detection rate in Hz. Right now it's 1Hz, which updates once per second.

That said, I'm also using a D435 on a Jetson Nano, and am noticing that I get much lower FPS off my depth stream (10-15) while rendering in RViz than I have set incoming from RealSense (30 FPS). Also wondering if there are any best practices for optimizing memory/compute on small boards like Jetson.

matlabbe, 2019-10-15:
The detection rate is set by default at 1 Hz, I rarely set it higher unless the robot is moving fast. In the logs, the update time is between 150 and 200 ms, so only 15-20% of one core would be used for rtabmap.

I would suggest not running rviz on the jetson. And if you run rviz on remote computer, use an ethernet cable, and don't show raw point clouds from realsense. If you only show `/rtabmap/mapData`, the bandwidth would be not too high.

If rtabmap is publishing slower than 1 Hz with its processing time around 200 ms, the input topics may be poorly synchronized, causing some lags. If you have odom, scan, image topics, verify all their frame rate with `rostopic hz` (on the jetson).

jungladicitta, 2019-10-15:
@matlabbe thanks a lot for so detailed answer.

What do mean by poorly synchronized? you mean the input topics should be publishing with the same rate, for example 100 Hz?

I will try to set the same rate to the input topics, and then running rviz on remote computer.

jungladicitta, 2019-10-15:
here is my input topics for rtabmap:
```
nvidia@tegra-ubuntu:~$ rostopic hz /camera/color/camera_info 
subscribed to [/camera/color/camera_info]
average rate: 13.375
	min: 0.013s max: 0.139s std dev: 0.04586s window: 12
average rate: 13.240
	min: 0.012s max: 0.139s std dev: 0.04470s window: 24
average rate: 12.122
	min: 0.012s max: 0.139s std dev: 0.04621s window: 35
average rate: 11.998
	min: 0.012s max: 0.139s std dev: 0.04869s window: 46
average rate: 12.024
	min: 0.012s max: 0.139s std dev: 0.04828s window: 59
^Caverage rate: 11.872
	min: 0.012s max: 0.139s std dev: 0.04759s window: 66
nvidia@tegra-ubuntu:~$ rostopic hz /camera/color/image_rect_color
subscribed to [/camera/color/image_rect_color]
average rate: 14.930
	min: 0.021s max: 0.106s std dev: 0.01705s window: 15
average rate: 15.008
	min: 0.021s max: 0.106s std dev: 0.01222s window: 30
^Caverage rate: 14.958
	min: 0.021s max: 0.106s std dev: 0.01100s window: 38
nvidia@tegra-ubuntu:~$ rostopic hz /camera/depth/image_rect_raw
subscribed to [/camera/depth/image_rect_raw]
average rate: 15.319
	min: 0.008s max: 0.105s std dev: 0.02354s window: 15
average rate: 15.039
	min: 0.008s max: 0.105s std dev: 0.01939s window: 30
average rate: 15.084
	min: 0.008s max: 0.111s std dev: 0.01912s window: 45
^Caverage rate: 15.082
	min: 0.008s max: 0.111s std dev: 0.01981s window: 52
nvidia@tegra-ubuntu:~$ rostopic hz /odom
subscribed to [/odom]
average rate: 10.019
	min: 0.081s max: 0.118s std dev: 0.01225s window: 9
average rate: 10.008
	min: 0.081s max: 0.118s std dev: 0.01042s window: 19
average rate: 10.000
	min: 0.081s max: 0.118s std dev: 0.00904s window: 29
average rate: 9.997
	min: 0.081s max: 0.118s std dev: 0.00870s window: 39
average rate: 10.003
	min: 0.075s max: 0.123s std dev: 0.00947s window: 49
average rate: 10.001
	min: 0.075s max: 0.123s std dev: 0.00885s window: 60
average rate: 10.001
	min: 0.075s max: 0.123s std dev: 0.00819s window: 70
average rate: 10.001
	min: 0.075s max: 0.123s std dev: 0.00788s window: 80
^Caverage rate: 10.002
	min: 0.075s max: 0.123s std dev: 0.00774s window: 83
nvidia@tegra-ubuntu:~$ rostopic hz /scan
subscribed to [/scan]
average rate: 8.868
	min: 0.099s max: 0.124s std dev: 0.00915s window: 8
average rate: 8.700
	min: 0.098s max: 0.150s std dev: 0.01421s window: 17
average rate: 8.826
	min: 0.097s max: 0.150s std dev: 0.01357s window: 26
average rate: 9.015
	min: 0.091s max: 0.150s std dev: 0.01304s window: 35
average rate: 8.885
	min: 0.091s max: 0.159s std dev: 0.01445s window: 44
average rate: 8.830
	min: 0.091s max: 0.159s std dev: 0.01451s window: 52
average rate: 8.837
	min: 0.076s max: 0.159s std dev: 0.01628s window: 61
average rate: 8.917
	min: 0.076s max: 0.159s std dev: 0.01544s window: 71
average rate: 8.946
	min: 0.076s max: 0.159s std dev: 0.01487s window: 80
```
What do you think? Are these topics poorly synchronized?

nacorti, 2019-10-15:
@matlabbe appreciate the detailed response, thank you.

wrt resource consumption, I find that my CPU usage (all four cores) hovers around 90%, while GPU fluctuates pretty wildly and RAM/Swap is pretty underutilized. Is there an rtabmap setting that will use more of my RAM? Wondering if it'll get my CPU usage down a bit.

wrt topic framerates, my image and depth framerates are synced at about 15-25 hz, but my odom/pose is coming in at around 200 hz. What's happening is as my pose changes, rtabmap projects the stale depth/image map into an incorrect/inaccurate place on the map (until the next depth map comes in). Is there an rtabmap parameter that could prevent this from happening? `approx_sync` is false, and when it's set to true, I get errors because my timestamps aren't exactly the same. The alternative is downsampling the odom topic, which I'd like to avoid if possible.

jungladicitta, 2019-10-21:
@matlabbe 
I have checked input topics rates for rtabmap and they are ok.
I have tried to run rviz on remote computer with using 5 GHz wifi and is still very slow.
Moreover, tf refreshes with big delay. Approximately 5-10 seconds.
what else can you recommend to check?

matlabbe, 2019-10-21:
@nacorti When you say CPU usage at 90%, how much is form rtabmap? From this [post](https://github.com/introlab/rtabmap_ros/issues/358#issuecomment-540386762) above, rtabmap is about 20% of one core. Actually point_cloud_to_laser_scan node is 117% usage. As for odometry topic published at 200 Hz, you can increase the `queue_size` parameter of rtabmap node. It will help to synchronize topics with very different frame rates. Another approach is to use TF for odometry by setting `odom_frame_id` parameter. 

@jungladicitta Be careful when launching RVIZ on remote computer to not subscribe to point cloud or uncompressed image topics. Check your network bandwidth if it is stable. I would suggest to connect through ethernet to debug easily without wifi latencies and bandwidth limitations.

nacorti, 2019-10-23:
@matlabbe 

Wasn't able to deduce how much CPU usage was from Rtabmap vs RViz - I don't have an Ubuntu machine handy besides my Jetson to take your advice about running RViz separately over an ethernet cable. But, I completed a series of steps that (I think) increased my performance pretty significantly

1) I turned off the IR streams from my D435, so I just get Depth and Color. Reduces load over USB/superfluous ROS topics
2) Followed your advice about `queue_size`, set to 200
3) Followed your advice from [rtabmap #449](https://github.com/introlab/rtabmap/issues/449), I increased `Grid/CellSize` and decreased `Vis/MaxFeatures`
4) I actually increased the `DetectionRate` to 30Hz, which allows me to use fewer features and thus do less pre-processing on the image bc subsequent images aren't as different as they would be at 1HZ

Right now the big issue I have is running out of RAM once my map gets too large, but I think that's a limitation of the Nano more than anything. Going to try `Mem/ImagePostDecimation` and `Mem/ReduceGraph` to see if any change.

I'm also in the process of building OpenCV4 w GPU, then rebuilding RTabMap with OpenCV4 and seeing if that can boost framerate. I'll let you know how it goes.

Comments/critiques always welcome, and of course thank you for your usual prompt and informative replies :)

jungladicitta, 2019-10-23:
@matlabbe every topics published by camera is uncompressed right? I mean
```
/camera/depth/image_rect_raw
/camera/color/image_rect_color
```

matlabbe, 2019-10-23:
@nacorti For the RAM problem, it depends of what you want to keep in the database. If you don't care about keeping raw RGB/depth image, you can set `Mem/BinDataKept` to false. If you don't need occupancy grids, set `RGBD/CreateOccupancyGrid` to false. 30 Hz is quite high, by decreasing the number of features in the images, you can also decrease the loop closure detection performance. If you don't care about global map optimization and loop closures (only in odometry mode), you can disable features extraction with `Kp/MaxFeatures=-1`. Mem/ImagePostDecimation can be a good way to keep raw RGB/Depth images in database at lower resolution, if you want to keep them. You may also check if it is only rtabmap that is taking a lot of RAM, or it could be also RVIZ or rtabmapviz if you show the point clouds.

@jungladicitta Yes, without suffix like "/compressed", they are the raw images.

jungladicitta, 2019-10-24:
I tried to connect everything to jetson via usb 3.0 hub, I mean camera, wheels and lidar, and now it looks ok in rviz.
I think problem was that I was transfering camera raw data through wi-fi.

closed 2019-10-24 by jungladicitta