introlab/rtabmap_ros #363: Build with ORB SLAM2

yubaoliu, 2019-11-05:
Hi,

I try to build with ORB SLAM2, I see this error:

`[ 82%] Generating Javascript code from rtabmap_ros/SetLabel.srv
[ 82%] Generating EusLisp code from rtabmap_ros/Point3f.msg
/usr/bin/ld: cannot find -lpangolin
collect2: error: ld returned 1 exit status
rtabmap_ros/CMakeFiles/rtabmap_rgbd_sync.dir/build.make:1222: recipe for target '/home/yubao/data/catkin_ws/devel/lib/rtabmap_ros/rgbd_sync' failed
make[2]: *** [/home/yubao/data/catkin_ws/devel/lib/rtabmap_ros/rgbd_sync] Error 1
CMakeFiles/Makefile2:16659: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_rgbd_sync.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_rgbd_sync.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....`

Here is my flow:
1. Build Pangolin standalone and add to library link path
` export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$HOME/data/software/Pangolin/lib`

`export Pangolin_DIR="$HOME/data/software/Pangolin/lib/cmake/Pangolin"`

2. Build ORB SLAM2 succeed
3.  Set ORB ROOT PATH
`export ORB_SLAM2_ROOT_DIR=/home/yubao/data/catkin_ws/src/ORB_SLAM2`

[image]

[image]

4. Build Rtabmap sucessfully

5. Build rtabmap_ros under my ROS workspace use `catkin_make`, then I got that error.
(Note: it succeeds without build ORB SLAM2 option)

Thanks

matlabbe, 2019-11-06:
On my system with ORB_SLAM2, I **make install** Pangolin, so I don't need to modify LD_LIBRARY_PATH for it. You may look at this docker file which shows how to install ORB_SLAM2: 
https://github.com/introlab/rtabmap/blob/d4cb623d9ca522229bfeff4766bf8e22f9a9c17d/docker/jfr2018/Dockerfile#L162-L170

yubaoliu, 2019-11-06:
Thanks,
But I don't want to install Pangolin to my system path. Actually I can build ORB_SLAM2 standalone successfully. 
The problem happens when I build ORB_SLAM2 in Rtabmap/Rtabmap_ros. I set the Cmake DIR path and set the library link path LD_LIBRARY_PATH. However, Rtabmap_ros still cannot find it when compiling using catkin_make.

matlabbe, 2019-11-16:
The `-lpangolin` flag would be coming from imported dependencies of rtabmap cmake config file. You can try to remove it from the installed `RTABMapConfig.cmake`, then build rtabmap_ros again. If it is building, we may look to remove this dependency flag from [here](https://github.com/introlab/rtabmap/blob/5f12459437d07aef772ead2ce48a0bd8412ccf07/CMakeLists.txt#L787).

zhuoyan28, 2024-03-24:
Can someone tell
1, RTAB-MAP commands to run KITTI image dataset alone, such as binocular images?
Where will the output be stored in the ros environment?
2. ORBSLAM+RTAB-MAP running KITTI image dataset, such as binocular image command?
Where will the output be stored in ros environment?

matlabbe, 2024-03-25:
1. For the kitti image dataset, you could process it with the command line tool `rtabmap-kitti_dataset` (type in terminal alone and you will see usage and an example at the end). It is independent of ros. The database could be use in ros afterwards if needed, though I don't think there is a reason to do so with kitti dataset (unless it was your own dataset where a robot would navigate in afterwards).

2. The command would look like:
```
rtabmap-kitti_dataset \
       --Rtabmap/PublishRAMUsage true\
       --Rtabmap/DetectionRate 2\
       --Rtabmap/CreateIntermediateNodes true\
       --RGBD/LinearUpdate 0\
       --GFTT/QualityLevel 0.01\
       --GFTT/MinDistance 7\
       --Odom/Strategy 5\
       --OdomORBSLAM2/VocPath ~/ORB_SLAM2/vocabulary/ORBvoc.txt\
       --OdomORBSLAM2/ThDepth 40\
       --OdomORBSLAM2/Fps 10\
       --OdomORBSLAM2/MaxFeatures 2000
       --Mem/STMSize 30\
       --Kp/MaxFeatures 750\
       --Vis/MaxFeatures 1500\
       --gt "~/KITTI/devkit/cpp/data/odometry/poses/07.txt"\
       ~/KITTI/dataset/sequences/07

```

Note that if you want to reproduce the results of the [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf) (with same parameters), see https://github.com/introlab/rtabmap/tree/master/docker/jfr2018

zhuoyan28, 2024-03-25:
really appreciate your patience! Thanks.

---- Replied Message ----
| From | ***@***.***> |
| Date | 03/25/2024 09:05 |
| To | introlab/rtabmap_ros ***@***.***> |
| Cc | Dan ***@***.***>,
Comment ***@***.***> |
| Subject | Re: [introlab/rtabmap_ros] Build with ORB SLAM2 (#363) |

For the kitti image dataset, you could process it with the command line tool rtabmap-kitti_dataset (type in terminal alone and you will see usage and an example at the end). It is independent of ros. The database could be use in ros afterwards if needed, though I don't think there is a reason to do so with kitti dataset (unless it was your own dataset where a robot would navigate in afterwards).

The command would look like:

rtabmap-kitti_dataset \
       --Rtabmap/PublishRAMUsage true\
       --Rtabmap/DetectionRate 2\
       --Rtabmap/CreateIntermediateNodes true\
       --RGBD/LinearUpdate 0\
       --GFTT/QualityLevel 0.01\
       --GFTT/MinDistance 7\
       --Odom/Strategy 5\
       --OdomORBSLAM2/VocPath ~/ORB_SLAM2/vocabulary/ORBvoc.txt\
       --OdomORBSLAM2/ThDepth 40\
       --OdomORBSLAM2/Fps 10\
       --OdomORBSLAM2/MaxFeatures 2000
       --Mem/STMSize 30\
       --Kp/MaxFeatures 750\
       --Vis/MaxFeatures 1500\
       --gt "~/KITTI/devkit/cpp/data/odometry/poses/07.txt"\
       ~/KITTI/dataset/sequences/07

Note that if you want to reproduce the results of the paper (with same parameters), see https://github.com/introlab/rtabmap/tree/master/docker/jfr2018

zhuoyan28, 2024-04-09:
Hello author, two questions please
1、According to the dockerfile of jfr2018, the file run_all.sh only keeps rtab-map(f2m) with orb2-rtab(os2) and the results can be reproduced. Then.
I want to reproduce the result of stereo of ORBSLAM2 standalone , how can I do it?
2, I want to modify the code of rtab-map, how to run the modified code?

matlabbe, 2024-04-11:
The jfr2018 docker is a snapshot of the code used for the paper. To use latest rtabmap, use docker under `latest` folder: https://github.com/introlab/rtabmap/tree/master/docker/jfr2018/latest

Normally, `run_kitti_dataset.sh` should use the right rtabmap built with orb-slam when using docker approach. Make sure to change the image name in case you rebuilt your own docker image: https://github.com/introlab/rtabmap/blob/4e7110930c594187104c52083f2cf469157276c1/docker/jfr2018/run_kitti_datasets.sh#L96

For "ORBSLAM2 standalone", do you mean using only ORBSLAM2 evaluation tools? I think they have one for kitti: https://github.com/raulmur/ORB_SLAM2/tree/master/Examples/Stereo

To do custom tests, either you change the code and rebuild the container, or build rtabmap on host machine with dependencies you need. You can use `run_kitti_dataset.sh` locally without docker container if you set [DOCKER](https://github.com/introlab/rtabmap/blob/4e7110930c594187104c52083f2cf469157276c1/docker/jfr2018/run_kitti_datasets.sh#L4) variable to 0.

zhuoyan28, 2024-04-25:
The algorithm system was installed by myself, not the docker file
the command is :
rtabmap-kitti_dataset \
       --Rtabmap/PublishRAMUsage false\
       --Rtabmap/DetectionRate 2\
       --Rtabmap/TimeThr 0 \
       --Rtabmap/MemoryThr 0\
       --Rtabmap/LoopThr 0.11\
       --GFTT/QualityLevel 0.01\
       --GFTT/MinDistance 7\
       --Odom/Strategy 0\
       --OdomORBSLAM2/VocPath /home/shen/rtab_ws/src/ORB_SLAM2/Vocabulary/ORBvoc.txt\
       --OdomORBSLAM2/ThDepth 40\
       --OdomORBSLAM2/Fps 10\
       --OdomORBSLAM2/MaxFeatures 3000\
       --OdomF2M/MaxSize 3000 \
       --Odom/KeyFrameThr 0.3\
       --Mem/STMSize 30\
       --Mem/RehearsalSimilarity 0.2\
       --Kp/MaxFeatures 750\
       --Vis/MaxFeatures 1500\
      --Vis/CorNNDR 0.6\
      --Vis/CorGuessWinSize 20\
      --Vis/MinInliers 20\
       --gt "/home/shen/study_learning/datasets/kitti/devkit/cpp/data/odometry/poses/04.txt"\
       /home/shen/study_learning/datasets/kitti/dataset/sequences/04

but the result for kitti-04 is 
[image]

[image]
How to reproduce your results？

matlabbe, 2024-05-12:
To recover results from the paper, you should use `rtabmap-report --scale`:
```
```