introlab/rtabmap_ros #364: Velodyne with RGB stream from Zed

FarBuildMan, 2019-11-06:
Hi @matlabbe 

I would like to do RGBD mapping with velodyne lidar and zed camera and external odometry source. I can get RGB data from Zed camera and PointXYZ data from VLP16. Eventually, I would like to use XYZ values to register depth value for respective RGB pixel. In other words I want to colorize velodyne pointcloud. i have the form at http://official-rtab-map-forum.67519.x6.nabble.com/RTAB-setup-with-a-3D-LiDAR-td1080.html . 
It was helpful to create my own launch file. But i have a problem UASSERT(!laserScan->data.empty()); condition fails and it stops mapping.

My launch file is here:
      <node pkg="rtabmap_ros" type="pointcloud_to_depthimage" name="pointcloud_to_depthimage"  output="screen">
        <remap from="/rtabmap/cloud"			to="/velodyne_points"/>
        <remap from="camera_info"       	to="/$(arg zed_namespace)/$(arg zed_node_name)/rgb/camera_info" />
        <param name="fixed_frame_id"		value="odom_link"/>
        <param name="approx_sync"		type="bool"		value="true"/>
        <param name="fill_holes_size"		type="int"		value="5"/>
        <param name="queue_size"		type="int"		value="100"/>
        <param name="wait_for_transform"		type="double"		value="0.1"/>
      </node>
For your information I am successful to do SLAM with Zed+external odometry with RTABMap. 

Any help from fellow community members is greatly appreciated too.

Cheers.

matlabbe, 2019-11-16:
The assert tells that the input scan should not be empty. Check why is `/velodyne_points` sometimes empty.

FarBuildMan, 2019-11-18:
"rostopic echo/hz /velodyne_points" seems the topic isn't empty and is publishing at 8-10 Hz and is_dense=true. My external odometry is publishing at 150 Hz. I made wait_for_transform=6, queue_size=100 and approx_sync=true. Do you think the huge difference between publishing rates can be the reason?

matlabbe, 2019-11-18:
To make sure, is this [line](https://github.com/introlab/rtabmap/blob/cf1997804ffadde7a830e7dcc8d00c4750ff01ab/corelib/src/util3d.cpp#L2850) causing the assert? That cloud is created here:
https://github.com/introlab/rtabmap_ros/blob/b19a82ef34c24da8ccf28d184f84e95a81ea6d9a/src/nodelets/pointcloud_to_depthimage.cpp#L199-L200
with `pointCloud2Msg` the cloud received. If some messages are not empty, I don't know why the callback is called with empty clouds. You could add a log to see the actual size of `pointCloud2Msg`:
```cpp
ROS_WARN("pointCloud2Msg size = %d", pointCloud2Msg->data.size());
```

For the topic frame rate, actually it is the camera info and point cloud messages that are synchronized in that node. Odometry messages are not used, only TF is used. If you have point clouds published at 10 Hz and camera info at 30 Hz, you could set `queue_size` as low as 3, but I would keep it between 5 and 10 in case the topic rates are varying.

If you have odometry updated very fast, you could keep `wait_for_transform` easily under 1 second.

FarBuildMan, 2019-11-18:
Thanks for your prompt reply. I added the log check as you suggested and attach the screenshot here.
[image]

As you it first has some contents but loses suddenly. However, "rostopic echo /velodyne_points" still returns non-empty and non-zero value with is_dense=true. 
I tried **queue_size=3;5** and **wait_for_transform=0.1;0.5;0.8**.

matlabbe, 2019-11-19:
It may be difficult to see in terminal with `rostopic echo /velodyne_points` as a lot of info would be printed. You may try:
```
$ rostopic echo /velodyne_points/width
```

In the log, `pointCloud2Msg size = 0`, which means it really received an empty cloud. Can you record a rosbag of the velodyne cloud until `pointcloud_to_depthimage` asserts?
```
$ rosbag record /velodyne_points
```

It is not a problem of `queue_size` or `wait_for_transform`.

FarBuildMan, 2019-11-19:
it seems **/velodyne_points/width** gets zero roughly once in every 20 messages which is almost 2 secs. 
[image]
 I recorded velodyne_points it's here https://drive.google.com/file/d/11FX5lINVUguR__rfarJoF3cS6cHv-6E0/view?usp=sharing .
Actually I digged into code, it seems like rtabmap **util2d::getDepth** returns 0, hence **util3d::projectDepthTo3D** assigns `pt.x = pt.y = pt.z = std::numeric_limits<float>::quiet_NaN();`. Therefore **Cloud with only NaN values created!** warning pops-up in previous screenshot.

Do you think turning off the assert is okay? Since /velodyne_points/width gets 0 once in a while.

matlabbe, 2019-11-19:
You may look at why the node publishing the velodyne topic is publishing empty clouds. I still updated the code in this [commit](https://github.com/introlab/rtabmap_ros/commit/26b8fb5027dfc25af990a15fe358d7fdd8699cd5) to handle the case of an empty cloud, which could happen if we hide completely the sensor.

For the other warning `Cloud with only NaN values created! `: which node is logging this warning? Is it rtabmapviz? If so, it is maybe that the depth image returned is empty even if the cloud is not empty. This happens if no points can be projected in the camera, because the TF is wrong between the camera and the lidar. Can you show TF with the lidar point cloud in RVIZ? or record a bag with:
```bash
$ rosbag record /tf /tf_static /velodyne_points /zed/zed/rgb/camera_info
```
well, "`/zed/zed/rgb/camera_info`" the actual name of your camera_info topic.

jessearmand, 2019-11-26:
This could be related to a change of `PointCloud2` structure in `velodyne_laserscan` package, which i think is related to ros-drivers/velodyne#292, and it's fixed in os-drivers/velodyne#316.

However I can see the registered depth image using [velodyne driver 1.5.2](https://github.com/ros-drivers/velodyne/tree/1.5.2) and the changes in 26b8fb5.

Would that affect the depth image projection?

matlabbe, 2019-12-01:
Can you show a screenshot of the RGB image and the generated depth image from velodyne? If the velodyne sends empty clouds, the depth image will be empty, thus causing "Cloud with only NaN values created!" warning when rtabmap is receiving the empty image.

jessearmand, 2019-12-01:
@matlabbe You can check this [video](https://youtu.be/7ms6MiTCIcI). I ran this with the changes in 26b8fb5 and [velodyne driver 1.5.2](https://github.com/ros-drivers/velodyne/tree/1.5.2). I've also seen the same result with the latest master branch in velodyne driver.

You can see the depth image on the left panel above the RGB stream.

I didn't see a completely empty image, so i guess there's no more problem? But, I did saw lines in the image that looks like the ring of the laser scans.

matlabbe, 2019-12-01:
It seems ok, the only problem I see is the drift at 12:00 where the odometry doesn't estimate correctly that the robot is moving (the wheels appear turning on place in rviz). What kind of odometry do you use? You may have also a TF problem, there is a TF flickering between two places on the robot, I cannot read which one is it.

jessearmand, 2019-12-02:
For this turtlebot, I’m using realsense T265. The wheel odometry from OpenCR is fed into the realsense camera `topic_odom_in`

Yup I noticed it’s rotating in place. Most likely because the TF data from Intel NUC which is connected to T265 is interrupted because of WiFi connection.

Have you tested how good RealSense T265 is for odometry? 

I’m using it because it’s relatively more accurate than the wheel odometry coming from OpenCR.

matlabbe, 2020-04-01:
We use wheel odometry most of the time on our ground robots (the wheel odometry is quite accurate). I didn't try to do a comparison. We have an environment with low visual features (mostly corridors with white walls), so visual odometry does always at some point strange drifts (when there is a lack of visual features).  I expect T265 odometry to suffer with the same problem. However, in a house, I think it could perform fairly and consistently as long the robot is not facing directly a room's corner without textures. Note that magnetic fields can be a problem for VIO approaches, which could cause also strange drifts.

jessearmand, 2020-04-02:
Interesting, I didn't know how magnetic field interference could affect T265. 

Do you mean VIO is affected by magnetic field? or the camera components such as the gyroscope.

matlabbe, 2020-04-03:
It depends of the VIO approach, if the approach fuses magnetometer, it can indeed be a problem in areas with magnetic field. A VIO approach could fuse only accelerometer and gyroscope with camera to avoid to be sensible to magnetic field, however IMU's yaw orientation will drift over time (which is normally corrected with magnetometer). For T265, I don't know what they are fusing, it is maybe written somewhere in their documentation. A way to find out is the move a magnet around the camera and see if it affects the orientation of the odometry.