introlab/rtabmap_ros #366: rtabmap_ros/rgbd_sync using Realsense R200  Did not receive data since 5 seconds!

CesMak, 2019-11-06:
Hey there,

I want to use rtabmap_ros/rgbd_sync for my Realsense R200 camera. This camera has a sw_registered depth path topic for this reason I changed the launch file to:

```

<launch>

  <!-- Realsense: -->
      <include file="$(find realsense_camera)/launch/r200_nodelet_rgbd.launch">
    </include>

  <arg name="compressed" default="false"/>
  <arg name="compressed_rate" default="0"/>

  <group ns="camera">
    <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="load rtabmap_ros/rgbd_sync camera_nodelet_manager" output="screen">
      <remap from="rgb/image"         to="rgb/image_rect_color"/>
      <remap from="depth_registered/sw_registered"       to="depth_registered/image_raw"/>
      <remap from="rgb/camera_info"   to="rgb/camera_info"/>
      <param name="compressed_rate" value="$(arg compressed_rate)"/>
    </node>
  </group>

</launch>
```

If I launch this file I get the Error:

Did not receive data since 5 seconds!

I also had a look on rostopic hz which worked fine or: ?
```

 rostopic hz /camera/rgb/image_rect_color /camera/depth/image /camera/rgb/camera_info 
subscribed to [/camera/rgb/image_rect_color]
subscribed to [/camera/depth/image]
subscribed to [/camera/rgb/camera_info]
           topic                rate   min_delta   max_delta   std_dev    window
================================================================================
/camera/rgb/image_rect_color   31.02   0.004314    0.03681     0.005284   33    
/camera/depth/image            30.12   0.0234      0.03758     0.002175   33    
/camera/rgb/camera_info        30.13   0.0298      0.0368      0.001721   33    
```

Is there anything I miss (maybe the /camera/depth_registered/camera_info topic ?)

matlabbe, 2019-11-16:
On the client you may be using the rtabmap_ros binaries, which unfortunatly don't have `rgbd_sync` and `rgbd_relay`. This has been fixed in https://github.com/introlab/rtabmap_ros/commit/e6e7df5beb8108dfb173dbf487d58b4c3bcdafba but until new binaires are released, you would have to build rtabmap_ros from source. To avoid recompiling also rtabmap, you can just recompile rtabmap_ros using the same version than the binaries:
```bash
$ sudo apt remove ros-melodic-rtabmap-ros
$ cd ~/catkin_ws/src
$ git clone -b melodic-devel https://github.com/introlab/rtabmap_ros.git
$ cd ~/catkin_ws
$ catkin_make
```

On remote PC, I would rather do:
```
$ rostopic hz /odom /camera/rgbd_image/compressed
```
to get the real frame rate and bandwidth usage you can get over WiFi.

ju-mingyue, 2021-07-08:
@matlabbe  
Hello, I also encountered the same problem and I don’t know how to solve it.
[image]
I use jetsonnano, ros is melodic .

matlabbe, 2021-07-22:
To debug, you should start by checking the framerate of the topics shown and see if some of them are not published or if there is a large difference of rate between them:
```
rostopic hz /t265/odom/sample /camera/rgb/image_rect_color /camera/depth_registered/image_raw /camera/rgb/camera_info
```

PS: Please don't join screenshots of text in github issues, it makes difficult for Google to find the same issue afterwards for someone else.

ju-mingyue, 2021-07-23:
@matlabbe 
Thank you very much for your help, and thank you for your suggestions.

hariharan382, 2022-07-19:
I having same issue ,but how can I debug from the frame rates

matlabbe, 2022-10-10:
@hariharan382 
1) you can know if the topic is published, if one rate is missing, debug why it is not published
2) you can see if the frame rate is different between the topic, if so, how much? `queue_size` parameter may nee to be increased if the rate is very different.