introlab/rtabmap_ros #368: Custom user data visualization

pfautley, 2019-11-19:
Hi! I'm working off of the Wifi Signal Strength Mapping (User Data Usage) example from the wiki, but modifying it to instead display temperature data on the map. I'm having some issues getting the code to work correctly and I could use some help in sorting it all out. I've modified the code to accept data from a temperature data type and send the data to ROS in a way similar to the example:

```
      ros::Time stamp = ros::Time::now();

      // Create user data [level] with the value
      cv::Mat data(1, 2, CV_64FC1);
      data.at<double>(0) = double(g_temp.temperature);
      // ROS_INFO("stored in data stucture: %f", data.at<double>(0));
      // we should set stamp in data to be able to
      // retrieve it from rtabmap map data to get precise
      // position in the graph afterward
      data.at<double>(1) = stamp.toSec();

      rtabmap_ros::UserData dataMsg;
      dataMsg.header.frame_id = frameId;
      dataMsg.header.stamp = stamp;
      // ROS_INFO("sensor ID: %d", frameId);
      rtabmap_ros::userDataToROS(data, dataMsg, true);
      tempPub.publish<rtabmap_ros::UserData>(dataMsg);
```

I then grab the data in the same way as the tutorial using the wifi sub node, but I don't seem to be getting any of the signatures with UserData, so I am repeatedly getting this error message:

> [ WARN] [1574122972.650465470]: No sensor signal detected yet in user data of map data

Here's where I think I may be running into problems. I'm running a custom mapping.launch to support my current setup. I tried to modify it to incorporate important information, but as I am relatively new to ROS and rtabmap I'm not sure what is important and what isn't. I'm happy to provide any additional information necessary to start sorting this out. If someone wouldn't mind helping me identify how to successfully modify this code for this purpose I would be very appreciative. Thanks!

matlabbe, 2019-11-19:
Can you show `rqt_graph`? Is the user data message connected to rtabmap node?

pfautley, 2019-11-19:
I'm honestly not sure if they're connected. I grabbed the result of rqt_graph and I'll post it here, but I apologize in advance for how messy it is. 
[image]

matlabbe, 2019-11-19:
You can click on tf in rqt_graph to hide tf stuff. You can also mouse over rtabmap node to see better inputs and outputs in the screenshot. 

Which is the topic that you want to connect to rtabmap? Is it `/ENV_Sensor_Data` or `/temperature`? They are both not connected to rtabmap. Under rtabmap node, your would have to remap `user_data_async`:
```xml
<node pkg="rtabmap_ros" type="rtabmap" name="rtabmap">
   ...
   <remap from="user_data_async" to="/temperature"/>
</node>
```

pfautley, 2019-11-19:
I think we're getting closer to the source of the problem here, but I'm not quite there. I may have done this in a very roundabout way, so I appreciate your help in sorting this out. I think it may be helpful for me to describe the intended function.

/ENV_Sensor_Data holds an array of sensor data being sent from an arduino. I split off the temperature data into /temperature (which holds a sensor message of type temperature). 

In my sensor pub node I subscribe to the /temperature topic (with subscriber "tempSub" to topic "temperature") and then update a global variable with the most recent data and a flag indicating that there is new data to send. I then convert that data to userdata using the method above and publish that data as a userData (with publisher "tempPub" to topic "temp_signal"). 

Next, I run a sensor sub node that looks for signatures that contain user data and that data is never found. 

> if(!node.sensorData().userDataCompressed().empty())

This statement is never true.

I think the correct topic for user_data_async to be connected to would be "temp_signal", but when I changed that just now I'm still not getting user data to come through. 

Attached is an updated rqt_graph with the modified connections.
[image]

matlabbe, 2019-11-24:
Sorry for the delay, I was trying to find a way to make the wifi example working on my new laptop to test if the code is still working. I've updated the wifi tutorial with a python example for the publisher: http://wiki.ros.org/rtabmap_ros/Tutorials/WifiSignalStrengthMappingUserDataUsage

The tutorial is working on my side. Can you check if you can reproduce the problem with the tutorial?

pfautley, 2019-11-26:
Thanks for getting back to me! I'm not able to run the tutorial unfortunately. I'm trying to integrate several packages together, so I need to use a custom launch file. When trying to run the launch file in the tutorial I am met with a bunch of warnings and repeated indications that the scan is empty. Plus, when I try to run the wifi signal pub I am told that there is an invalid interface and am unable to use sudo with rosrun.

matlabbe, 2019-12-01:
You may use the Python version of the publisher.

Otherwise, I would need a rosbag without rtabmap running, and the launch file for rtabmap that you are using.

pfautley, 2019-12-07:
Thanks so much for the python example! I really appreciate you taking the time to help me through this. I'm sorry for taking so long to get back to you on this I had to make some changes to the robot and needed to take it down for a few days. I'm back up and running now and I just ran a version of the python code (without wifi, but a constant value for dBm) and I saw the same issues. Attached is a copy of my mapping launch file I'm sure this is the problem, because the rtabmap_ros mapping launch file is so extensive that there must be something in there that I need to do that I'm not already doing. 

[myMapping.launch.txt](https://github.com/introlab/rtabmap_ros/files/3934471/myMapping.launch.txt)

I will try to create a rosbag on Monday (The robot lost a charge before I could get to it today). I've never made one, but it seems simple enough. Do you want one with the publisher and subscriber running without the mapping launch file?

matlabbe, 2019-12-13:
Don't use subscribe_user_data if the rate of /temp_signal is not as fast as the camera. The tutorial is using async approach: Try this way:
```
<param name="subscribe_user_data"  type="bool"   value="false"/>
<remap from="user_data_async" to="/temp_signal"/>
```

Look in rqt_graph, rtabmap should be linked to /temp_signal.