introlab/rtabmap_ros #370: Localization and Navigation

jerryt3204, 2019-11-22:
Hi, recently i'm able to create a map and i'm trying to Localization and Navigation.
I did the steps from this video: 
https://www.youtube.com/watch?v=JgP2X-ARj-A&list=PLZPXCuPULoic67j3KWDuAKvORBMPbQp7J&index=23

But i got 2 questions:
The first one is when i click "download map", it only showed up a part of the map. I saw the video above and other videos, they showed up the whole map. This is my screenshot:
[image]
 This is not my whole map, did i miss some parameters ??

The second question is i thought my localization is not working well. Because sometimes my lidar data will drift like this picture(ignore the yellow text):
Sometimes my move_base command failed, i thought maybe because of this problem...
[image]
Is that probably because of TF?

Thanks in advance !!!

matlabbe, 2019-11-22:
Hi jerry,

1) Which parameters did you use? Is `Rtabmap/TimeThr` used? Set it to 0 unless you really want to make use (and understand!) of RTAB-Map's memory management approach.
2) Which parameters did you use? If the robot drifts a lot on rotation, rtabmap may not be able to correct the whole drift depending on the Icp parameters (if you use `Reg/Strategy=1`).

cheers,
Mathieu

jerryt3204, 2019-11-23:
Hi @matlabbe , thanks for quick response.

I've tried the "Rtabmap/TimeThr" to 0, it looks better, but it also looks like not the same to what i saw in database-viewer(3D-Map). It looks like broken to pieces.
And the second question, my robot not only drift rotation, but also displacement.

Here is my launch file to include rtabmap's launch file (part):
<arg name="localization"            value="false"/>
    <arg name="approx_sync"             value="true"/>  

    <arg name="rtabmap_args"            value="--delete_db_on_start"/>
    <arg name="depth_topic"             value="/camera/aligned_depth_to_color/image_raw"/>
    <arg name="rgb_topic"               value="/camera/color/image_raw"/>
    <arg name="camera_info_topic"       value="/camera/color/camera_info"/>

    <arg name="rtabmapviz"              value="true"/>
    <arg name="rviz"                    value="false"/>

    <arg name="subscribe_scan"          value="true"/>
    <arg name="frame_id"                value="base_link"/>      
    <arg name="odom_topic"              value="/odom"/>          

    <arg name="tag_linear_variance"     value="0.0001" />
    <arg name="tag_angular_variance"    value="0.0005" />        

    <arg name="visual_odometry"         value="false"/>
    <arg name="icp_odometry"            value="false"/>
    <arg name="queue_size"              value="20"/>             

And this is my rtabmap's launch file (part):

      <param name="Grid/FromDepth"            type="string" value="false"/>      
      <param name="Optimizer/Strategy"                      value="1"/>          
      <param name="RGBD/NeighborLinkRefining" type="string" value="true"/>
      <param name="RGBD/ProximityBySpace"     type="string" value="true"/>
      <param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/>
      <param name="RGBD/OptimizeMaxError"     type="string" value="0.1"/>
      <param name="Rtabmap/TimeThr"           type="string" value="0"/>          
      <param name="Reg/Strategy"              type="string" value="1"/>          
      <param name="Reg/Force3DoF"             type="string" value="true"/>
      <param name="Grid/NoiseFilteringRadius" type="string" value="0.05"/>
      <param name="Grid/NoiseFilteringMinNeighbors" type="string" value="5"/>
      <param name="Vis/MinInliers"            type="string" value="5"/>
      <param name="Vis/InlierDistance"        type="string" value="0.1"/>
      <param name="Kp/DetectorStrategy"       type="string" value="0"/>          
      <param name="Mem/RehearsalSimilarity"   type="string" value="0.45"/>
      <param name="RGBD/ProximityPathMaxNeighbors" type="string" value="10"/>
      <param name="RGBD/SavedLocalizationIgnored"  type="bool"   value="true"/>

I want to solve the drifting problem, and also improving my map accuracy.
Did i tune something wrong?

Thank you,
Jerry

matlabbe, 2019-11-23:
If you want to re-use a map that was previously built with `Rtabmap/TimeThr > 0` and show all nodes on localization, set `Mem/InitWMWithAllNodes` to true. If the map was created with `Rtabmap/TimeThr=0`,  all nodes should be shown. Can you share the database?

jerryt3204, 2019-11-23:
Yes, of course. Here is my db:
https://mega.nz/#!aCZSnArZ!hvaPH8ufEzkQgSneFfhIBuEU12FMCiuO5qRUwlv5o3E

I built with Rtabmap/TimeThr = 0

matlabbe, 2019-11-23:
```
$ roslaunch rtabmap_ros rtabmap.launch database_path:=~/Downloads/rtabmap.db rviz:=true
$ rosservice call /rtabmap/publish_map  1 1 0
```
[image]

All clouds seem to be there. The only thing is that the clouds have an offset on z-axis, that can be corrected by using the appropriate `frame_id` (e.g. `base_footprint`) for rtabmap node.

`Icp/MaxCorrespondenceDistance` could be increased if the robot drifts between two frames. For best scan matching, I build rtabmap with libpointmatcher support.

`RGBD/OptimizeMaxError` should be  > 1 (default 3, it is a ratio of standard deviation, not absolute distance). Setting it too low would make all loop closures rejected.

cheers,
Mathieu

jerryt3204, 2019-11-26:
Hi matlabbe,

Thanks for your advice first.
I've notice that offset problem, and i already fix it by modifying some TF relationship.
And i also notice that when i'm doing localization and navigation, "cloud max depth" is also important, i think that's one of the reason why i cannot see the whole map.

Everything works well now, next i will try to tune it better.
Thank you.