introlab/rtabmap_ros #371: rtabmap pointcloud for same device looks different on 2 different computers

lingfei93, 2019-11-26:
Hello,

I am using a Kinect DK to map my surroundings. I am using rtabmap_ros to create a 3D map of the surrounding. On one computer, rtabmap_ros seems to work very well and is able to create a decent 3D and 2D map of the surrounding. The normal one looks like this.

3D map:
[image]
2D map:
[image]

Which is all fine, until I move the computing to my laptop, then i get really weird yellow spots in the 3D map that seems like noise, but I'm having issues filtering it. The weird one looks like this:

[image]

There seems to be a lot of yellow cloud noise that seems to be around the origin of the camera. This does not seem to affect the 3D map, which turns out to be fine, but it really affects my 2D map, as seen here
2D Map: 
[image]
I'm really curious as to what the issue is, as I am subscribing to rtabmap/rtabmap/proj_map, I thought I'd post the issue here and check. I've checked the native Kinect DK viewer and the sensor data is fine.

I'd appreciate any help and please do ping me if I have left out some information.

matlabbe, 2019-11-28:
Can you record a rosbag wit kinect data?

lingfei93, 2019-11-29:
[bag_files.zip](https://github.com/introlab/rtabmap_ros/files/3904793/bag_files.zip)
I've uploaded bag files. Only have the scan data, as the files are quite big in general. Hope that's okay.

matlabbe, 2019-12-01:
If the bags are too large, it would be better to share the databases created from rtabmap (default `~/.ros/rtabmap.db`). I cannot do much with only scan_map topic.

lingfei93, 2019-12-02:
Hi, i've uploaded them to a google drive link, but the topics only have around ~5 seconds of data. hope thats okay.

Thank you so much for your time.

https://drive.google.com/open?id=14ISuYr4AJ_pJ26mxeFErqfvJxmHMRtIz

lingfei93, 2019-12-03:
Hi,

Sorry, on closer inspection, I realized that the depth image that I was looking

closed 2019-12-03 by lingfei93