introlab/rtabmap_ros #374: RTABMap Freezes and stops after around 800 frames

lingfei93, 2019-12-17:
Hi, I've been able to consistently reproduce this error, so I thought I'd get some guidance here. I have a Ubuntu 18.04 running ROS Melodic and using a Kinect DK to run rtabmap. All seems to work really well, except that RTABMap seems to hang after around 700-900 frames consistently. It does not seem to be the size of the map as well(i think?), as I am running it just spinning inside a small room. I've attached a picture of the DB as well as the log right before it hangs. I'd appreciate any guidance or request of information. Have attached the database here as well!

https://drive.google.com/open?id=1dLcpuImzE1eZHVqjgMZfxDBEF0dd1a1_

[ INFO] [1576578122.627487829]: rtabmap (661): Rate=1.00s, Limit=0.000s, RTAB-Map=1.1116s, Maps update=0.0157s pub=0.0002s (local map=403, WM=403)
[ WARN] (2019-12-17 18:22:02.666) MainWindow.cpp:1599::processStats() Processing time (1.107876s) is over detection rate (1.000000s), real-time problem!
[ WARN] [1576578123.326835718]: Image processing thread is running behind.
Expected max loop time: 0.033333333
Actual loop time: 0.043315573

[ INFO] [1576578123.648076427]: rtabmap (662): Rate=1.00s, Limit=0.000s, RTAB-Map=1.0030s, Maps update=0.0118s pub=0.0001s (local map=404, WM=404)
[ INFO] [1576578125.207263058]: rtabmap (663): Rate=1.00s, Limit=0.000s, RTAB-Map=1.0427s, Maps update=0.0239s pub=0.0002s (local map=405, WM=405)
[ WARN] (2019-12-17 18:22:05.270) MainWindow.cpp:1599::processStats() Processing time (1.037903s) is over detection rate (1.000000s), real-time problem!
[ INFO] [1576578126.469146129]: rtabmap (664): Rate=1.00s, Limit=0.000s, RTAB-Map=1.2039s, Maps update=0.0288s pub=0.0001s (local map=406, WM=406)
[ WARN] (2019-12-17 18:22:06.532) MainWindow.cpp:1599::processStats() Processing time (1.199211s) is over detection rate (1.000000s), real-time problem!
[ INFO] [1576578127.431312503]: rtabmap (665): Rate=1.00s, Limit=0.000s, RTAB-Map=0.9403s, Maps update=0.0186s pub=0.0000s (local map=407, WM=407)
[ INFO] [1576578128.813812589]: rtabmap (666): Rate=1.00s, Limit=0.000s, RTAB-Map=0.8457s, Maps update=0.0120s pub=0.0000s (local map=408, WM=408)
[ INFO] [1576578130.234248792]: rtabmap (667): Rate=1.00s, Limit=0.000s, RTAB-Map=0.9732s, Maps update=0.0244s pub=0.0000s (local map=409, WM=409)
[ INFO] [1576578130.455274674]: rtabmap (668): Rate=1.00s, Limit=0.000s, RTAB-Map=0.1595s, Maps update=0.0282s pub=0.0000s (local map=409, WM=409)
[ INFO] [1576578132.134502768]: rtabmap (669): Rate=1.00s, Limit=0.000s, RTAB-Map=0.7928s, Maps update=0.0131s pub=0.0000s (local map=410, WM=410)
[ WARN] [1576578133.334479435]: Image processing thread is running behind.
Expected max loop time: 0.033333333
Actual loop time: 0.035075459

[ INFO] [1576578133.738766429]: rtabmap (670): Rate=1.00s, Limit=0.000s, RTAB-Map=1.1558s, Maps update=0.0210s pub=0.0000s (local map=411, WM=411)
[ WARN] (2019-12-17 18:22:13.790) MainWindow.cpp:1599::processStats() Processing time (1.145777s) is over detection rate (1.000000s), real-time problem!
[ INFO] [1576578135.036985675]: rtabmap (671): Rate=1.00s, Limit=0.000s, RTAB-Map=1.2431s, Maps update=0.0220s pub=0.0000s (local map=412, WM=412)
[ WARN] (2019-12-17 18:22:15.116) MainWindow.cpp:1599::processStats() Processing time (1.238242s) is over detection rate (1.000000s), real-time problem!
[ INFO] [1576578136.156480160]: rtabmap (672): Rate=1.00s, Limit=0.000s, RTAB-Map=1.1011s, Maps update=0.0159s pub=0.0000s (local map=413, WM=413)
[ WARN] (2019-12-17 18:22:16.191) MainWindow.cpp:1599::processStats() Processing time (1.097243s) is over detection rate (1.000000s), real-time problem!
[ INFO] [1576578137.717829841]: rtabmap (673): Rate=1.00s, Limit=0.000s, RTAB-Map=1.1000s, Maps update=0.0231s pub=0.0000s (local map=414, WM=414)
[ WARN] (2019-12-17 18:22:17.753) MainWindow.cpp:1599::processStats() Processing time (1.088005s) is over detection rate (1.000000s), real-time problem!
[ INFO] [1576578138.749904925]: rtabmap (674): Rate=1.00s, Limit=0.000s, RTAB-Map=1.0014s, Maps update=0.0113s pub=0.0000s (local map=415, WM=415)
[ INFO] [1576578140.160577271]: rtabmap (675): Rate=1.00s, Limit=0.000s, RTAB-Map=0.9082s, Maps update=0.0185s pub=0.0001s (local map=416, WM=416)
[ INFO] [1576578141.708167461]: rtabmap (676): Rate=1.00s, Limit=0.000s, RTAB-Map=1.0869s, Maps update=0.0127s pub=0.0000s (local map=417, WM=417)
[ WARN] (2019-12-17 18:22:21.774) MainWindow.cpp:1599::processStats() Processing time (1.078426s) is over detection rate (1.000000s), real-time problem!
[ INFO] [1576578142.760352722]: rtabmap (677): Rate=1.00s, Limit=0.000s, RTAB-Map=1.0046s, Maps update=0.0148s pub=0.0001s (local map=418, WM=418)
[ WARN] (2019-12-17 18:22:22.800) MainWindow.cpp:1599::processStats() Processing time (1.001007s) is over detection rate (1.000000s), real-time problem!
[ INFO] [1576578142.949274530]: rtabmap (678): Rate=1.00s, Limit=0.000s, RTAB-Map=0.1682s, Maps update=0.0174s pub=0.0001s (local map=418, WM=418)
[ WARN] [1576578143.597958029]: Image processing thread is running behind.
Expected max loop time: 0.033333333
Actual loop time: 0.034395645

[ WARN] (2019-12-17 18:22:24.781) Rtabmap.cpp:2919::process() Rejecting all added loop closures (4, first is 679 <-> 568) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 3.047221 (edge 328->505, type=1, abs error=0.099012 m, stddev=0.032492). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2019-12-17 18:22:24.782) Rtabmap.cpp:2953::process() Loop closure 679->568 rejected!
[ WARN] (2019-12-17 18:22:24.782) Rtabmap.cpp:2953::process() Loop closure 679->574 rejected!
[ WARN] (2019-12-17 18:22:24.782) Rtabmap.cpp:2953::process() Loop closure 679->332 rejected!
[ WARN] (2019-12-17 18:22:24.782) Rtabmap.cpp:2953::process() Loop closure 679->508 rejected!
[ INFO] [1576578144.814511976]: rtabmap (679): Rate=1.00s, Limit=0.000s, RTAB-Map=1.0166s, Maps update=0.0235s pub=0.0000s (local map=419, WM=419)
[ WARN] (2019-12-17 18:22:24.931) MainWindow.cpp:1599::processStats() Processing time (1.010674s) is over detection rate (1.000000s), real-time problem!
[ WARN] (2019-12-17 18:22:25.940) Rtabmap.cpp:2919::process() Rejecting all added loop closures (4, first is 680 <-> 626) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 3.006039 (edge 328->505, type=1, abs error=0.097674 m, stddev=0.032492). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2019-12-17 18:22:25.940) Rtabmap.cpp:2953::process() Loop closure 680->626 rejected!
[ WARN] (2019-12-17 18:22:25.940) Rtabmap.cpp:2953::process() Loop closure 680->574 rejected!
[ WARN] (2019-12-17 18:22:25.940) Rtabmap.cpp:2953::process() Loop closure 680->458 rejected!
[ WARN] (2019-12-17 18:22:25.940) Rtabmap.cpp:2953::process() Loop closure 680->334 rejected!
[ INFO] [1576578145.972196989]: rtabmap (680): Rate=1.00s, Limit=0.000s, RTAB-Map=1.1076s, Maps update=0.0228s pub=0.0002s (local map=420, WM=420)
[ WARN] (2019-12-17 18:22:26.023) MainWindow.cpp:1599::processStats() Processing time (1.102000s) is over detection rate (1.000000s), real-time problem!
[ WARN] (2019-12-17 18:22:27.221) Rtabmap.cpp:2919::process() Rejecting all added loop closures (3, first is 681 <-> 328) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 3.001647 (edge 328->505, type=1, abs error=0.097531 m, stddev=0.032492). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2019-12-17 18:22:27.221) Rtabmap.cpp:2953::process() Loop closure 681->328 rejected!
[ WARN] (2019-12-17 18:22:27.221) Rtabmap.cpp:2953::process() Loop closure 681->574 rejected!
[ WARN] (2019-12-17 18:22:27.221) Rtabmap.cpp:2953::process() Loop closure 681->458 rejected!
[ INFO] [1576578147.240802303]: rtabmap (681): Rate=1.00s, Limit=0.000s, RTAB-Map=0.9279s, Maps update=0.0104s pub=0.0000s (local map=421, WM=421)
[ INFO] [1576578148.697954567]: rtabmap (682): Rate=1.00s, Limit=0.000s, RTAB-Map=1.0030s, Maps update=0.0210s pub=0.0001s (local map=422, WM=422)
[ INFO] [1576578148.949681899]: rtabmap (683): Rate=1.00s, Limit=0.000s, RTAB-Map=0.1812s, Maps update=0.0234s pub=0.0000s (local map=422, WM=422)
[... cut ...]
[ WARN] (2019-12-17 18:24:38.186) Rtabmap.cpp:2953::process() Loop closure 737->681 rejected!
[ WARN] (2019-12-17 18:24:38.186) Rtabmap.cpp:2953::process() Loop closure 737->684 rejected!
[ WARN] (2019-12-17 18:24:38.186) Rtabmap.cpp:2953::process() Loop closure 737->457 rejected!
[ WARN] (2019-12-17 18:24:38.186) Rtabmap.cpp:2953::process() Loop closure 737->566 rejected!
[ INFO] [1576578278.205983646]: rtabmap (737): Rate=1.00s, Limit=0.000s, RTAB-Map=0.7840s, Maps update=0.0132s pub=0.0002s (local map=470, WM=470)

matlabbe, 2019-12-19:
rtabmap adds a new node in the graph each time the robot turns or moves enough. Rotating indefinitely on-place will grow the map linearly. Looking at the processing time (rtabmap-databaseViewer->Statistics view):
[image]
A lot of time is required to do proximity detections. Proximity detection requires more an more time when we traverse the same area again and again (here turning on-place has the same effect). You could try to disable `RGBD/ProximityBySpace` to save some processing time.

The freeze problem is unclear with the log you've shown. Is it because the camera stopped to publish images?

About KinectDK, you could try to use RGB image with depth registered to RGB camera instead of using depth image with RGB registered to depth camera.

lingfei93, 2019-12-23:
Hi,
Thank you for your comprehensive reply. The freeze problem is unclear with me too, as there are no error logs, and RTABMap seems to be killed off. I would love to see what other sources of information I can seek when this symptom happens. 

I have checked that the camera is still publishing at a normal rate when the freeze happens, so i can only assume that the RTABMap node seems to have died?

Thanks for your suggestion about the KinectDK! Could I ask why using RGB to depth registered might be better? As I thought the RGB images will have a higher resolution. Just curious, that's all.

Thanks so much for looking at my issue!

matlabbe, 2020-01-07:
Feature extraction is done on RGB image, having black borders could create false features. The depth image is used by default as mask, so in general features won't be extracted on those fake borders, but it depends on the OpenCV version and the feature detector used. When the 3D point cloud is created for visualization, some parts may be black because some depth pixels may not have any corresponding RGB pixels (then assuming black pixels). Also if eventually you want to create texture mesh of the environments (like this [one](https://skfb.ly/6OyUy)), it is better to have original RGB images for projection, otherwise a lot of black will be projected on the mesh.

cheers,
Mathieu

lingfei93, 2020-01-22:
Hi Mathieu,

Thank you for your help. I am using the rgb image and the quality of the Map has improved a lot. I still have a question though. After the map reaches a certain size, or time, RTABMap seems to stop publishing new node/map updates. What I mean is that RTABMap seems to stop publishing updates. I have disabled RGBD/ProximityBySpace, as well as setting a limit  to Rtabmap/TimeThr, but RTABMap node seems to stop publishing updates after the map grows sufficiently big. Does this mean computational power on my system is limited? But I'm confused because the memory management of TimeThr should manage this for me. Would appreciate some guidance if i'm going in the right direction here. Thank you so much!

matlabbe, 2020-01-31:
Do you have warnings on the terminal? If it was a processing problem, you would have timing spikes of 10 seconds in the plot. It looks like a synchronization problem, maybe image and odometry timestamps drift apart over time, so rtabmap node has difficulty to synchronize the input topics. What is the frame rate of the input topics (image and odometry)? (rostopic hz /your_rgb_image_topic /your_depth_image_topic /your_camera_info /your_odometry)

On the 10 seconds updates, were the robot moving?

Can you show the rtabmap part of the launch file you are using?

lingfei93, 2020-02-05:
Hi Mathieu,

Thanks for your reply. I would think that it seems like it is a timing issue. When i switched over to doing RTABMap's visual odometry, the rtabmap system was able to work fine. I will see what I can do to synchronize the image and odometry timestamp together.

I have attached the frame rate here: [sample_rate.txt](https://github.com/introlab/rtabmap_ros/files/4158787/sample_rate.txt)

Over the 10 seconds, the robot was moving!

I attach the rtabmap of the launch file is here. [rtabmap.txt](https://github.com/introlab/rtabmap_ros/files/4158820/rtabmap.txt)

Thanks again for taking time to look into my issue. I really appreciate it.