introlab/rtabmap_ros #376: The generated map is of poor quality and cannot be used for navigation

diyuhuodong, 2019-12-24:
Hello,
     I use REALSENSE T265 as an odometer, and D435 generates a point cloud map, but the map produced is of poor quality and cannot be used for navigation. I don't know where the problem is. My data link is [here](https://drive.google.com/file/d/11JxGNn4JIwALWn--UltVMv3sKV_eSPWx/view?usp=sharing), help analyze and see if there is a problem with the configuration.

diyuhuodong, 2019-12-24:
Here a screenshot of my data:
[image]
My map is spliced everywhere.

diyuhuodong, 2019-12-24:
I got the following TF tree:
[image]

diyuhuodong, 2019-12-24:
My nodes and topics look like this：
[image]

diyuhuodong, 2019-12-24:
And here my launchfile:
<launch>
    
    <arg name="device_type_camera1"    		default="t265"/>
    <arg name="device_type_camera2"    		default="d4.5"/>	
    <arg name="serial_no_camera1"    			default="908412111092"/>
    <arg name="serial_no_camera2"    			default="845112071317"/>
    <arg name="camera1"              			default="t265"/>		
    <arg name="camera2"              			default="d400"/>		
    <arg name="clip_distance"             default="-2"/>
    <arg name="use_rviz"                  default="false"/>
    <arg name="use_rtabmapviz"            default="false"/>
    

    <include file="$(find realsense2_camera)/launch/rs_d400_and_t265.launch">
            <arg name="device_type_camera1"             value="$(arg device_type_camera1)"/>
            <arg name="device_type_camera2"             value="$(arg device_type_camera2)"/>
            <arg name="serial_no_camera1"               value="$(arg serial_no_camera1)"/>
            <arg name="serial_no_camera2"               value="$(arg serial_no_camera2)"/>
            <arg name="camera1"                         value="$(arg camera1)"/>
            <arg name="camera2"                         value="$(arg camera2)"/>
            <arg name="clip_distance"                   value="$(arg clip_distance)"/>
            
    </include>

    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
            <arg name="depth_topic"        value="/$(arg camera2)/aligned_depth_to_color/image_raw"/>
            <arg name="frame_id"           value="$(arg camera2)_link"/>
            <arg name="visual_odometry"    value="false"/>
            <arg name="odom_topic"         value="/$(arg camera1)/odom/sample"/>
            <arg name="rgb_topic"          value="/$(arg camera2)/color/image_raw"/>
            <arg name="camera_info_topic"  value="/$(arg camera2)/color/camera_info"/>
            <arg name="queue_size"         value="200"/>
            <arg name="rviz"               value="$(arg use_rviz)"/>
            <arg name="rtabmapviz"         value="$(arg use_rtabmapviz)"/>
   </include>
</launch>

My rtabmap.launch file  
Part of my rtabmap.launch file
    
    
    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="$(arg output)" args="$(arg rtabmap_args)" launch-prefix="$(arg launch_prefix)">
      <param     if="$(arg stereo)" name="subscribe_depth"  type="bool" value="false"/>
      <param unless="$(arg stereo)" name="subscribe_depth"  type="bool" value="true"/>
      <param name="subscribe_rgbd"       type="bool"   value="$(arg subscribe_rgbd)"/>
      <param name="subscribe_stereo"     type="bool"   value="$(arg stereo)"/>
      <param name="subscribe_scan"       type="bool"   value="$(arg subscribe_scan)"/>
      <param name="subscribe_scan_cloud" type="bool"   value="$(arg subscribe_scan_cloud)"/>
      <param name="subscribe_user_data"  type="bool"   value="$(arg subscribe_user_data)"/>
      <param if="$(arg visual_odometry)" name="subscribe_odom_info" type="bool" value="true"/>
      <param if="$(arg icp_odometry)"    name="subscribe_odom_info" type="bool" value="true"/>
      <param name="frame_id"             type="string" value="$(arg frame_id)"/>
      <param name="map_frame_id"         type="string" value="$(arg map_frame_id)"/>
      <param name="odom_frame_id"        type="string" value="$(arg odom_frame_id)"/>
      <param name="publish_tf"           type="bool"   value="$(arg publish_tf_map)"/>
      <param name="ground_truth_frame_id"       type="string" value="$(arg ground_truth_frame_id)"/>
      <param name="ground_truth_base_frame_id"  type="string" value="$(arg ground_truth_base_frame_id)"/>
      <param name="odom_tf_angular_variance" type="double" value="$(arg odom_tf_angular_variance)"/>
      <param name="odom_tf_linear_variance"  type="double" value="$(arg odom_tf_linear_variance)"/>
      <param name="odom_sensor_sync"         type="bool"   value="$(arg odom_sensor_sync)"/>
      <param name="wait_for_transform_duration"  type="double"   value="$(arg wait_for_transform)"/>
      <param name="database_path"        type="string" value="$(arg database_path)"/>
      <param name="approx_sync"          type="bool"   value="$(arg approx_sync)"/>
      <param name="config_path"          type="string" value="$(arg cfg)"/>
      <param name="queue_size"           type="int" value="$(arg queue_size)"/>
      <param name="scan_normal_k"        type="int" value="$(arg scan_normal_k)"/>
      <param name="landmark_linear_variance"   type="double" value="$(arg tag_linear_variance)"/>
      <param name="landmark_angular_variance"  type="double" value="$(arg tag_angular_variance)"/>      
      <param name="Grid/MaxGroundAngle"     type="string"     value="60"/>
[... 9 lines cut ...]
      <remap from="rgb/image"       to="$(arg rgb_topic_relay)"/>
      <remap from="depth/image"     to="$(arg depth_topic_relay)"/>
      <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
      
      <remap from="rgbd_image"      to="$(arg rgbd_topic_relay)"/>
	
      <remap from="left/image_rect"        to="$(arg left_image_topic_relay)"/>
      <remap from="right/image_rect"       to="$(arg right_image_topic_relay)"/>
      <remap from="left/camera_info"       to="$(arg left_camera_info_topic)"/>
      <remap from="right/camera_info"      to="$(arg right_camera_info_topic)"/>
      
      <remap from="scan"                   to="$(arg scan_topic)"/>
      <remap from="scan_cloud"             to="$(arg scan_cloud_topic)"/>
[... cut ...]
      <remap from="tag_detections"         to="$(arg tag_topic)"/>
      <remap from="odom"                   to="$(arg odom_topic)"/>
      <remap from="imu"                    to="$(arg imu_topic)"/>
      
      
      <param     if="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="false"/>
      <param unless="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="true"/>
      <param name="Mem/InitWMWithAllNodes" type="string" value="$(arg localization)"/>
    </node>

matlabbe, 2020-01-07:
Ok this is the realsense launch file from there: https://github.com/IntelRealSense/realsense-ros/blob/432eb1c19809a637622604bd8f0f3438c04cc17e/realsense2_camera/launch/rs_rtabmap.launch

I guess you would have to update this [transform](https://github.com/IntelRealSense/realsense-ros/blob/432eb1c19809a637622604bd8f0f3438c04cc17e/realsense2_camera/launch/rs_d400_and_t265.launch#L48): 
`<node pkg="tf" type="static_transform_publisher" name="t265_to_d400" args="0 0 0 0 0 0 /$(arg tf_prefix_camera1)_link /$(arg tf_prefix_camera2)_link 100"/>`
because I assume that the cameras cannot be perfectly superposed. Otherwise, from the original realsense launch file, they use standard rtabmap launch file, what modifications you did on rtabmap launch file?

However, if the transform between the cameras is correctly updated and you still have similar problems, maybe it is because T265 is drifting for some reason.

I see in the database at some points you seem dragging the camera/robot in front of the tracking camera. This can cause large drift if the tracking camera is obstructed while moving. I see also that the RGB camera has a lot of motion blur at some point, try not moving too fast, at least not rotating too fast for better tracking.

cheers,
Mathieu