introlab/rtabmap_ros #377: RTABMAP ros is crashing suddenly in the middle of playing bagfile

UditSinghParihar, 2019-12-24:
Hello,
I have been using RTABMAP for a long time and it is quite a good package. Previously I have collected some bag files with wheel odometry. And used to ran RTABMAP with visual odometry or wheel odometry, and it used to work.
Last few days whenever I am trying to run RTABMAP again with those bagfiles, rtabmap always crashes at some particular point in the bagfile, with the below error message.
I tried my setup with RTABMAP `0.19.3` and `0.19.6`, both build from source (clean build by deleting everything) and it giving the same error in both setups on two different bag files (and error is coming exactly at the same point in bagfile).
I am on Ubuntu 16.04 (ROS Kinetic), 64 bit architecture.

`[rtabmap/rtabmap-2] process has died [pid 17937, exit code -11, cmd /home/cair/catkin_ws2/devel/lib/r
tabmap_ros/rtabmap --delete_db_on_start rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/
depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info rgbd_image:=rgbd_image_relay left
/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect l
eft/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info 
scan:=/scan scan_cloud:=/scan_cloud user_data:=/user_data user_data_async:=/user_data_async gps/fix:=
/gps/fix tag_detections:=/tag_detections odom:=odom imu:=/imu/data __name:=rtabmap __log:=/home/cair/
.ros/log/c8bd584c-2695-11ea-bcfe-507b9d90c688/rtabmap-rtabmap-2.log].
log file: /home/cair/.ros/log/c8bd584c-2695-11ea-bcfe-507b9d90c688/rtabmap-rtabmap-2*.log.`

The video of crashing can be seen [here](https://drive.google.com/open?id=15ZDUp-jWfvhIQjTIMcaI464pcepH4aGv) (at 25 seconds mark and it always happen at 25 secs mark only).

After following the suggestion given [here](https://github.com/introlab/rtabmap_ros/issues/28) to debug, below is the output of from gdb backtracing: 

[image]

UditSinghParihar, 2019-12-30:
Ok, I figured out the solution. The problem was with pcl which I earlier built from source, that was being used by rtabmap instead of the one which is mentioned in the preinstallation step (from apt repositories). So I uninstalled the local built pcl (`sudo make uninstall` in the build directory of local pcl), and did:    
`sudo apt-get install ros-melodic-rtabmap ros-melodic-rtabmap-ros`     
`sudo apt-get remove ros-melodic-rtabmap ros-melodic-rtabmap-ros`     
. Now building `rtabmap` and `rtabmap_ros` uses apt installed pcl and bagfile is working without segmentation fault.

closed 2019-12-30 by UditSinghParihar

matlabbe, 2020-01-06:
Great that you've found a solution. Here some more details explaining the problem:

In the gdb log, we can see it is crashing in Eigen functions. With the video, we can also see that it is crashing on loop closure, where a graph optimization is done. Graph optimization (g2o ot GTSAM) is using Eigen. The problem when building pcl from source is that they now force compilation flag "`-march=native`", which affects how memory is managed in Eigen. When building rtabmap, it will also use that flag. Now, if g2o and GTSAM are not rebuilt with that flag, it will likely crash like that. 

In your case, using PCL binaries solved the problem, because they were not built with that compilation flag, so everything else is compatible out of the box. The other solution, if you would want to keep latest PCL version built from source, is to rebuild g2o and GTSAM (is used) with "`-march-native`" compilation flag.

cheers,
Mathieu