introlab/rtabmap_ros #378: How to remotely launch rtabmapviz separately？

willzoe, 2020-01-06:
I have a robot controlled by TX2 and want to run rtabmap on TX2 to build a map.

But there is no display on the robot, so I want to run the rtabmap algorithm on TX2, and then use a laptop to remotely display the rtabmap GUI, just like remotely starting rviz.

Is there any way?

matlabbe, 2020-01-06:
Hi, 
If you are able to use RVIZ remotely, starting rtabmapviz would work in similar way. We only have to make sure to start `rtabmapviz` on the remote computer in same namespace than `rtabmap` node. If `rtabmap` node is started in `rtabmap` namespace:

```bash
$ export ROS_NAMESPACE=rtabmap
$ rosrun rtabmap_ros rtabmapviz
```
or in one line:
```bash
$ export ROS_NAMESPACE=rtabmap && rosrun rtabmap_ros rtabmapviz
```
also possible (but services between `rtabmapviz` and `rtabmap` won't work):
```bash
$ rosrun rtabmap_ros rtabmapviz mapData:=/rtabmap/mapData
```

cheers,
Mathieu

closed 2020-01-08 by willzoe

knnurl, 2024-06-16:
great explanation! thanks.