introlab/rtabmap_ros #379: When I use a ZED stereo camera, what is the difference between using RGB-D Odometry and?

willzoe, 2020-01-08:
I have a zed stereo camera, and I want to put it on turtlebot for outdoor environment mapping and navigation.

`zed_wrapper` provides an example, it uses odometry generated by sdk, and passes the rgb image (left sensor) and depth image generated by sdk to `rtabmap` node. I modified this example to use the `RGB-D Odometry` of rtabmap instead of the odometry generated by the zed sdk.

The tutorial [wiki.ros.org/rtabmap_ros/Tutorials/StereoOutdoorMapping](wiki.ros.org/rtabmap_ros/Tutorials/StereoOutdoorMapping) uses the Bumblebee2 stereo camera, which uses `Stereo Odometry`, and passes the left and right sensor images to the `rtabmap` node.

I want to know what is the difference between the results obtained using these methods?

matlabbe, 2020-01-08:
If you want to use visual odometry from rtabmap outdoor with the zed, use stereo odometry. Farther features can be used to estimate odometry using stereo mode, which will give better odometry. Zed's depth is often limited in range, so very far features are ignored in RGB-D odometry. Maybe there are options on zed side to estimate depth farther than by default.

cheers,
Mathieu

willzoe, 2020-01-09:
The default maximum range of the depth map generated by zed is 15 meters, and the maximum range can be reached by modifying parameters.
Is the difference between stereo odometry and RGB-D odometry only in the estimated scene depth?

If I use stereo mode, what parameters can I modify to make stereo odometry better than RGB-D odometry?

matlabbe, 2020-01-09:
> Is the difference between stereo odometry and RGB-D odometry only in the estimated scene depth?
Yes

> If I use stereo mode, what parameters can I modify to make stereo odometry better than RGB-D odometry?
It will be automatically better. The default parameters were tuned with KITTI and EuRoC stereo datasets.

15 meters is too short for outdoor. In stereo mode, some features can be extracted over 1-10 km (with relatively bad depth estimation, but at least they are super useful for orientation estimation). With local bundle adjustment done in odometry, the 3d pose of those features is refined with new frames.

cheers,
Mathieu

closed 2020-01-12 by willzoe