introlab/rtabmap_ros #383: pointcloud orientation issue

redvinaa, 2020-01-24:
Hi,
I'm trying to simulate a diff drive robot with a realsense D435i in gazebo. The mapping works, but the pointcloud is displayed pointing upwards. 
There is no point in moving the camera link, because all the cameras are pointing forward already. The imu doesn't affect the pointcloud, because if I remove the imu sensor the pointcloud is the same.
[image]
[image]
[image]
Rtabmap doesn't show any relevant error messages.
I don't know what files to include that might help resolve the problem, if someone asks me to I will upload the given file.

matlabbe, 2020-02-05:
Hi,
What is strange to me is that `base_link` is not the parent frame of the camera. Normally, you would have `/map -> odom -> base_link -> camera_base ....`

The `frame_id` you set for `rgbd_odometry` and `rtabmap` should be `base_link`, assuming you fix the urdf to have `base_link` the parent of all frames on the robot.

Try fixing your TF tree to see if it helps, otherwise it would be that the frame set in image topics is wrong, or an optical rotation is needed between `camera_base` and `camera_link`. What is the frame set in camera topics?

threeal, 2021-04-19:
Hi i also have the same issue. I just curious what may cause the point cloud transformation to change? Does it is transformed according to the published `tf` topic? or `robot_description`? or else?

matlabbe, 2021-05-04:
The wrong frame_id may be set in the topic.

yaqub93, 2023-07-15:
I also had this problem. It solved by adding dummy frame with this transformation:
```
  <joint name="camera_optical_joint" type="fixed">
    <origin xyz="0 0 0" rpy="-1.5707 0 -1.5707"/>
    <parent link="camera_frame"/>
    <child link="camera_optical_frame"/>
  </joint>
  <link name="camera_optical_frame"/>
```

closed 2024-07-26 by redvinaa

peakyquest, 2024-12-15:
I am having similar issue. I am working on sdf model and gazebo ignition, doesnt know how to solve it ?

matlabbe, 2024-12-16:
Based on https://github.com/introlab/rtabmap_ros/issues/383#issuecomment-1636592584, make sure the simulated camera (gazebo sensor description) is using the optical frame `camera_optical_frame`, not the `camera_frame` frame.

For ignition, the only doc I've found is here: https://gazebosim.org/docs/latest/migrating_gazebo_classic_ros2_packages/#libgazebo-ros-camera-so, the parameter is `gz_frame_id`:
```xml
<sensor name="camera" type="camera">
  <always_on>true</always_on>
  <visualize>true</visualize>
  <update_rate>30</update_rate>
  <topic>camera/image_raw</topic>
  <gz_frame_id>camera_rgb_frame</gz_frame_id>
  <camera name="intel_realsense_r200">
    <camera_info_topic>camera/camera_info</camera_info_topic>
    ... <!-- all the content of <camera> from the original -->
  </camera>
</sensor>
```

peakyquest, 2024-12-16:
[model.txt](https://github.com/user-attachments/files/18153817/model.txt)
[rtabmap_sync_launch.py.txt](https://github.com/user-attachments/files/18153837/rtabmap_sync_launch.py.txt)

I am using the following model.sdf file which is similar to the one you have shown in your comment. 
and also I have added the launch file. can you figure it out what I am doing wrong.

matlabbe, 2024-12-21:
In the launch fife, you should use `base_link` for `frame_id`. In the sdf, you need to add a new link for the camera optical frame. I didn't test it but something like this maybe:
```xml
   <link name="base_link">
      <pose> 0 0 17.425E-3 0 0 0</pose>
      <inertial>
        <mass>0.025</mass>
        <inertia>
          <ixx>16.571710E-06</ixx>
          <iyy>16.655602E-06</iyy>
          <izz>29.261652E-06</izz>
          <ixy>0.830806E-06</ixy>
          <ixz>0.718277E-06</ixz>
          <iyz>1.800197E-06</iyz>
        </inertia>
      </inertial>
      <collision name="base_link_collision">
        <geometry>
          <box>
            <size>0.10 0.10 34.850E-3</size>
          </box>
        </geometry>
      </collision>
      <visual name="base_link_visual">
        <geometry>
          <mesh>
            <uri>../../../meshes/collada_files/cf2_assembly.dae</uri>
          </mesh>
        </geometry>
      </visual>
    </link>

   <link name="camera_rgb_frame"/>
   <joint name="camera_optical_joint" type="fixed">
        <parent link="base_link"/>
        <child link=camera_rgb_frame"/>
        <origin xyz="0 0 0" rpy="${-pi/2} 0 ${-pi/2}"/>
        <axis xyz="0 0 1" />
    </joint>

    <gazebo reference="camera_rgb_frame">
      <sensor name="camera" type="rgbd_camera">
        <pose relative_to="crazyflie/base_footprint">0 0 0.1 0 0 0</pose>
        <always_on>true</always_on>
        <visualize>true</visualize>
        <update_rate>30</update_rate>
        <camera_info_topic>/camera/rgbd/camera_info</camera_info_topic>
        <topic>/camera/rgbd/image_raw</topic>
        <ignition_frame_id>camera_rgb_frame</ignition_frame_id>
        <camera name="intel_realsense_r200">
          <horizontal_fov>1.047</horizontal_fov>
          <image>
            <width>320</width>
            <height>240</height>
          </image>
          <clip>
            <near>0.1</near>
            <far>100</far>
          </clip>
        </camera>
        <plugin
          filename="RosGzPointCloud"
          name="ros_gz_point_cloud::PointCloud">
          <namespace>custom_params</namespace>
          <topic>pc2</topic>
          <frame_id>camera_rgb_frame</frame_id>
        </plugin>
      </sensor>
    </gazebo>

```