introlab/rtabmap_ros #385: Issue Compiling on ROS On Windows

uchuuengineer, 2020-02-21:
I am running ROS Melodic on Windows. I managed to build RTabMap from source with CMake GUI, however running catkin_make for the RTAB-Map ros package fails building at 56%. 

Please let me know if it is possible to solve this issue. Thank you.

```
OUTPUT:
c:\ws\rtabmap_ros>catkin_make
Base path: c:\ws\rtabmap_ros
Source space: c:\ws\rtabmap_ros\src
Build space: c:\ws\rtabmap_ros\build
Devel space: c:\ws\rtabmap_ros\devel
Install space: c:\ws\rtabmap_ros\install
####
#### Running command: "nmake cmake_check_build_system" in "c:\ws\rtabmap_ros\build"
####

Microsoft (R) Program Maintenance Utility Version 14.24.28316.0
Copyright (C) Microsoft Corporation.  All rights reserved.

CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Using CATKIN_DEVEL_PREFIX: C:/ws/rtabmap_ros/devel
-- Using CMAKE_PREFIX_PATH: C:/opt/ros/melodic/x64;C:/opt/rosdeps/x64;C:/opt/vcpkg/installed/x64-windows
-- This workspace overlays: C:/opt/ros/melodic/x64
-- Found PythonInterp: C:/opt/python27amd64/python.exe (found suitable version "2.7.15", minimum required is "2")
-- Using PYTHON_EXECUTABLE: C:/opt/python27amd64/python.exe
-- Using default Python package layout
-- Using empy: C:/opt/python27amd64/lib/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: C:/ws/rtabmap_ros/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/opt/python27amd64/Scripts/nosetests-2.7.exe
-- catkin 0.7.18
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - rtabmap_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'rtabmap_ros'
-- ==> add_subdirectory(src/rtabmap_ros)
CMake Deprecation Warning at src/rtabmap_ros/CMakeLists.txt:7 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at src/rtabmap_ros/CMakeLists.txt:12 (cmake_policy):
  The OLD behavior for policy CMP0042 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
[... 362 lines cut ...]
C:\PROGRA~1\RTABMA~1.6-W\include\RTABMA~1.19\rtabmap/core/Parameters.h(647): warning C4305: 'return': truncation from 'double' to 'float'
C:\PROGRA~1\RTABMA~1.6-W\include\RTABMA~1.19\rtabmap/core/Parameters.h(699): warning C4305: 'return': truncation from 'double' to 'float'
C:\PROGRA~1\RTABMA~1.6-W\include\RTABMA~1.19\rtabmap/core/Parameters.h(708): warning C4305: 'return': truncation from 'double' to 'float'
C:\PROGRA~1\RTABMA~1.6-W\include\RTABMA~1.19\rtabmap/core/Parameters.h(723): warning C4305: 'return': truncation from 'double' to 'float'
C:\PROGRA~1\RTABMA~1.6-W\include\RTABMA~1.19\rtabmap/core/Parameters.h(729): warning C4305: 'return': truncation from 'double' to 'float'
C:\PROGRA~1\RTABMA~1.6-W\include\RTABMA~1.19\rtabmap/core/Parameters.h(730): warning C4305: 'return': truncation from 'double' to 'float'
C:\PROGRA~1\RTABMA~1.6-W\include\RTABMA~1.19\rtabmap/core/Parameters.h(731): warning C4305: 'return': truncation from 'double' to 'float'
C:\PROGRA~1\RTABMA~1.6-W\include\RTABMA~1.19\rtabmap/core/Parameters.h(732): warning C4305: 'return': truncation from 'double' to 'float'
C:\PROGRA~1\RTABMA~1.6-W\include\RTABMA~1.19\rtabmap/core/Parameters.h(736): warning C4305: 'return': truncation from 'double' to 'float'
C:\PROGRA~1\RTABMA~1.6-W\include\RTABMA~1.19\rtabmap/core/Parameters.h(737): warning C4305: 'return': truncation from 'double' to 'float'
C:\PROGRA~1\RTABMA~1.6-W\include\RTABMA~1.19\rtabmap/core/Parameters.h(738): warning C4305: 'return': truncation from 'double' to 'float'
C:\opt\rosdeps\x64\include\boost-1_66\boost/iostreams/positioning.hpp(96): warning C4996: 'std::fpos<_Mbstatet>::seekpos': warning STL4019: The member std::fpos::seekpos() is non-Standard, and is preserved only for compatibility with workarounds for old versions of Visual C++. It will be removed in a future release, and in this release always returns 0. Please use standards-conforming mechanisms to manipulate fpos, such as conversions to and from streamoff, or an integral type, instead. If you are receiving this message while compiling Boost.IOStreams, a fix has been submitted upstream to make Boost use standards-conforming mechanisms, as it does for other compilers. You can define _SILENCE_FPOS_SEEKPOS_DEPRECATION_WARNING to acknowledge that you have received this warning, or define _REMOVE_FPOS_SEEKPOS to remove std::fpos::seekpos entirely.
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\include\iosfwd(74): note: see declaration of 'std::fpos<_Mbstatet>::seekpos'
C:\ws\rtabmap_ros\src\src\rtabmap_ros\src\OdometryROS.cpp(275): error C2131: expression did not evaluate to a constant
C:\ws\rtabmap_ros\src\src\rtabmap_ros\src\OdometryROS.cpp(275): note: failure was caused by call of undefined function or one not declared 'constexpr'
C:\ws\rtabmap_ros\src\src\rtabmap_ros\src\OdometryROS.cpp(275): note: see usage of 'std::vector<std::string,std::allocator<std::string>>::size'
C:\ws\rtabmap_ros\src\src\rtabmap_ros\src\OdometryROS.cpp(278): error C3863: array type 'char *['function']' is not assignable
NMAKE : fatal error U1077: 'C:\PROGRA~2\MIB055~1\2019\COMMUN~1\VC\Tools\MSVC\1424~1.283\bin\Hostx64\x64\cl.exe' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
Invoking "nmake" failed
```

matlabbe, 2020-02-21:
It seems the compiler doesn't like:
https://github.com/introlab/rtabmap_ros/blob/f54e060787be3982e86e5df599f42316c501f713/src/OdometryROS.cpp#L277-L278

getMyArg() is from nodelet API: http://docs.ros.org/jade/api/nodelet/html/classnodelet_1_1Nodelet.html#a1dffeed8c34b7a7b631023435ab6f4f5, and would return a vector of strings. Not sure how to fix that.

For the char* array, we could make a new array if it doesn't like a non-constant size:
```cpp
char ** argv = new char*[argList.size()];
// delete after usage...
delete [] argv;
```

uchuuengineer, 2020-02-25:
Thank you! Replacing line 278 with ` char ** argv = new char*[argList.size()];` helped me get farther in the building process!

uchuuengineer, 2020-03-01:
Not sure if a follow-up question is proper etiquette, but does anyone suspect a reason why this code compiles in most cases, but required a syntax change for mine? Is it possible it is due to difference in versions of CMake or Python?

I am asking because I have been trying to solve other build errors. Thank you again.

matlabbe, 2020-03-10:
Well, windows compatibility has been done for rtabmap standalone version. I've never tried rtabmap_ros on Windows, so there could be some incompatibility issues with the compiler used, cmake version used and/or c++ standard used.

closed 2020-03-11 by uchuuengineer

MikeHallettUK, 2021-04-08:
Hi ( to uchuuengineer ) ...

Did you ever get **rtabmap_ros** to compile and work on **Windows 10** ?

Thanks

uchuuengineer, 2021-04-08:
Hi MikeHallettUK,

I reframed the question and opened it as another issue last year and matlabbe came to the rescue. This is the answer that helped me: https://github.com/introlab/rtabmap_ros/issues/391#issuecomment-598014764