introlab/rtabmap_ros #390: Fail build rtabmap_ros melodic ubuntu 18.04

AlexandreBruliau, 2020-03-05:
Hello , 
Im actually facing an issue when i tried to build the pakage rtabmap_ros .
Indeed , I get this error but I dont know where it might comes from .
[image]

I also attached what cmake returns : 
[image]
Can someone help me please ?

matlabbe, 2020-03-05:
Make sure to have latest [rtabmap](https://github.com/introlab/rtabmap) library version installed. Uninstall binary version to make sure `rtabmap_ros` is not including headers from two different rtabmap versions during compilation (the error above maybe because it included `Rtabmap.h` from binary version instead of the one built from source):
```bash
$ sudo apt remove ros-melodic-rtabmap
```
cheers,
Mathieu

AlexandreBruliau, 2020-03-06:
Thanks you very much but I encoutered an other problem.
I am simulating the robot in gazebo <with a Kinect camera and a laser from the package depth_image to laser-scan and i would like to get an accurate 2d occupancy grid.
The algorithm is contructing well the map when the robot goes in the linear direction.
However , the map is getting awful when I give an angular velocity the robot.
I first thought to change the parameter landmark_angular_variance but it doesn't solve actually the problem.
[image]
[image]

matlabbe, 2020-03-10:
The rotation error you are seeing is from odometry. Look at the code you are using that is providing odometry, maybe some optimizations have to be done there. Can you provide your launch file (copy paste into a code area, not a screenshot)?

Can you add a simulated lidar instead of using fake scan from depth image, you will get a lot better results as you could enable icp refinements like in this [setup](http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot#Kinect_.2B-_Odometry_.2B-_2D_laser). Otherwise, look at this [setup](http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot#Kinect_.2B-_Odometry_.2B-_Fake_2D_laser_from_Kinect).

cheers,
Mathieu

AlexandreBruliau, 2020-03-10:
Hello , as i will have to test it on a real camera on a robot , I cant use a simple lidar. Even with the setup that you advice me to use , the map get disorted each time the robot turns on himself.
[image]

About the camera Im using in gazebo is the 'libgazebo_ros_openni_kinect' that is defined like this in gazebo : 
`<gazebo reference="camera">
  <sensor name="camera" type="depth">
    <update_rate>100</update_rate>
    <camera>
      <horizontal_fov>2</horizontal_fov>
      <image>
        <width>640</width>
        <height>480</height>
        <format>R8G8B8</format>
      </image>
      <clip>
        <near>0.05</near>
        <far>30</far>
      </clip>
    </camera>
    <plugin name="camera_controller" filename="libgazebo_ros_openni_kinect.so">
      <baseline>0.2</baseline>
      <alwaysOn>true</alwaysOn>
      <updateRate>100</updateRate>
      <cameraName>camera</cameraName>
      <imageTopicName>/camera/color/image_raw</imageTopicName>
      <cameraInfoTopicName>/camera/color/camera_info</cameraInfoTopicName>
      <depthImageTopicName>/camera/depth/image_raw</depthImageTopicName>
      <depthImageInfoTopicName>/camera/depth/camera_info</depthImageInfoTopicName>
      <pointCloudTopicName>/camera/depth/points</pointCloudTopicName>
      <frameName>camera</frameName>
      <pointCloudCutoff>0.1</pointCloudCutoff>
      <pointCloudCutoffMax>30</pointCloudCutoffMax>
      <distortionK1>0.00000001</distortionK1>
      <distortionK2>0.00000001</distortionK2>
      <distortionK3>0.00000001</distortionK3>
      <distortionT1>0.00000001</distortionT1>
      <distortionT2>0.00000001</distortionT2>
      <CxPrime>0</CxPrime>
      <Cx>0</Cx>
      <Cy>0</Cy>
      <focalLength>0</focalLength>
      <hackBaseline>0</hackBaseline>
    </plugin>
  </sensor>
</gazebo>`

here is the launch file I was using and the node i use for odometry is not the problem because it follows accurrately the real trajectory.

`
<launch>
  

  
           
  
     
        
  <arg name="stereo"                    default="false"/>
  <arg     if="$(arg stereo)" name="depth"  default="false"/>
  <arg unless="$(arg stereo)" name="depth"  default="true"/>
 
  
  <arg name="rtabmapviz"              default="false" /> 
  <arg name="rviz"                    default="true" />
  
  
  <arg name="localization"            default="false"/>
  
  
  <arg name="use_sim_time"            default="false"/>
[... 318 lines cut ...]
    </node>
  
  </group>
  
  
  <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(arg rviz_cfg)"/>
  <node if="$(arg rviz)" pkg="nodelet" type="nodelet" name="points_xyzrgb" args="standalone rtabmap_ros/point_cloud_xyzrgb" output="$(arg output)">
    <remap if="$(arg stereo)" from="left/image"        to="$(arg left_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="right/image"       to="$(arg right_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="left/camera_info"  to="$(arg left_camera_info_topic)"/>
    <remap if="$(arg stereo)" from="right/camera_info" to="$(arg right_camera_info_topic)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/image"         to="$(arg rgb_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="depth/image"       to="$(arg depth_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/camera_info"   to="$(arg camera_info_topic)"/>
    <remap from="rgbd_image"        to="$(arg rgbd_topic_relay)"/>
    <remap from="cloud"             to="voxel_cloud" />

    <param name="decimation"  type="double" value="4"/>
    <param name="voxel_size"  type="double" value="0.0"/>
    <param name="approx_sync" type="bool"   value="$(arg approx_sync)"/>
  </node>

</launch>`

Thanks you for you help

matlabbe, 2020-04-01:
Hi,
 [rtabmap.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/rtabmap.launch) is not meant to be edited manually, but used in another launch file like this:
```xml
<launch>
<include file="$(find rtabmap_ros)/launch/rtabmap.launch">
  <arg name="rtabmapviz"              value="false" /> 
  <arg name="rviz"                    value="true" />
  <arg name="rgb_topic"               value="/camera/color/image_raw" />
  <arg name="depth_topic"             value="/camera/depth/image_raw" />
  <arg name="camera_info_topic"       value="/camera/color/camera_info" />
  <arg name="visual_odometry"         value="false"/>       
  <arg name="odom_topic"              value="/segway/odom"/>
  <arg name="frame_id"                value="base_link"/>
  <arg name="odom_frame_id"           value="odom"/>
  <arg name="subscribe_scan"          value="true"/>
  <arg name="scan_topic"              value="/laser/scan"/>
  <arg name="appox_sync"               value="false"/>
</include>
</launch>
```
So not sure if I can spot all differences you made. The error may be in tf. If you are creating a map, you can share the resulting database (`~/.ros/rtabmap.db`), or record a rosbag:
```
$ rosbag record /tf /tf_static /camera/color/image_raw /camera/depth/image_raw /camera/color/camera_info /laser/scan
```