introlab/rtabmap_ros #391: Library limit of 65535 objects exceeded - Building RTABMap ROS on Windows

uchuuengineer, 2020-03-11:
I am trying to build RTABMap ROS  for ROS Melodic on Windows, but a library limit error seems to be the cause of my problems. I think I managed to make this error go away temporarily by adding /bigobj flags in CmakeLists.txt, but then I just run into a problem when I'm 89% done with compilation that says  "don't know how to make 'C:\ws\rtabmap_ros\devel\lib\rtabmap_sync.lib'." I suspect it is because I was improperly dealing with this library limit error and after trying for about two weeks, I can't seem to fix it. I linked the original library limit error I got below. I appreciate any help you can give me.

```
c:\ws\rtabmap_ros>catkin_make -DVTK_DIR=C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build -DCMAKE_BUILD_TYPE=RELEASE
Base path: c:\ws\rtabmap_ros
Source space: c:\ws\rtabmap_ros\src
Build space: c:\ws\rtabmap_ros\build
Devel space: c:\ws\rtabmap_ros\devel
Install space: c:\ws\rtabmap_ros\install
####
#### Running command: "cmake c:\ws\rtabmap_ros\src -DVTK_DIR=C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build -DCMAKE_BUILD_TYPE=RELEASE -DCATKIN_DEVEL_PREFIX=c:\ws\rtabmap_ros\devel -DCMAKE_INSTALL_PREFIX=c:\ws\rtabmap_ros\install -G NMake Makefiles" in "c:\ws\rtabmap_ros\build"
####
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is MSVC 19.24.28319.0
-- The CXX compiler identification is MSVC 19.24.28319.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.24.28314/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.24.28314/bin/Hostx64/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.24.28314/bin/Hostx64/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.24.28314/bin/Hostx64/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: C:/ws/rtabmap_ros/devel
-- Using CMAKE_PREFIX_PATH: C:/opt/ros/melodic/x64;C:/opt/rosdeps/x64;C:/opt/vcpkg/installed/x64-windows
-- This workspace overlays: C:/opt/ros/melodic/x64
-- Found PythonInterp: C:/opt/python27amd64/python.exe (found suitable version "2.7.15", minimum required is "2")
-- Using PYTHON_EXECUTABLE: C:/opt/python27amd64/python.exe
-- Using default Python package layout
-- Found PY_em: C:\opt\python27amd64\lib\site-packages\em.pyc
-- Using empy: C:/opt/python27amd64/lib/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: C:/ws/rtabmap_ros/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/opt/python27amd64/Scripts/nosetests-2.7.exe
-- catkin 0.7.21
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - rtabmap_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'rtabmap_ros'
-- ==> add_subdirectory(rtabmap_ros)
CMake Deprecation Warning at rtabmap_ros/CMakeLists.txt:7 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

[... 969 lines cut ...]
        with
        [
            MReq=nav_msgs::GetMap::Request,
            MRes=nav_msgs::GetMap::Response,
            T=rtabmap_ros::CoreWrapper
        ]
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(54): note: see reference to class template instantiation 'boost::arg<9>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(53): note: see reference to class template instantiation 'boost::arg<8>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(52): note: see reference to class template instantiation 'boost::arg<7>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(51): note: see reference to class template instantiation 'boost::arg<6>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(50): note: see reference to class template instantiation 'boost::arg<5>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(49): note: see reference to class template instantiation 'boost::arg<4>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(48): note: see reference to class template instantiation 'boost::arg<3>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(47): note: see reference to class template instantiation 'boost::arg<2>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(46): note: see reference to class template instantiation 'boost::arg<1>' being compiled
[ 68%] Linking CXX shared library C:\ws\rtabmap_ros\devel\bin\rtabmap_sync.dll
LINK: command "C:\PROGRA~2\MIB055~1\2019\COMMUN~1\VC\Tools\MSVC\1424~1.283\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\rtabmap_sync.dir\objects1.rsp /out:C:\ws\rtabmap_ros\devel\bin\rtabmap_sync.dll /implib:C:\ws\rtabmap_ros\devel\lib\rtabmap_sync.lib /pdb:C:\ws\rtabmap_ros\devel\bin\rtabmap_sync.pdb /dll /version:0.0 /machine:x64 /INCREMENTAL:NO /DEF:CMakeFiles\rtabmap_sync.dir\exports.def C:\ws\rtabmap_ros\devel\lib\rtabmap_ros.lib C:\opt\rosdeps\x64\lib\opencv_dnn341.lib C:\opt\rosdeps\x64\lib\opencv_ml341.lib C:\opt\rosdeps\x64\lib\opencv_objdetect341.lib C:\opt\rosdeps\x64\lib\opencv_shape341.lib C:\opt\rosdeps\x64\lib\opencv_stitching341.lib C:\opt\rosdeps\x64\lib\opencv_superres341.lib C:\opt\rosdeps\x64\lib\opencv_videostab341.lib C:\opt\rosdeps\x64\lib\opencv_calib3d341.lib C:\opt\rosdeps\x64\lib\opencv_features2d341.lib C:\opt\rosdeps\x64\lib\opencv_flann341.lib C:\opt\rosdeps\x64\lib\opencv_highgui341.lib C:\opt\rosdeps\x64\lib\opencv_photo341.lib C:\opt\rosdeps\x64\lib\opencv_video341.lib C:\opt\rosdeps\x64\lib\opencv_videoio341.lib C:\opt\rosdeps\x64\lib\opencv_imgcodecs341.lib C:\opt\rosdeps\x64\lib\opencv_imgproc341.lib C:\opt\rosdeps\x64\lib\opencv_core341.lib C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build\lib\Release\vtkDomainsChemistryOpenGL2-8.2.lib C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build\lib\Release\vtkDomainsChemistry-8.2.lib C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build\lib\Release\vtkFiltersFlowPaths-8.2.lib C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build\lib\Release\vtkFiltersGeneric-8.2.lib C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build\lib\Release\vtkFiltersHyperTree-8.2.lib C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build\lib\Release\vtkFiltersParallelImaging-8.2.lib C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build\lib\Release\vtkFiltersPoints-8.2.lib C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build\lib\Release\vtkFiltersProgrammable-8.2.lib C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build\lib\Release\vtkFiltersSMP-8.2.lib C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build\lib\Release\vtkFiltersSelection-8.2.lib C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build\lib\Release\vtkFiltersTexture-8.2.lib C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build\lib\Release\vtkFiltersTopology-8.2.lib C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build\lib\Release\vtkFiltersVerdict-8.2.lib C:\Users\cdobrin1\Documents\github\VTK-8.2.0\VTK-8.2.0\build\lib\Release\vtkverdict-8.2.lib C:\Use [... cut ...]
LINK : fatal error LNK1189: library limit of 65535 objects exceeded
NMAKE : fatal error U1077: 'C:\opt\python27amd64\Lib\site-packages\cmake\data\bin\cmake.exe' : return code '0xffffffff'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
Invoking "nmake" failed```

matlabbe, 2020-03-12:
Hi,

I've taken some time to try rtabmap_ros on Windows. I tested to far without VTK, so without UI. For this example, this [commit](https://github.com/introlab/rtabmap/commit/5371f94c8f673e3bab1d2ea3373c4cab047606b2) (rtabmap) and this [one](https://github.com/introlab/rtabmap_ros/commit/aec4fe6bcc95b6e0f96c4b7488f4839a74bb25ad) (rtabmap_ros) are required.

1. Install ROS melodic on Windows following this tutorial: http://wiki.ros.org/Installation/Windows
2. From inside a ROS configured command window (see tutorial above), git clone [rtabmap](https://github.com/introlab/rtabmap), then build without GUI support (there is a missing VTK issue when pcl_visualization module is trying to find it):
    ```bash
    $ cd rtabmap/build
    $ cmake -DWITH_QT=OFF -DCMAKE_INSTALL_PREFIX=C:\opt\rosdeps\x64 ..
    $ cmake --build . --config Release --target install
    ``` 
    I install rtabmap in default rosdep binaries path for convenience.
3. git clone [rtabmap_ros](https://github.com/introlab/rtabmap_ros) in your catkin workspace src folder.
4. catkin_make

Open Issues:
 * I see that you've built rtabmap with VTK support, is the standalone RTABMap.exe generated by the rtabmap library (not rtabmap_ros) working? I tried VTK6.3 but the Qt5 version installed with ROS is missing some libraries. I may try VTK8 as you, but I want to know if RTABMap.exe is not crashing at start. Not sure to which VTK version the pcl_visualization library is linked on. We should use the same one, otherwise rtabmap GUI would crash (and rtabmapviz on ROS).

 * The `Library limit of 65535 objects exceeded` is fixed by the second commit I referred. As you did, we should add `/bigobj` compilation flag, and for `rtabmap_sync`, I needed to reduce the number of functions compiled with the two new build options (both default to false on Windows). 

 * [demo_mapping.bag](http://wiki.ros.org/rtabmap_ros#Robot_mapping) is not working on Windows, because images in the bag are `compressedDepth` format. It seems that the ROS windows binaries don't contain the `compressedDepth` plugin, so `image_transport` is failing to do the conversion. I would need to convert the bag on Ubuntu to save raw images instead.

 * I didn't test with a camera (like Kinect), as there are no prebuilt binaries for cameras in ROS for windows. We would have to build from source openni2_launch, openni_launch or freenect_launch to try a Kinect camera for example. Not sure about realsense package on ROS if they work on Windows.

uchuuengineer, 2020-03-12:
Hi matlabbe, thank you so much for all your help. This is the first time I managed to build rtabmap_ros, but not without issues. I followed your installation instructions and rtabmap_ros finally built to 100%. I also modified line 81 in codec.cpp in the image_transport_plugins package as you recommended and rebuilt. However, running the first roslaunch command launched RViz with four error messages (OpenNI2.dll, g2o_core.dll, g2o_types_slam2d.dll, g2o_types_slam3d.dll not found).

 I am not very good with C++ editing and building code, but I hope I will be able to answer your questions somewhat. Most of the rtabmap_ros builds I attempted to used an RTABMap binary for its dependencies that I obtained from [here](https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts). If my memory is correct, I think I managed to built RTABMap from source once (and failed all the other times), but I had the same error as when I used the binary for dependencies, so I deleted the version I built from source. The already built RTABMap.exe was able to detect my ZED stereo camera, however it would crash upon calibration. I would get generate(?) errors saying VTKConfig.cmake could not be located so I manually tried to build with two versions of VTK on separate build attempts: the newest VTK version (8.2) on VTK's official site as well as the 6.3 version under the Installation for Windows section [here](https://github.com/introlab/rtabmap/wiki/Installation#windows).

If I understand correctly from your explanation, I should build OpenNI2 from source (and I guess also g2o) while pointing to it either when building RTABMap or RTABMap_ROS. I will try that and see how it works.

matlabbe, 2020-04-01:
I installed rtabmap binaries in `C:\opt\rosdeps\x64`, which is the default path for ros to find the dlls. If rtabmap is installed anywhere else, we need to add to PATH the bin directory, or copy dlls like you did (in a directory contained in the PATH).

The rtabmap standalone binaries may not be built with same PCL and opencv versions used in ROS. This may cause crashes for `rtabmap_ros` when mixing up libraries with different versions. Well, at least you managed to run the demo example. I suggest to continue using RVIZ for visualization under ROS, to avoid all building issues with VTK. The resulting database could still be opened with RTAB-Map standalone binaries afterwards.

cheers,
Mathieu

closed 2021-01-23 by uchuuengineer