introlab/rtabmap_ros #393: [melodic][Kinect v4a]Did not receive data since 5 seconds!

colorDogs, 2020-03-23:
Device: kinect v4 (v4a)
Command:  roslaunch azure_kinect_ros_driver slam_rtabmap.launch 

launch file : [launch](https://raw.githubusercontent.com/microsoft/Azure_Kinect_ROS_Driver/melodic/launch/slam_rtabmap.launch)

the launch file about rtabmap:
` 
  <include file="$(find rtabmap_ros)/launch/rtabmap.launch" ns="rtabmap" >
    <arg name="rgb_topic"          value="/k4a/rgb_to_depth/image_rect" />
    <arg name="depth_topic"        value="/k4a/depth/image_rect" />
    <arg name="camera_info_topic"  value="/k4a/depth/camera_info" />
    <arg name="approx_sync"        value="false" />
    <arg name="frame_id"           value="camera_base" />
    <arg name="args"               value="--delete_db_on_start --Odom/ResetCountdown 2" />
  </include>`

source code:
- [kinect v4  ros wrapper](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/launch/slam_rtabmap.launch)     branch[melodic]
- rtabmap      branch[melodic-devel]
- rtabmap_ros     branch[melodic-devel]

 roslaunch azure_kinect_ros_driver slam_rtabmap.launch 
... logging to /home/lamer/.ros/log/7f2a2fbc-6d07-11ea-acfa-50eb7167c20d/roslaunch-lamer-Mi-Gaming-Laptop-15-6-6633.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lamer-Mi-Gaming-Laptop-15-6:40323/

SUMMARY
========

PARAMETERS
 * /k4a/k4a_ros_bridge/color_enabled: True
 * /k4a/k4a_ros_bridge/color_resolution: 1536P
 * /k4a/k4a_ros_bridge/depth_enabled: True
 * /k4a/k4a_ros_bridge/depth_mode: NFOV_UNBINNED
 * /k4a/k4a_ros_bridge/fps: 30
 * /k4a/k4a_ros_bridge/point_cloud: False
 * /k4a/k4a_ros_bridge/required: True
 * /k4a/k4a_ros_bridge/rgb_point_cloud: False
 * /k4a/manager/num_worker_threads: 16
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.4
 * /rtabmap/rtabmap/rgbd_odometry/approx_sync: False
 * /rtabmap/rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rtabmap/rgbd_odometry/frame_id: camera_base
 * /rtabmap/rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rtabmap/rgbd_odometry/publish_tf: True
 * /rtabmap/rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rtabmap/rgbd_odometry/subscribe_rgbd: False
 * /rtabmap/rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmap/rgbd_odometry/wait_imu_to_init: False
 * /rtabmap/rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/rtabmap/approx_sync: False
 * /rtabmap/rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/rtabmap/frame_id: camera_base
 * /rtabmap/rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/rtabmap/map_frame_id: map
[... 313 lines cut ...]
/rtabmap/rtabmap/rtabmap subscribed to (exact sync):
   /rtabmap/rtabmap/odom,
   /k4a/rgb_to_depth/image_rect,
   /k4a/depth/image_rect,
   /k4a/depth/camera_info,
   /rtabmap/rtabmap/odom_info
[ WARN] [1584968968.219677953]: /rtabmap/rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap/rtabmapviz subscribed to (exact sync):
   /rtabmap/rtabmap/odom,
   /k4a/rgb_to_depth/image_rect,
   /k4a/depth/image_rect,
   /k4a/depth/camera_info,
   /rtabmap/rtabmap/odom_info
[ WARN] [1584968972.343166219]: /rtabmap/rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap/rgbd_odometry subscribed to (exact sync):
   /k4a/rgb_to_depth/image_rect,
   /k4a/depth/image_rect,
   /k4a/depth/camera_info
[ WARN] [1584968972.661113566]: /rtabmap/rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap/rtabmap subscribed to (exact sync):
   /rtabmap/rtabmap/odom,
   /k4a/rgb_to_depth/image_rect,
   /k4a/depth/image_rect,
   /k4a/depth/camera_info,
   /rtabmap/rtabmap/odom_info

sibeichen, 2020-03-24:
I have the same problem ,this is my configuration    

  
  <include file="$(find rtabmap_ros)/launch/rtabmap.launch" ns="rtabmap" >
    <arg name="rgb_topic"          value="/k4a/rgb_to_depth/image_rect" />
    <arg name="depth_topic"        value="/k4a/depth/image_rect" />
    <arg name="camera_info_topic"  value="/k4a/depth/camera_info" />
    <arg name="approx_sync"        value="false" />
    <arg name="frame_id"           value="camera_base" />
    <arg name="args"               value="--delete_db_on_start --Odom/ResetCountdown 2" />
  </include>

sibeichen, 2020-03-24:
`  
  <include file="$(find rtabmap_ros)/launch/rtabmap.launch" ns="rtabmap" >
    <arg name="rgb_topic"          value="/k4a/rgb_to_depth/image_rect" />
    <arg name="depth_topic"        value="/k4a/depth/image_rect" />
    <arg name="camera_info_topic"  value="/k4a/depth/camera_info" />
    <arg name="approx_sync"        value="false" />
    <arg name="frame_id"           value="camera_base" />
    <arg name="args"               value="--delete_db_on_start --Odom/ResetCountdown 2" />
  </include>`

matlabbe, 2020-04-04:
The root of the problem is coming from `rgbd_odmetry` node that is not computing odometry:
```
[ WARN] [1584968962.342825111]: /rtabmap/rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap/rgbd_odometry subscribed to (exact sync):
   /k4a/rgb_to_depth/image_rect,
   /k4a/depth/image_rect,
   /k4a/depth/camera_info
```
As those 3 topics seem published as shown in your rostopic hz, the last thing to check is if these topics have exactly the same stamp. For example, if you set `approx_sync` to true, does it work? However, it is better that `approx_sync` is false if we can, to make sure everything is perfectly synchronized.

colorDogs, 2020-04-04:
Thank you very much for your reply.
I tried using 'approx_sync = true', it works , but the pose is always lost....Can't create map successfully.
Are there any successful examples of using k4a?

colorDogs, 2020-04-04:
> The root of the problem is coming from `rgbd_odmetry` node that is not computing odometry:
>
> ```
> [ WARN] [1584968962.342825111]: /rtabmap/rtabmap/rgbd_odometry: Did not receive data since 5 seconds! [...]

Thank you very much for your reply.
I tried using 'approx_sync = true', it works , but the pose is always lost....Can't create map successfully.
Are there any successful examples of using k4a?

matlabbe, 2020-04-06:
The default parameters of `rgbd_odometry` assumes VGA resolution of images. If you are using 720p or 1080p, it will likely cause problems. Either use lower resolution images or change features extracted to `ORB_OCTREE` type, which will scale better with high resolution images, add `--Vis/FeatureType 10 --Kp/DetectorStrategy 10` to `args` for [rtabmap.launch](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/f953f9196e07fd7660aa29411c9e69445513eaef/launch/slam_rtabmap.launch#L60) inlude.

In the [example ](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/launch/slam_rtabmap.launch) you are using, I would prefer to use the depth image registered to color camera instead of the color image registered to depth camera. I don't know if the registered depth to color camera is available, if so you could give a try. Guessing which topics that could be:
```xml
<arg name="rgb_topic"          value="/k4a/rgb/image_rect" />
<arg name="depth_topic"        value="/k4a/depth_to_color/image_rect" />
<arg name="camera_info_topic"  value="/k4a/rgb/camera_info" />
```

I don't own yet a K4A, but if you can record a small rosbag of the topics and /tf it could be useful to see if there is another problem.

cheers,
Mathieu

colorDogs, 2020-04-07:
> The default parameters of `rgbd_odometry` assumes VGA resolution of images. If you are using 720p or 1080p, it will likely cause problems. Either use lower resolution images or change features extracted to `ORB_OCTREE` type, which will scale better with high resolution images, add `--Vis/FeatureType 10 --Kp/DetectorStrategy 10` to `args` for [rtabmap.launch](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/f953f9196e07fd7660aa29411c9e69445513eaef/launch/slam_rtabmap.launch#L60) inlude.
> 
> In the [example ](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/launch/slam_rtabmap.launch) you are using, I would prefer to use the depth image registered to color camera instead of the color image registered to depth camera. I don't know if the registered depth to color camera is available, if so you could give a try. Guessing which topics that could be:
> 
> ```
> <arg name="rgb_topic"          value="/k4a/rgb/image_rect" />
> <arg name="depth_topic"        value="/k4a/depth_to_color/image_rect" />
> <arg name="camera_info_topic"  value="/k4a/rgb/camera_info" />
> ```
> 
> I don't own yet a K4A, but if you can record a small rosbag of the topics and /tf it could be useful to see if there is another problem.
> 
> cheers,
> Mathieu

I used "--Vis / FeatureType 10 --Kp / DetectorStrategy 10" to args , looks the same as using 'approx_sync = true'.
I think there is a problem with "depth_to_color".
I recorded rosbag for 5 seconds
[rosbag - pls click here](https://anya711.com/statics/rtabmap_k4a_bag.zip)

Thanks

colorDogs, 2020-04-17:
Is there any progress on this?

matlabbe, 2020-04-18:
Hi,

I got some time to look at this. Thx for the bag. Not sure if it is a problem from the bag recording, but there are only 132 images for 10 seconds (~13 Hz). When the camera is rotating faster, those missing gaps don't help odometry and is likely to get lost The image resolution of K4A is 2048x1536. Because of this high resolution, default parameters tuned for 640x480 will work poorly. It works better if we decimate the images. For convenience, I updated the code in the commit above, to be able to do that easily with rtabmap.launch. Here is an example:
```bash
$ roslaunch rtabmap_ros rtabmap.launch args:="-d" \
   depth_topic:=/k4a/depth_to_rgb/image_raw \
   rgb_topic:=/k4a/rgb/image_raw \
   camera_info_topic:=/k4a/rgb/camera_info \
   use_sim_time:=true \
   frame_id:=camera_base \
   rgbd_sync:=true \
   rgbd_decimation:=4 \
   approx_sync:=false

$ rosbag play --clock 2020-04-07-17-34-31.bag
```
Note that as explained in this above [post](https://github.com/introlab/rtabmap_ros/issues/393#issuecomment-609915074), we can also try to add `--Vis/FeatureType 10 --Kp/DetectorStrategy 10` to `args` argument. Also with this bag, even with those parameters, odometry get lost at the end because the camera is looking right into window where there are almost to visual features to extract.

cheers,
Mathieu

colorDogs, 2020-04-21:
It's working!!!  It's a little slow, but it's finally working. ( ~ 10 Hz)
words are powerless to express my gratitude.
Thank you

closed 2020-04-21 by colorDogs