introlab/rtabmap_ros #394: Error when running robot mapping demo & d435 with global mapping

dwbzxc, 2020-03-24:
Hi.I got 2 problems when running rtabmap.

**- Error when running robot mapping demo**

When I try running"`demo_robot_mapping.launch`" with "`demo_mapping.bag`" as ros wiki said,nothing shows up in rtabmaprviz interface,just as the picture shows.
[image]
The terminal displays the following message:

> [ WARN] [1585014354.389112881, 1368730038.388771955]: /rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
> /rtabmap/rtabmapviz subscribed to (approx sync):
>    /az3/base_controller/odom,
>    /data_throttled_image/compressed,
>    /data_throttled_image_depth/compressedDepth,
>    /data_throttled_camera_info,
>    /jn0/base_scan

I've tried to `rostopic echo` all topics above and they are all sending messages.So,I wonder how I can fix this.

**- No point cloud accumulation when using realsense D435 to run `rtabmap.launch` as ros wiki said.**

I've never changed the launch file` rs_camera.launch` and `rtabmap.launch`,but when running them as guidance,there is only local point cloud.If I move the camera further,the former point cloud disappears and is not storaged.The terminal displays the following message:

> [rtabmap/rtabmap-2] process has died [pid 21634, exit code -11, cmd /home/dzx/rtabmap_ros/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start rgb/image:=/camera/color/image_raw depth/image:=/camera/aligned_depth_to_color/image_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image_relay left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud user_data:=/user_data user_data_async:=/user_data_async gps/fix:=/gps/fix tag_detections:=/tag_detections odom:=odom imu:=/imu/data __name:=rtabmap __log:=/home/dzx/.ros/log/f2cd60ae-6d77-11ea-9921-28d244204ace/rtabmap-rtabmap-2.log].
> log file: /home/dzx/.ros/log/f2cd60ae-6d77-11ea-9921-28d244204ace/rtabmap-rtabmap-2*.log

These problems have confused for a long time.
Thank you!

matlabbe, 2020-04-04:
Which rtabmap version are you using? Can you show exactly which commands you used for the robot mapping demo?

dwbzxc, 2020-04-04:
I used the Kinetic version and I ran these two commands as the wiki said:
$ roslaunch rtabmap_ros demo_robot_mapping.launch
$ rosbag play --clock demo_mapping.bag

But I just reinstalled my Ubuntu system and rtabmap，and it works right now.So I don't have to solve this problem anymore.Thanks for you replying!

closed 2020-04-06 by matlabbe