introlab/rtabmap_ros #396: Change value of Cloud max depth (rtabmap_ros)

borcherd, 2020-03-25:
I am working on a project where I do a SLAM with rtabmap in an outdoor environment simulated in Gazebo, and navigate autonomously in that SLAM map. For the navigation I need an octomap. I know that rtabmap publishes a octomap to rtabmap/octomap_full but it looks like the range of the sensor isn't big enough to map almost anything but the street. I've tried changing the param Grid/RangeMax but that changes nothing. I have a vlp16 lidar and a realsense depth camera on my robot, tried with both and still no difference.

[image]
Here you can see how far the depth camera sees, and how far the octomap goes.

matlabbe, 2020-04-02:
`Grid/RangeMax` should have done the trick. Can you share a database?

borcherd, 2020-04-22:
> `Grid/RangeMax` should have done the trick. Can you share a database?

Yes, in combination with changing the depth decimation it worked.

closed 2020-04-22 by borcherd

iportolesrb, 2021-09-07:
> > `Grid/RangeMax` should have done the trick. Can you share a database?
> 
> Yes, in combination with changing the depth decimation it worked.

What did you do with the decimation to make it work? I have the same issue.

matlabbe, 2021-09-09:
Increasing `Grid/DepthDecimation` would generate less cells, if it is a range problem, increase `Grid/RangeMax`.

iportolesrb, 2021-09-10:
> Increasing `Grid/DepthDecimation` would generate less cells, if it is a range problem, increase `Grid/RangeMax`.

I tried to increase `Grid/RangeMax` but nothing changes.

matlabbe, 2021-09-13:
Which topic are you looking at exactly?

iportolesrb, 2021-09-14:
The mapData topic.

matlabbe, 2021-09-14:
For mapData, if you visualize with MapCloud plugin in rviz, there is an [Cloud max depth (m)](http://wiki.ros.org/rtabmap_ros#Map_Cloud_Display) option set to 4 meters by default. In rtabmapviz, that option is under Preferences->3D Rendering panel, under Map column. rtabmap-export has also a max depth of 4 meters by default, it can be changed like this (to disable it):
```
rtabmap-export --max_range 0 rtabmap.db
```

iportolesrb, 2021-09-15:
Oh, ok, that solved it thank you.