introlab/rtabmap_ros #397: rtabmap nodes crash immediately in ROS 2 Eloquent w/ "Failed to find library 'rtabmap_ros__rosidl_typesupport_introspection_cpp'"

pjreed, 2020-03-28:
OS: Ubuntu Linux 18.04
Distro: ROS Eloquent

I know ROS 2 support is still experimental, but I feel like the nodes should at least start without crashing.  After building the `rtabmap_ros` package and trying to launch one of the default launch files with `ros2 launch rtabmap_ros turtlebot3_scan.launch.py`, both nodes throw an exception and die like so:
```bash
$ ros2 launch rtabmap_ros turtlebot3_scan.launch.py 
[INFO] [launch]: All log files can be found below /home/preed/.ros/log/2020-03-27-23-30-45-844130-rover-nuc-26382
[INFO] [launch]: Default logging verbosity is set to INFO
1585369845.847427 [0]       ros2: using network interface eno1 (udp/192.168.2.100) selected arbitrarily from: eno1, docker0
[INFO] [rtabmap-1]: process started with pid [26401]
[INFO] [rtabmapviz-2]: process started with pid [26402]
[rtabmap-1] 1585369846.089611 [0]    rtabmap: using network interface eno1 (udp/192.168.2.100) selected arbitrarily from: eno1, docker0
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_depth = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_rgb = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_stereo = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_rgbd = false (rgbd_cameras=1)
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_odom_info = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_user_data = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_scan = true
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_scan_cloud = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap: queue_size    = 10
[rtabmap-1] [INFO] [rtabmap]: rtabmap: approx_sync   = true
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): map_filter_radius          = 0.000000
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): map_filter_angle           = 30.000000
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): map_cleanup                = true
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): map_always_update          = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): map_empty_ray_tracing      = true
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): cloud_output_voxelized     = true
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): cloud_subtract_filtering   = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[rtabmap-1] [INFO] [rtabmap]: rtabmap: frame_id      = base_footprint
[rtabmap-1] [INFO] [rtabmap]: rtabmap: map_frame_id  = map
[rtabmap-1] [INFO] [rtabmap]: rtabmap: use_action_for_goal  = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap: tf_delay      = 0.050000
[rtabmap-1] [INFO] [rtabmap]: rtabmap: tf_tolerance  = 0.100000
[rtabmap-1] [INFO] [rtabmap]: rtabmap: odom_sensor_sync   = false
[rtabmap-1] Failed to find library 'rtabmap_ros__rosidl_typesupport_introspection_cpp'
[rtabmap-1] 
[rtabmap-1] >>> [rcutils|error_handling.c:107] rcutils_set_error_state()
[rtabmap-1] This error state is being overwritten:
[rtabmap-1] 
[rtabmap-1]   'type support not from this implementation, at /tmp/binarydeb/ros-eloquent-rmw-cyclonedds-cpp-0.4.4/src/rmw_node.cpp:991'
[rtabmap-1] 
[rtabmap-1] with this new error message:
[rtabmap-1] 
[rtabmap-1]   'type_support is null, at /tmp/binarydeb/ros-eloquent-rmw-cyclonedds-cpp-0.4.4/src/rmw_node.cpp:1212'
[rtabmap-1] 
[rtabmap-1] rcutils_reset_error() should be called after error handling to avoid this.
[rtabmap-1] <<<
[rtabmap-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[rtabmap-1]   what():  could not create publisher: type_support is null, at /tmp/binarydeb/ros-eloquent-rmw-cyclonedds-cpp-0.4.4/src/rmw_node.cpp:1212, at /tmp/binarydeb/ros-eloquent-rcl-0.8.4/src/rcl/publisher.c:173
[ERROR] [rtabmap-1]: process has died [pid 26401, exit code -6, cmd '/home/preed/src/ros2/rtabmap/install/rtabmap_ros/lib/rtabmap_ros/rtabmap -d --ros-args --params-file /tmp/launch_params_n8tzjyd5 -r scan:=/scan'].
[rtabmapviz-2] 1585369846.227603 [0] rtabmapviz: using network interface eno1 (udp/192.168.2.100) selected arbitrarily from: eno1, docker0
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_depth = false
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_rgb = false
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_stereo = false
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_rgbd = false (rgbd_cameras=1)
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_odom_info = false
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_user_data = false
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_scan = true
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_scan_cloud = false
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: queue_size    = 10
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: approx_sync   = true
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: Using configuration from "/home/preed/.ros/rtabmapGUI.ini"
[rtabmapviz-2] libpng warning: iCCP: known incorrect sRGB profile
[rtabmapviz-2] libpng warning: iCCP: known incorrect sRGB profile
[rtabmapviz-2] libpng warning: iCCP: known incorrect sRGB profile
[rtabmapviz-2] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rtabmapviz-2] [INFO] [rtabmapviz]: Reading parameters from the ROS server...
[rtabmapviz-2] [ERROR] [rtabmapviz]: Can't call rtabmap parameters service, is the node running?
[rtabmapviz-2] [INFO] [rtabmapviz]: Parameters read = 0
[rtabmapviz-2] Failed to find library 'rtabmap_ros__rosidl_typesupport_introspection_cpp'
[rtabmapviz-2] 
[rtabmapviz-2] >>> [rcutils|error_handling.c:107] rcutils_set_error_state()
[rtabmapviz-2] This error state is being overwritten:
[rtabmapviz-2] 
[rtabmapviz-2]   'type support not from this implementation, at /tmp/binarydeb/ros-eloquent-rmw-cyclonedds-cpp-0.4.4/src/rmw_node.cpp:991'
[rtabmapviz-2] 
[rtabmapviz-2] with this new error message:
[rtabmapviz-2] 
[rtabmapviz-2]   'type_support is null, at /tmp/binarydeb/ros-eloquent-rmw-cyclonedds-cpp-0.4.4/src/rmw_node.cpp:1433'
[rtabmapviz-2] 
[rtabmapviz-2] rcutils_reset_error() should be called after error handling to avoid this.
[rtabmapviz-2] <<<
[rtabmapviz-2] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[rtabmapviz-2]   what():  could not create subscription: type_support is null, at /tmp/binarydeb/ros-eloquent-rmw-cyclonedds-cpp-0.4.4/src/rmw_node.cpp:1433, at /tmp/binarydeb/ros-eloquent-rcl-0.8.4/src/rcl/subscription.c:168
[ERROR] [rtabmapviz-2]: process has died [pid 26402, exit code -6, cmd '/home/preed/src/ros2/rtabmap/install/rtabmap_ros/lib/rtabmap_ros/rtabmapviz --ros-args --params-file /tmp/launch_params_7ln5gpq9 -r scan:=/scan'].
```

I'm using `rmw_cyclonedds_cpp` as my default DDS implementation, but `fastrtps` and `opensplice` both fail with similar errors.  I can also verify that, like the error message says, no library named `rtabmap_ros__rosidl_typesupport_introspection_cpp` was built inside either my `build` or `install` directory.  Am I missing something?

matlabbe, 2020-04-03:
Hi,
Just tested this on my default Ubuntu 18.04 / eloquent setup:
```
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:~/ros2_ws/src/turtlebot3_simulations/turtlebot3_gazebo/models
export TURTLEBOT3_MODEL=burger
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
ros2 launch rtabmap_ros turtlebot3_scan.launch.py
```
No prob:
[image]

I didn't play with DDS implementation, so I guess I am using the default eloquent one. How can I get a similar setup than yours? I've found this issue that may be related: https://github.com/ros2/rosidl/issues/441
Not sure if it is related to code here:
https://github.com/introlab/rtabmap_ros/blob/4b4a5f0a1f2d414fbae5b93ca20378eeb9431628/CMakeLists.txt#L416-L422

cheers,
Mathieu

pjreed, 2020-04-03:
To switch to a different DDS implementation, you just need to install its relevant packages, recompile any packages you've built form source, set the appropriate `RMW_IMPLEMENTATION` environment variable, and then run things as normal:

```bash
sudo apt install ros-eloquent-*cyclonedds*
# clean and rebuild your workspace
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
ros2 launch ...
```

After you've installed and rebuilt, you can switch back and forth just by changing `RMW_IMPLEMENTATION` (or unsetting it to use FastRTPS).

I think Dirk's last comment in ros2/rosidl#441 may be the exact issue I'm having:

> Atm the generated library for the interfaces must have the basename ${PROJECT_NAME}.

I made one of the changes he suggested, and now it launches for me!  I'll open a PR shortly.

closed 2020-04-03 by pjreed

matlabbe, 2020-04-03:
Great! Thank you for the pull request. I tested with `rmw_fastrtps_cpp` and `rmw_cyclonedds_cpp` and the demo above is working. However, with `rmw_cyclonedds_cpp`, rtabmapviz _sometimes_ fails to read parameters when opening preferences. 
```
[rtabmapviz-2] [INFO] [rtabmapviz]: Parameters read = 0
[rtabmapviz-2] [WARN] [rtabmapviz]: Failed to get RTAB-Map parameters from ROS server, the rtabmap node may be not started or some parameters won't work...
```
I didn't see this problem with `rmw_fastrtps_cpp`, which seems to correctly call the parameters service _every time_. If this is coming a new problem, I'll let someone open a new issue.

pjreed, 2020-04-03:
Interesting; you might also try with OpenSplice (`sudo apt install ros-eloquent-*opensplice*`, then `export RMW_IMPLEMENTATION=rmw_opensplice_cpp`) and see if it has the same behavior.  I will keep an eye out to see if I see what might be going wrong there.

PR #400 (2020-04-03): Use project name as basename for interfaces
As noted by Dirk Thomas in ros2/rosidl#441, libraries for generated
interfaces must use ${PROJECT_NAME} as their basename.  This PR
modifies rtabmap_ros so that it does so.

It also tweaks a couple of rosdep keys because nav2_common was used in
some launch files but not included, and message_runtime does not exist
in ROS 2.

Fixes #397
commit 160ac38e (2020-04-03): Use project name as basename for interfaces

As noted by Dirk Thomas in ros2/rosidl#441, libraries for generated
interfaces must use ${PROJECT_NAME} as their basename.  This PR
modifies rtabmap_ros so that it does so.

It also tweaks a couple of rosdep keys because nav2_common was used in
some launch files but not included, and message_runtime does not exist
in ROS 2.

Fixes #397