introlab/rtabmap_ros #399: Building rtabmap from source with ros

seanxu112, 2020-04-03:
I am currently trying to install RTABMAP from source in order to calibrate camera and use different functionalities of RTABMAP standalone version. I am following the procedure as the repo says, basically pulling and making. The first problem I encountered was the following:
`undefined reference to `pcl::search::Search<pcl::PointXYZ>::getName`
For all point types, while making. I searched online, it was a problem with have to install pcl from source or install pcl 1.8 version. In order to prevent pcl_ros from functioning, I pulled pcl 1.8, compiled and changed the install folder on my Desktop. Then I added this line for rtabmap's CMakeLists.txt:
`list(APPEND CMAKE_PREFIX_PATH "/home/nvidia/pcl_1_8_0_install")`
And I also changed the required PCL version from 1.7 to 1.8 for me to enforce this version. 
At first make completed, but when I clone rtabmap_ros into my catkin_ws, there was some error for catkin_make -j1. Also there is warning saying that there might be a conflict for 2 pcl versions. Thus, I uninstalled pcl 1.8 and changed CMakeLists.txt in rtabmap back because I saw [this post ](https://github.com/introlab/rtabmap/issues/127#issuecomment-253615144)

Now I am just getting an error compiling rtabmap:
`c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-6/README.Bugs> for instructions.
tools/ExtractObject/CMakeFiles/extractObject.dir/build.make:62: recipe for target 'tools/ExtractObject/CMakeFiles/extractObject.dir/main.cpp.o' failed
make[2]: *** [tools/ExtractObject/CMakeFiles/extractObject.dir/main.cpp.o] Error 4
`
Also can I just use make -j4 for compiling rtabmap?

Sincerely,

Sean Xu

seanxu112, 2020-04-03:
I removed rtabmap directory and cloned it again, make is successful, but got this error at `sudo make install`

`[100%] Built target noEventsExample
Linking CXX executable CMakeFiles/CMakeRelink.dir/rtabmap
/usr/bin/aarch64-linux-gnu-ld: warning: libg2o_opengl_helper.so, needed by /opt/ros/kinetic/lib/libg2o_types_slam2d.so, not found (try using -rpath or -rpath-link)
/opt/ros/kinetic/lib/libg2o_types_slam3d.so: undefined reference to `g2o::opengl::drawPyramid(float, float)'
/opt/ros/kinetic/lib/libg2o_types_slam3d.so: undefined reference to `g2o::opengl::drawBox(float, float, float)'
/opt/ros/kinetic/lib/libg2o_types_slam2d.so: undefined reference to `g2o::opengl::drawPoint(float)'
/opt/ros/kinetic/lib/libg2o_types_slam2d.so: undefined reference to `g2o::opengl::drawArrow2D(float, float, float)'
collect2: error: ld returned 1 exit status
app/src/CMakeFiles/rtabmap.dir/build.make:962: recipe for target 'app/src/CMakeFiles/CMakeRelink.dir/rtabmap' failed
make[2]: *** [app/src/CMakeFiles/CMakeRelink.dir/rtabmap] Error 1
CMakeFiles/Makefile2:480: recipe for target 'app/src/CMakeFiles/rtabmap.dir/preinstall' failed
make[1]: *** [app/src/CMakeFiles/rtabmap.dir/preinstall] Error 2
Makefile:155: recipe for target 'prei`

Would it be because I forgot to sudo make uninstall before I removed the whole directory?

matlabbe, 2020-04-04:
It is strange, your g2o binaries seem broken or missing some libraries about opengl. What is your system?

For pcl, it is easier to keep only one version installed. This will avoid conflicts when linking with pcl_ros.

seanxu112, 2020-04-04:
I am using Jetson TX2 with ubuntu 16.04, ros kinetic. If g2o is the problem, should i just sudo apt-get remove ros-kinetic-libg2o and install it again? For pcl, it is mostly because of `undefined reference to pcl::search::Search<pcl::PointXYZ>::getName` problem I decided to install pcl 1.8

matlabbe, 2020-04-04:
For g2o, you can remove the binaries and rebuild/install from source. For PCL with jetpack 3 on Kinetic, I don't remember having seen this kind of PCL error: https://github.com/introlab/rtabmap_ros#build-from-source-for-nvidia-jetson-with-opencv-4-tegra

seanxu112, 2020-04-07:
Since I actually compiled rtabmap on my other machine, I am not pushing hard on compiling on tx2. Will probably try to compile it again in the future and will let you know the progress then.

seanxu112, 2020-11-03:
Hello Mathieu, 

I am finally trying to compile rtabmap and rtabmap_ros on TX2 again. I am trying to use libpointmatcher for better outlier rejection. This time I am able to compile rtabmap but when I was trying to compile rtabmap_ros version 0.20.0-kinetic I got internal compiler error:

c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
algo/software/rtabmap_ros/src/rtabmap_ros/CMakeFiles/rtabmap_sync.dir/build.make:158: recipe for target 'algo/software/rtabmap_ros/src/rtabmap_ros/CMakeFiles/rtabmap_sync.dir/src/impl/CommonDataSubscriberRGBD.cpp.o' failed
make[2]: *** [algo/software/rtabmap_ros/src/rtabmap_ros/CMakeFiles/rtabmap_sync.dir/src/impl/CommonDataSubscriberRGBD.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
[ 97%] Building CXX object algo/software/rtabmap_ros/src/rtabmap_ros/CMakeFiles/rtabmap_plugins.dir/src/nodelets/undistort_depth.cpp.o
[ 97%] Built target rtabmap_rviz_plugins
[ 97%] Building CXX object algo/software/rtabmap_ros/src/rtabmap_ros/CMakeFiles/rtabmap_plugins.dir/src/nodelets/imu_to_tf.cpp.o
[ 98%] Building CXX object algo/software/rtabmap_ros/src/rtabmap_ros/CMakeFiles/rtabmap_plugins.dir/src/nodelets/rgbd_sync.cpp.o
[ 98%] Building CXX object algo/software/rtabmap_ros/src/rtabmap_ros/CMakeFiles/rtabmap_plugins.dir/src/nodelets/stereo_sync.cpp.o
[ 98%] Building CXX object algo/software/rtabmap_ros/src/rtabmap_ros/CMakeFiles/rtabmap_plugins.dir/src/nodelets/rgbd_relay.cpp.o
CMakeFiles/Makefile2:9358: recipe for target 'algo/software/rtabmap_ros/src/rtabmap_ros/CMakeFiles/rtabmap_sync.dir/all' failed
make[1]: *** [algo/software/rtabmap_ros/src/rtabmap_ros/CMakeFiles/rtabmap_sync.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 98%] Linking CXX shared library /home/nvidia/catkin_ws/devel/lib/librtabmap_plugins.so
[ 98%] Built target rtabmap_plugins
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j6" failed

I am using ubuntu 16.04, with gcc 5.5.0, both rtabmap and rtabmap_ros are on tag/0.20.0-kinetic.

seanxu112, 2020-11-05:
Resolved building rtabmap_ros by changing to kinetic-devel branch instead of 0.20.0-kinetic tag, will do further testing to make sure it is running as expected.

matlabbe, 2020-11-07:
That error `c++: internal compiler error: Killed (program cc1plus)` is often a lack of RAM. Try `catkin_make -j1`. In particular, all the `CommonDataSubscriber*****.cpp` files require around ~1.6GB to build. When building 6 at the  same time, You would need ~9.6 GB of RAM.