introlab/rtabmap_ros #401: Covariance

MotorTrick, 2020-04-22:
Hey, very nice work here.
I have a couple of question for you :) Can I get somehow the covariance of the last node that has been added to the graph? Is this information published? Or, even if I don't think so, maybe it's the current odom covariance? 

If not how can you compute the loop-closure covariance?

As far as I understand you add nodes to the graph, when a loop closure happens you optimize the graph and publish a pose+covariance in rtabmap/localization_pose topic. But this has to come from somewhere I think.

matlabbe, 2020-04-27:
In the graph, odometry covariance is set on consecutive nodes added to graph. This covariance is already published in the odometry topic. 

On loop closure, the covariance set in localization_pose topic is computed from the marginals after graph optimization. For g2o, see [here](https://github.com/introlab/rtabmap/blob/1a3c90e8988c9479f087bd9855fe7344850fc027/corelib/src/optimizer/OptimizerG2O.cpp#L1127) for the 6DoF case. For gtsam, see [here](https://github.com/introlab/rtabmap/blob/1a3c90e8988c9479f087bd9855fe7344850fc027/corelib/src/optimizer/OptimizerGTSAM.cpp#L663-L664).

cheers,
Mathieu

MotorTrick, 2020-04-27:
Perfect, may I ask also why sometimes the marginals computation is done twice in a row, within the same computation? If you try print the covariance for example [here](https://github.com/introlab/rtabmap/blob/1a3c90e8988c9479f087bd9855fe7344850fc027/corelib/src/optimizer/OptimizerGTSAM.cpp#L663-L664) sometimes I got two (consecutive) printouts even if rtabmap is set to pause when loop closures happens. I don't have other lines between these two so I think that this function is being called twice.

matlabbe, 2020-04-27:
If proximity detection is done, it may optimize a local path to localize on. You can show all debug logs with argument `--udebug` or by calling service `/rtabmap/rtabmap/log_debug`

eliabntt, 2020-04-28:
ok, very nice. I'll check this out. Thank you :)

MotorTrick, 2020-04-28:
Ok, perfect :) thank you so much :)

closed 2020-04-28 by MotorTrick