introlab/rtabmap_ros #403: No PointCloud2 data at all

czrcbl, 2020-04-23:
Hi,

I am trying to use `rtabmap` to build a map and segment ground and obstacles from it.
I am using ROS Melodic and a BumbleBee2 camera on simulation.
I am able to run `rtabmap` with no erros and I get a plausible output on `rtabmapviz`.
The problem is that all topics of the type `PointCloud2` published from `rtabmap` are either advertised but not being published to or receiving messages with empty point clouds.
This way I am unable to extract the information I am seeking.

This is the launch file I am using to start `rtabmap`:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<launch>
    <include file="$(find rtabmap_ros)/launch/stereo_mapping.launch">
        <arg name="rtabmapviz" value="true" /> 
        <arg name="rviz" value="false" />
        <arg name="stereo_namespace" value="/stereo_camera" />
        <arg name="approx_sync" value="true" />
    </include>
</launch>
```

I really have no idea of how to debug this since I don't see any error.

matlabbe, 2020-04-23:
Do you have a database to share? (should be saved by default in `~/.ros/rtabmap.db`)

czrcbl, 2020-04-23:
Here the database file:
https://send.firefox.com/download/9a9014697d33bc89/#L63N2_HfDJZC1Gd1XM2NWw

matlabbe, 2020-04-27:
The download always fails at 42%. Can you use something else than Firefox Send? (like google drive, dropbox or wetransfert)

matlabbe, 2020-04-27:
It seems opencv_contrib master is broken, as SIFT [has been moved](https://github.com/opencv/opencv_contrib/commit/1d77931b30b26c40b60c4fef931d8f0f048dc6ba) from nonfree to free, but only appears in 4.3.0 branch (SIFT just doesn't exist in master branch). Try rebuild opencv with opencv_contrib 4.3.0:
```
cd opencv_contrib
git checkout 4.3.0
... go make opencv
```

czrcbl, 2020-04-27:
Here a Dropbox link:
https://www.dropbox.com/s/la1ypxpg6wmu3t8/rtabmap.db?dl=0

matlabbe, 2020-04-27:
Try launching with an empty database with `rtabmap_args:="-d"`:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<launch>
    <include file="$(find rtabmap_ros)/launch/stereo_mapping.launch">
        <arg name="rtabmapviz" value="true" /> 
        <arg name="rtabmap_args" value="-d --Grid/RangeMax 0 --Grid/CellSize 0.5 --Grid/ClusterRadius 2 --Grid/FromDepth true" />
        <arg name="rviz" value="false" />
        <arg name="stereo_namespace" value="/stereo_camera" />
        <arg name="approx_sync" value="true" />
    </include>
</launch>
```
I increased cell size because the scale of the environment is large.

czrcbl, 2020-04-28:
Thank you. Now I am able to get the clouds.

closed 2020-04-28 by matlabbe