introlab/rtabmap_ros #404: Baseline value 0.00

ankitvora7, 2020-04-24:
Hi,

Even though my values in the camera_info topic seem to be correct, I'm still getting a 0 baseline value. This is the error 
``` The stereo baseline (0.000000) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.```

Here are my camera calibration values
```
header: 
  seq: 43821
  stamp: 
    secs: 4398
    nsecs: 200000000
  frame_id: "robot/left_camera/camera_info"
height: 480
width: 640
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [381.36246688113556, 0.0, 320.5, 0.0, 381.36246688113556, 240.5, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [381.36246688113556, 0.0, 320.5, -157.29401728003162, 0.0, 381.36246688113556, 240.5, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
```
and the right camera is 

```
header: 
  seq: 44788
  stamp: 
    secs: 4494
    nsecs: 900000000
  frame_id: "robot/right_camera/camera_info"
height: 480
width: 640
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [381.36246688113556, 0.0, 320.5, 0.0, 381.36246688113556, 240.5, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [381.36246688113556, 0.0, 320.5, -157.29401728003162, 0.0, 381.36246688113556, 240.5, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
```
The actual value should've been (-157.294/381.362) but it is 0.0. Any idea what might be going wrong ?

matlabbe, 2020-04-27:
The left camera_info should not have P(0,3) set (normally it is zero). The baseline is actually computed like [this](https://github.com/introlab/rtabmap/blob/1a3c90e8988c9479f087bd9855fe7344850fc027/corelib/include/rtabmap/core/StereoCameraModel.h#L104):
`left_.Tx() / left_.fx() - right_.Tx()/right_.fx()`

cheers,
Mathieu

surajmahangade, 2020-08-09:
I am facing a similar problem.
Can you please explain how exactly I solve this problem.
I have 2 cameras publishing their camera info independently. How exactly the camera info needs to be updated.

Thank you

supersaiyajinggod, 2020-10-12:
I am facing the same problem while i am  using the realsense d435i.
I solve the problems using another function which get the baseline from the tf tree.
The tf tree provide the accurate baseline which calibrated before leaving the factory, if you use the realsense_ros.
Modify the line 1789 in MsgConversion.cpp with:
	if (0) {
		stereoModel = rtabmap_ros::stereoCameraModelFromROS(leftCamInfoMsg, rightCamInfoMsg, localTransform);
	} else {
		stereoModel = rtabmap_ros::stereoCameraModelFromROS(leftCamInfoMsg, rightCamInfoMsg, frameId, listener, waitForTransform);
	}

matlabbe, 2020-10-12:
Two months ago, this has been added specifically for the D400 stereo camera info missing Tx (in stereo IR mode):
https://github.com/introlab/rtabmap_ros/blob/10ed0109b8c7c9e507808e62d2bfde68fcc08b28/src/MsgConversion.cpp#L1848-L1855