introlab/rtabmap_ros #405: Database management

MotorTrick, 2020-04-28:
Hi, I tried to search it out on the code but I've found some difficulties. 
By running the code I noticed that the database is being built as the program is running so I assume there are some functions around that add/edit/delete lines in it.
My question is where do you manage tables in the database? 
In particular I would be interested in accessing the database through code and modify some tables especially the one that manage nodes (by adding one column) and know where this function is called every time a node is added to the database. To do so I need to be sure obviously not to break anything when you update that specific row (if you ever do so).
In addition, related to my previous issue, I'd like to access the node pose (x,y, yaw) and covariance :)

MotorTrick, 2020-04-30:
Plus, I've seen in the gui (rtabmapviz) that I can visualize for each node on the graph a different color based on the number of features for that node. More or less it's the information that I'd like to retrieve. Is it possible to call  it from a function/api directly when needed?

matlabbe, 2020-05-26:
Nodes are saved to database in [Memory::moveToTrash()](https://github.com/introlab/rtabmap/blob/0902ccc340bfb99695688a2379acfcd8691e29ae/corelib/src/Memory.cpp#L2427). through `DBDriver` class (the interface between c++ objects and SQL).

The odometry pose is in Signature object: 
https://github.com/introlab/rtabmap/blob/0902ccc340bfb99695688a2379acfcd8691e29ae/corelib/include/rtabmap/core/Signature.h#L133

We keep odometry covariance in the links of the graph. It is the twist covariance, not the pose (thus bounded).

To get the actual pose covariance, you would have to get the marginals from g2o or gtsam after graph optimization. We currently return only [marginal of the last pose in the graph](https://github.com/introlab/rtabmap/blob/0902ccc340bfb99695688a2379acfcd8691e29ae/corelib/src/optimizer/OptimizerG2O.cpp#L1039-L1067).

We update TF `/map -> /odom` only on localization based o the current odometry and the localization transformation against the map's graph. So, if you get the pose as `/map -> /base_link`, it will always contains the new odometry pose corrected by last localization.

There is no api call to get colors on the graph, this is hard-coded for visualization purpose:
https://github.com/introlab/rtabmap/blob/0902ccc340bfb99695688a2379acfcd8691e29ae/guilib/src/MainWindow.cpp#L2121-L2139

cheers,
Mathieu

closed 2020-10-08 by MotorTrick