introlab/rtabmap_ros #406: Get Plan

MotorTrick, 2020-04-30:
Hi, I'm trying to use getPlan service to plan my path toward a custom Pose(NOT node) from my current position but I'm not able to have it running correctly: the answer is always blank without any planning.

matlabbe, 2020-05-26:
If the pose is too far from the graph or if we are not currently localized in the graph, it will fail. You can set rtabmap logger's level to debug (`rosservice call /rtabmap/rtabmap/log_debug`) to better see why rtabmap is failing.

eliabntt, 2020-05-27:
I've found the problem here. Orientation can't be 0,0,0,0 in the message otherwise it won't give output. After that there is a tolerance of x m from the graph based on the parameter you set but still looking at the source code it should be possible to have a plan up to the last node without having to increase this too much, hence avoiding to include other nodes if you are planning to have goals very far from the graph. ([Rtabmap.cpp:5049::ComputePath()](https://github.com/introlab/rtabmap/blob/master/corelib/src/Rtabmap.cpp#L5275)) In this way you can travel the graph and you'll have to worry about only about the local planning between the last node and the goal.

eliabntt, 2020-05-27:
@matlabbe I think the problem can be easily solved [here](https://github.com/introlab/rtabmap_ros/blob/0c4779c19e25f7a2967c2caad5ad1ffe8c04ef74/src/CoreWrapper.cpp#L3137) by checking if the quaternion is all zeros and forcing a 1 on the w dimension.
```
    if (req.goal.pose.orientation.x == 0 && req.goal.pose.orientation.y == 0 && req.goal.pose.orientation.z == 0 && req.goal.pose.orientation.w == 0)
        req.goal.pose.orientation.w = 1;
```

matlabbe, 2020-06-02:
@eliabntt thx for the suggestion, I've made the changes in the commit above. The request `frame_id` is now also optional (assume map frame by default) to be able to set minimal values in the request.

MotorTrick, 2020-06-03:
Perfect, this works! thanks to both of you!

closed 2020-06-03 by MotorTrick