introlab/rtabmap_ros #407: Occupancy grid vs 2D map

eliabntt, 2020-04-30:
Hi, I've a strange problem with my rtabmap. My occupancy grid seems correct while my 2D map is not. I'm trying to use my rgbd data to get obstacles in the map but I'm probably doing something wrong. I've attached my database [here](https://we.tl/t-zrezlQNHuh) with current settings if you can check it out.

Short story long I'm using a simulated Realsense D435 placed vertically on my robot, doing for now visual odometry and use these rgbd data to place obstacles in the map. For loop closure I'm using both rgbd+icp registration (strategy=2) and optimizer either gtsam or g2o. In general the throughput of rtabmap is quite good with the given settings (around 100/200 ms)

[image]

ninamwa, 2020-05-10:
Hi. Did you manage to use both LRD and depth to create the map? When i subscribe to both scan and rgbd it seems like only the scan is included in the 2D occupancy map. When i only subscribe to rgbd the map looks different (because of obstacles like tables etc).

eliabntt, 2020-05-12:
Hi @ninamwa I think I'll work more thoroughly on that tomorrow, for sure by the end of the week. There are a lot of parameters to test and check.

matlabbe, 2020-05-13:
For the occupancy grid, we cannot use both depth image and lidar at the same time (see `Grid/FromDepth` parameter to choose which one you want to use). I cannot download your database (link expired) but what I see is that some tuning against the Grid/ parameters for normal segmentation approach would be required. In the top red square, is there really an obstacle?

eliabntt, 2020-05-18:
Yup, it's a table and a couple of chairs. I managed to solve that by tuning some parameters. Obviously is less precise than a LRF but I'm getting closer results w.r.t. my scene. 
[image]

[Here](https://github.com/introlab/rtabmap_ros/files/4643267/rtabmap.txt) it's my current config if you can check I would much appreciate since I'm just starting with my Ph.D. :)

In your opinion is it correct to use localization_pose output within EKF? I'm planning to use wheel+IMU readings as high frequency input for the EKF (robot localization package). When a loop closure is detected I have a localization_pose output with a covariance computed (either from gtsam or g2o) and that will refine my EKF (avoiding or increasing drifting). My wheels and IMU odoms have static covariances but when fused together in EKF the localization cov increase constantly while moving as expected but when RTABMap localize itself in the environment I think this should be reflected. (About this I've also some other [doubts](https://github.com/introlab/rtabmap/issues/539) ).
Visual Odometry is too much unstable I think to be used within EKF.

matlabbe, 2020-05-22:
The `localization_pose` is discrete in time (like a GPS) as other odometry sources are continuous. There is an example here: http://official-rtab-map-forum.67519.x6.nabble.com/Filtering-rtabmap-localization-jumps-with-robot-localization-in-2D-td5931.html

eliabntt, 2020-06-03:
Yes, I've seen that one thanks :) right now I've this setting:

```
            <param name="frequency" value="50"/>
            <param name="sensor_timeout" value="0.2"/>
            <param name="two_d_mode" value="true"/>
            <param name="publish_tf" value="true"/>

            <param name="base_link_frame" value="base_link"/>
            <param name="odom_frame" value="odom"/>
            <param name="world_frame" value="odom"/>
            <param name="map_frame" value="map"/>

            <param name="transform_time_offset" value="0.1"/>

            <param name="odom0" value="/odom_calc"/>
            <param name="imu0" value="/imu/data"/>
            <param name="pose0" value="/rtabmap/localization_pose"/>

            <!-- The order of the values is x, y, z, roll, pitch, yaw, vx, vy, vz, vroll, vpitch, vyaw, ax, ay, az. -->
            <rosparam param="odom0_config">[false, false, false,
                false, false, false,
                true, true, false,
                false, false, true,
                false, false, false]</rosparam>

            <rosparam param="imu0_config">[false, false, false,
                false, false, true,
                false, false, false,
                false, false, true,
                true, true, true]
            </rosparam>

            <rosparam param="pose0_config">[true, true, false,
                false, false, true,
                false, false, false,
                false, false, false,
                false, false, false]</rosparam>

            <param name="imu0_differential" value="true"/>
            <param name="odom0_differential" value="true"/>
            <param name="pose0_differential" value="false"/>

            <param name="odom0_relative" value="false"/>
            <param name="imu0_relative" value="true"/>
            <param name="pose0_relative" value="false"/>

            <!-- todo true or false -->
            <param name="imu0_remove_gravitational_acceleration" value="true"/>

            <param name="print_diagnostics" value="false"/>

            <!-- ======== ADVANCED PARAMETERS ======== -->
            <param name="odom0_queue_size" value="5"/>
            <param name="imu0_queue_size" value="50"/>
            <param name="pose0_queue_size" value="5"/>

            <param name="dynamic_process_noise_covariance" value="true"/>
```

It's a bit different w.r.t. your example. My odometry is a custom one that I obtain through a custom plugin (mostly based on p3d) since my robot is omnidirectional. There I add noise directly to the velocities after having applied them through a PID controller. In the EKF here I fuse then the velocities of my odom source with the acceleration of the IMU and since we're in 2D-flat surface I'm not really interested in roll and yaw.
Another difference is the set (odom,world,map)_frame where you set both "world" and "map" to _map_ but I need this as odometry source and hence I set "world" to _odom_ frame. In this way I'm able to get both the tf map->odom and odom->base_link. [Here](http://docs.ros.org/melodic/api/robot_localization/html/integrating_gps.html?highlight=world_frame), they suggest to use two modules one with world = odom to fuse continuos data, one with world = map to fuse the previous module and the "GPS" but as of now it's working correctly as it is. What do you think about this?

matlabbe, 2020-06-03:
They also say:

> This is just a suggestion, however, and users are free to fuse the GPS data into a single instance of a robot_localization state estimation node.

In this case, is robot_localization publishing both map->odom and odom->base_link? If so, is map->odom matches /rtabmap/localization_pose or is it merged in /odom -> /base_link where /map->/odom is always Identity and /odom->base_link jumps on loop closure?

eliabntt, 2020-06-03:
In this case ekf_robot_localization is used as odometry so I've just odom->base_link from it. Instead rtabmap takes care of the transformation map->odom.

closed 2021-03-24 by eliabntt