introlab/rtabmap_ros #411: Some warnings when launching stereo_odometry

willzoe, 2020-05-04:
When I launch stereo_odometry, some warning messages will be prompted:
```
[ WARN] (2020-05-04 08:52:48.637) OdometryF2M.cpp:542::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=14) between -1 and 3" (guess=xyz=-0.296457,-1.958867,-0.121009 rpy=0.078903,-0.374390,1.155881)
[ WARN] (2020-05-04 08:52:48.638) OdometryF2M.cpp:311::computeTransform() Failed to find a transformation with the provided guess (xyz=-0.296457,-1.958867,-0.121009 rpy=0.078903,-0.374390,1.155881), trying again without a guess.
[ WARN] (2020-05-04 08:52:48.902) OdometryF2M.cpp:552::computeTransform() Trial with no guess succeeded!
```
After this, it seems that node has normally published odom.

But I want to know what these warning messages mean? Will they affect odom?

matlabbe, 2020-05-12:
It can happen if the camera has some auto exposure, motion blur, is shaking or did some fast movements. For features matching, it uses the previous motion as guess for the new one (e.g., constant motion model), this generally helps to get more and better matches, so more accurate odometry. If it fails to do so for one of the reason above, it assumes its guess is wrong and try a global features matching instead to recover. If it succeeds, odometry will continue as usual. We show the warning because at that time the odometry may be drifted more than usual.

`Vis/CorGuessWinSize` (default 20 pixels) can be increased to track better if the camera has larger changes in velocity. It could also be increased with higher resolution images (this parameter is tuned for 640x480 resolution by default).

willzoe, 2020-05-12:
If the above two methods for feature matching fail, will the odometer information not be released?

Does `Odom/ResetCountdown` restart the odometer after the odometer fails to be released?

matlabbe, 2020-05-13:
If `Odom/ResetCountdown` is not set, odometry will be lost and recover only if you move back the camera where it got lost. If `Odom/ResetCountdown` is set to 1, odometry will automatically reset after 1 fail.

willzoe, 2020-05-13:
Thank you very much.

I have some problems with odometer fusion. And I posted my question here #398 . 

Hope to get your help.

closed 2020-05-13 by willzoe