introlab/rtabmap_ros #412: Can not work with ICP 2D lidar in newest version

NamDinhRobotics, 2020-05-04:
Hello Mathieu,
My name Nam Dinh, 
Currently, I am working with 2D Lidar slam fusion, then find a perfect SLAM is Rtabmap.
When I tried with :
roslaunch rtabmap_ros demo_hector_mapping.launch hector:=false
in Rtabmap version 0.19.3 in Ros Kinetic, it's success and works well.
But, when updating the new version 0.19.7 or 0.20.0, in Rviz which can not show the same results( without /odom in Fixed Frame in rviz)
Would you help me and give suggestions for that?
Best regards,
Nam
[image]

matlabbe, 2020-05-26:
There were 2 bugs at the same time when we added intensity to converted scans and the complexity check by default. The commits above should fix the problem. Thnaks for reporting!

closed 2020-05-26 by matlabbe

NamDinhRobotics, 2020-05-26:
Great! Thank you so much.