introlab/rtabmap_ros #413: how to ignore robot body that is partially in camera view

tianshiz, 2020-05-05:
Part of my robot is within view of the camera so when I start mapping those parts become mapped. 
I tried setting Grid/RangeMin but it doesn't seem to work. If I set RangeMin to 5m, nothing gets plotted anymore. 
I'm using 3 RGBD cameras stacked ontop of one another, oriented 30deg apart.

I'm on ROS melodic. here is my param:

> 
# RTAB-Map configuration

subscribe_depth:        false
subscribe_rgbd:         true
subscribe_rgb:          false
subscribe_stereo:       false
subscribe_scan:         false
subscribe_scan_cloud:   false
subscribe_user_data:    false
subscribe_odom_info:    false
rgbd_cameras:   3

database_path:      "~/.ros/rtabmap.db"
config_path:        "~/.ros/rtabmap.cfg"

frame_id:       "base_link"
map_frame_id:   "map"
odom_frame_id:  "odom"                   # odometry from odom msg to have covariance - Remapped by launch file
odom_tf_angular_variance:  1     # If TF is used to get odometry, this is the default angular variance
odom_tf_linear_variance:   1     # If TF is used to get odometry, this is the default linear variance
tf_delay:       0.02
publish_tf:     true                 # Set to false if fusing different poses (map->odom) with UKF

odom_sensor_sync:               false
wait_for_transform_duration:    0.2
approx_sync:                    true
rgbd_sync:                      false
queue_size: 10

scan_normal_k:  0

Grid:
    3D:                     true
    FlatObstacleDetected:   true
    MapFrameProjection:     false  #puts cloud in map or local robot frame
    GroundIsObstacle:       false
    PreVoxelFiltering:      true
    RayTracing:             true
    FromDepth:              true
    NormalsSegmentation:    true # [Segment ground from obstacles using point normals, otherwise a fast passthrough is used.]
    CellSize:               0.1
    ClusterRadius:          0.1
    MinClusterSize:         3
    DepthDecimation:        4
    DepthRoiRatios:         [0.0, 0.0, 0.0, 0.0]
    FootprintHeight:        1.6
    FootprintLength:        2
    FootprintWidth:         1.5
    MaxGroundAngle:         30.0
    MinGroundHeight:        0.0  #[Maximum ground height (0=disabled). Should be set if "Grid/NormalsSegmentation" is true.]
    MaxGroundHeight:        0.0
    MaxObstacleHeight:      1.7 #any higher obstacles won't block the robot
    NoiseFilteringMinNeighbors: 0
    NoiseFilteringRadius:   0.0
    NormalK:                20
    RangeMin:               5.0
    RangeMax:               20.0
Vis:
    MaxDepth: 20.0
    MinDepth: 4.5

GridGlobal:
    Eroded:                 false       # Erode obstacle cells
    FootprintRadius:        1        # Footprint radius (m) used to clear all obstacles under the graph
    FullUpdate:             true        # When the graph is changed, the whole map will be reconstructed instead of moving individually each cells of the map. Also, data added to cache won't be released after updating the map. This process is longer but more robust to drift that would erase some parts of the map when it should not
    MaxNodes:               0           # Maximum nodes assembled in the map starting from the last node (0=unlimited)
    MinSize:                1.0        # Minimum map size (m)
    OccupancyThr:           0.55        # Occupancy threshold (value between 0 and 1)
    ProbClampingMax:        0.971       # Probability clamping maximum (value between 0 and 1)
    ProbClampingMin:        0.1192      # Probability clamping minimum (value between 0 and 1)
    ProbHit:                0.7         # Probability of a hit (value between 0.5 and 1)
    ProbMiss:               0.4         # Probability of a miss (value between 0 and 0.5)
    UpdateError:            0.01        # Graph changed detection error (m). Update map only if poses in new optimized graph have moved more than this value

matlabbe, 2020-05-12:
Hi,

using Grid/RangeMin will filter everything inside 5 meters radius from the robot. Are the robot parts fixed in the depth images? You could create manually a mask to apply to depth images before sending them to rtabmap. If you want to filter moving parts on your robot (like arms), you may take a look at this package that is using the URDF of the robot to create the mask automatically for the depth image: https://github.com/blodow/realtime_urdf_filter

cheers,
Mathieu

tianshiz, 2020-05-13:
In reality, the robot parts can't be more than 0.5 from my sensors, not sure why it's not making a difference in MapCloud. Is Grid/RangeMin the right settings for that?

matlabbe, 2020-05-13:
Yes, but in your config file it was set to 5 meters, not 0.5 meter, so everything inside 5 meters around the robot was filtered.

tianshiz, 2020-05-14:
sorry, I was unclear with this config file. I've tried multiple iterations of Grid/RangeMin. In the end, it looks like it doesn't matter what rangemin I use, everything around the robot still gets mapped. I checked the rosparam /rtabmap/Grid/RangeMin and it shows the correct value set.

tianshiz, 2020-05-14:
does Grid/RangeMin only apply for building the occupancygrid and not the cloud_map? I think that's my problem here. So likely as you said the only way to do this is to mask the depth images or do realtime urdf filter

matlabbe, 2020-05-25:
`cloud_map` topic and `grid_map` topic uses `Grid/` parameters. For the MapCloud rviz plugin, it doesn't use Grid/ parameters, but directly the parameters in the plugin. 

Can you share a database?

tianshiz, 2020-05-28:
ah, that was the problem! I didn't realize it was an rviz setting.

closed 2020-05-28 by tianshiz