introlab/rtabmap_ros #414: librtabmap_core.so.0.19: cannot open shared object   after loading ros-kinetic-rtabmap-ros

patrickpoirier51, 2020-05-08:
Bonjour

I just loaded on my UBUNTU , sudo apt-get install ros-kinetic-rtabmap-ros
And when I launch , I get 
_error while loading shared libraries: librtabmap_core.so.0.19: cannot open shared object file: No such file or directory_

I see you suggest adding path in bashrc, that is pointing to /opt/ros/melodic/lib/x86_64-linux-gnu , so I changed melodic to kinetic but still get the same error.

Problem seem that librtabmap is older release
ls /opt/ros/melodic/lib/x86_64-linux-gnu/librtabmap_core*
librtabmap_core.so  librtabmap_core.so.0.17  librtabmap_core.so.0.17.6

I don't mind building from source, but I guess you need to fix this.

Salutations de la ville de Quebec

matlabbe, 2020-05-08:
Bonjour!

Which Ubuntu are you on? 16.04 is kinetic and 18.04 should be melodic. Normally doing `source /opt/ros/melodic/setup.bash` should setup the paths correctly.

Ubuntu 18.04:
```bash
sudo apt install ros-melodic-rtabmap ros-melodic-rtabmap-ros
source /opt/ros/melodic/setup.bash
```
Ubuntu 16.04:
```bash
sudo apt install ros-kinetic-rtabmap ros-kinetic-rtabmap-ros
source /opt/ros/kinetic/setup.bash
```

Salutations de Sherbrooke!

patrickpoirier51, 2020-05-08:
Ayoye, thats a fast reply, Thanks  :-)

Yes Ubuntu 16.04 and the source is in my bashrc file
source /opt/ros/kinetic/setup.bash
source ~/catkin_ws/devel/setup.bash

As I wrote above I don't think its an environment issue but an obsolete library that is installed :+1: 17 instead of 19.

Merci pour ton aide

patrickpoirier51, 2020-05-08:
I tested on my I7 Ubuntu 18.04 system , loaded  ros-melodic-rtabmap ros-melodic-rtabmap-ros and it works no problem

And as expected the /ros/melodic/lib has release 19 libraries

matlabbe, 2020-05-08:
Kinetic is on 0.19.3 too. Maybe just an updgrade of the package is required.
`sudo apt install ros-kinetic-rtabmap` should upgrade it to 0.19.

Cordialement,
Mathieu

NamDinhRobotics, 2020-05-08:
please try this command first:
sudo ldconfig

Regards,

patrickpoirier51, 2020-05-08:
Bonjour

OK my bad , I must have installed it on this system last year , so here is what I did:
**sudo apt-get --only-upgrade install ros-kinetic-rtabmap**
And libs are 0.19.3 now :-)

Thanks for your excellent support,  

Au plaisir et salutation a toute l'equipe

closed 2020-05-08 by matlabbe