introlab/rtabmap_ros #415: Problem with Eigen::DenseCoeffsBase

NamDinhRobotics, 2020-05-08:
Hello all,
I get a problem with Eigen when I tried with Rtabmap ROS version >= 0.19.7
I tried: roslaunch rtabmap_ros demo_robot_mapping.launch, then run: rosbag play --clock demo_mapping.bag
The errors are shown as:

Failed to find match for field 'intensity'.
agv_slam: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:353: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index, Eigen::Index) [with Derived = Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = float; Eigen::Index = long int]: Assertion `row >= 0 && row < rows() && col >= 0 && col < cols()' failed

please help me!
Best Regards,

matlabbe, 2020-05-12:
Was PCL also installed from source? Which version? If so, are pcl_ros and pcl_conversion packages rebuilt too?

NamDinhRobotics, 2020-05-19:
Sorry for the late response.
I use PCL 1.8,
I solved the problem by not using G2O,
Regards,

matlabbe, 2020-05-25:
This may be related to http://official-rtab-map-forum.67519.x6.nabble.com/PCL-version-tp6373p6382.html
With PCL 1.8 built from source, it is built with `-march=native` by default, so g2o should be built with `-DBUILD_WITH_MARCH_NATIVE=ON`

closed 2020-05-25 by matlabbe