introlab/rtabmap_ros #417: ROS2, which map topics do rtabmap publish to?

ninamwa, 2020-05-10:
Hi. 
In the documentation for RTABMAP belonging to ROS1, it says that the rtabmap node publish to the following topics: 

> grid_map (nav_msgs/OccupancyGrid):
> Mapping Occupancy grid generated with laser scans. Use parameters with prefixes map_ and grid_ below.

> proj_map (nav_msgs/OccupancyGrid):
> Mapping Occupancy grid generated from projection of the 3D point clouds on the ground. Use parameters with prefixes map_, grid_ and proj_ below.

When running the RTABMAP node for ROS2 I can use `ros2 node info rtabmap` to see that the node publish nav_msgs/OccupancyGrid to two topics: `/map` and `/grid_prob_map`. 
Are these two topics the same as grid_map and proj_map from the documentation? In that case, which are which? 

I am using both a 2D lidar scan and a 3D camera, which means i want to save a 2Dmap using both the laser scan and the projected 3D point clouds. I am using the nav2_map_server to save the map which means i have to subscribe to a topic with nav_msgs/OccupancyGrid messages.

matlabbe, 2020-05-12:
`/map` is `/proj_map`. `/grid_prob_map` is `/map` but with probabilities (values between 0 and 100) in cells instead of only -1,0 or 100.

`/proj_map` has been removed because even in ROS1, `/proj_map` is deprecated and it is the same topic than `/grid_map`. Well, I just updated the documentation about that.

`/map` (ROS2) or `/grid_map` (ROS1) are created from lidar **or** camera, but not both. In general, if you have a 2D lidar, you would want the map to be created with the lidar. For all 3D obstacles avoidance, that would be done by move_base's local costmap using both lidar and camera.

ninamwa, 2020-05-13:
Hi. Thank you for your answer. I am still a bit confused. 
In this tutorial http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot multiple examples combining lasers and a camera is shown. Is this just to make 3D maps?
As far as i can see R

matlabbe, 2020-05-13:
In the examples, when the lidar is used, by default `Grid/FromDepth` is set to false, creating a 2d occupancy grid. The camera is used for loop closure detection.

It is possible to merge two lidar scans into a single fake one, that is then provided to rtabmap. (laser scan to PointCloud2 -> aggregate the clouds -> then convert the combined PointCloud2 back to laser scan).

If you have many sensors from which you want to detect obstacles, you can convert all data to many PointCloud2 and use [rtabmap_ros::PointCloudAggregator](https://github.com/introlab/rtabmap_ros/blob/ros2/src/nodelets/point_cloud_aggregator.cpp) to merge them into a single PointCloud2 that can be fed to rtabmap (`scan_cloud` input topic with `subscribe_scan_cloud:=true`).

Note as I stated previously, the dynamic obstacles or some 3D obstacles don't need to be in the static map, as long as the local costmaps include them when revisiting the area (in which we can combine lidars and cameras easily).

For your last question, yes, you can switch from `Grid/FromDepth` to false to `Grid/FromDepth` to true for the second session.

ninamwa, 2020-05-15:
Thank you for a very clarifying answer! I will test the different methods and see which gives the best result! :)

closed 2020-05-15 by ninamwa

ninamwa, 2020-05-20:
Hi again. I tried merging my two laserscans by converting them to pointclouds, and then using the pointcloud aggregator to merge the clouds. When visualizing the cloud in Rviz, it looks good, and the 2D grid map created do also look correct, except that no walls are included? 
The first image is from using two laserscans merged into a pointcloud (scan_cloud topic), while the second image is from using only one scan (scan topic). An RGBD camera is also used in both cases. 
The shape of the map is all correct, but as you can see the black walls saying there is an obstacle disappears. Any ideas? 
[image]
[image]

The reason why i did not convert the pointcloud back into a laser scan as you suggested, is because i want to merge the two laserscans with the pointclouds from my cameras as well. Right now, this do not seem possible as the pointcloud from the camera has a field named rgb, while the pointclouds from the laserscans (transformed by the laserscan to pointcloud node) have a field named index. This causes an error in pcl_conversions.h as the name of the fields are not exact the same:
`PCL_ERROR ("[pcl::concatenatePointCloud] Name of field %d in cloud1, %s, does not match name in cloud2, %s\n", i, cloud1.fields[i].name.c_str (), cloud2.fields[i].name.c_str ());`

The other fields in the clouds (x,y,z) are similar. 
Have you used this method to merge rgbd and laserscan data? What can i do to make the pointclouds similar? 

Thankyou!

matlabbe, 2020-05-25:
The clearing of the walls may be caused by the ray tracing. Can you share the database?

You would have to convert all the different point clouds to the most basic ones, i.e., only XYZ channels, before concatenating them. pcl::copyPointCloud can be used to convert to PointXYZ format.

alexandru-oana, 2021-03-24:
> When running the RTABMAP node for ROS2 I can use `ros2 node info rtabmap` to see that the node publish nav_msgs/OccupancyGrid to two topics: `/map` and `/grid_prob_map`.

I've installed rtabmap + rtabmap_ros on foxy on Ubuntu 20.04, but i get **Unable to find node rtabmap.**

Any idees ?

devyani-1605, 2025-03-01:
I am implementing RTAB-Map ROS2 on a D435 camera for autonomous drone mapping. However, as soon as my drone becomes unstable, RTAB-Map gets stuck and shows the error:
[ERROR] (2025-03-01 [21:49:29](https://www.youtube.com/watch?v=oDilesu1los&t=78569s).631) Rtabmap.cpp:1406::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
and the map freezes and shows the red screen

matlabbe, 2025-03-16:
Red screens are explained here: https://github.com/introlab/rtabmap/wiki/Kinect-mapping#lost-odometry-red-screens

rtabmap is loosing odometry somehow. Please open another issue with launch file you are using.