introlab/rtabmap_ros #418: libvtk-dev conflict on debian buster - rpi4

giancan, 2020-05-14:
Hi all,
I have a debian buster on a raspberry pi4 and I am trying to install rtabmap on it, in ros or outside.
From what I understand, there is a conflict between the libpcl (which in buster requires vtk7) and rtabmap (which requires vtk6), correct?

Is there any (easy?) workaround?

matlabbe, 2020-05-14:
RTAB-Map works with vtk5, vtk6, vtk7, vtk8 (not tested more than that). If pcl is link to vtk7, when doing cmake for rtabmap, hard-code the minimal version of vtk to 7 [here](https://github.com/introlab/rtabmap/blob/55509c6c27712a25e6f3e4ce379e37b917fba57b/CMakeLists.txt#L272), or launch cmake with `-DVTK_DIR="path/to/VTKConfig.cmake/directory"` to that vtk 7 is found instead of vtk6.

giancan, 2020-05-14:
@matlabbe thanks. Actually it was (probably) not necessary. I was following another procedure (perhaps wrong):
cloned the source in catkin_ws/src folder and then run
rosdep install --from-paths src --ignore-src -r -y
to check for dependencies... and there I was getting the error with vtk.
Your reply reminded me I should have cmake the source and then make so I did it.
Well, now I get another error. A series of errors.
All are related with CV_CAP_(something).
I do have OpenCV4 installed, if that matters. Perhaps I should go for OpenCV3?

matlabbe, 2020-05-25:
RTAB-Map works with OepnCV2, OpenCV3 and OpenCV4. I'll be receiving a RPI4 soon, so I could see better what are the problems. In general, try to stick with default binary versions of the dependencies like OpenCV, pcl, vtk...