introlab/rtabmap_ros #420: static objects appearing multiple times on the map

MatanVoichek, 2020-05-16:
hello, 
I am a student that have been using Rtab-map for a project with the end goal of making an autonomous robot capable of navigating an unmapped environment. I am having difficulty however with getting Rtabmap to remove falsely placed points from the cloud during mapping, and the result is that any map I make becoming worse the longer I loop around the same area. I understand there are ways to clean the map after the process is done but due to the goals of the project I can't use use those. Is there a way to make rtabmap clean objects off the map if it sees they are not there? 

because I cant reach the university anymore I am currently using Gazebo to simulate a turtlebot and kinect camera using the instructions on segment 6 of the [mapping and navigation on turtlebot](http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot) tutorial. (I turned rtabmapvis on in the rtabmap launch file because I like it better then Rviz, but the problem accours there as well). 
[image]
you can see in this picture for example that the fire hydrant appears multiple times in the 3-D point cloud, and even after driving the robot around it several times looking at it from diffrent angles (and going through the copies that shouldn't exist), rtabmap did not remove the errors.
is there a way to make Rtabmap clean mistakes while it is running?

I am using ubuntu 14.04 and ros indigo. 
thank you for your time.
Matan.

MatanVoichek, 2020-05-23:
solved: I had to leave my computer for a while and forgot to pause the simulation and rtabmap, so the robot stayed in place without getting new images to add to the map graph and when I came back after a while the map fixed itself and cleared all the extra iterations of the same objects. I think the problem was just that my processor wasn't strong enough and by simply giving rtabmap time to work on loop closures it managed to fix itself.

closed 2020-05-23 by MatanVoichek

matlabbe, 2020-05-25:
What we are seeing is the odometry drift. Detecting loop closures helps to reduce that error.