introlab/rtabmap_ros #421: roslaunch eror

Het369, 2020-05-22:
I have installed from source
Tried many launch files (demo_stereo_outdoor.launch, demo_mapping.launch, rtabmap.launch) but got the same error 

`
... logging to /home/het3/.ros/log/22a2d710-9c07-11ea-b943-aced5c308ea0/roslaunch-het-hpOMEN-18815.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://het-hpOMEN:45207/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /rtabmap/rtabmap/Grid/DepthDecimation: 4
 * /rtabmap/rtabmap/Grid/FlatObstacleDetected: true
 * /rtabmap/rtabmap/Kp/DetectorStrategy: 6
 * /rtabmap/rtabmap/Kp/MaxDepth: 0
 * /rtabmap/rtabmap/RGBD/CreateOccupancyGrid: true
 * /rtabmap/rtabmap/Rtabmap/TimeThr: 700
 * /rtabmap/rtabmap/Vis/EstimationType: 1
 * /rtabmap/rtabmap/Vis/MaxDepth: 0
 * /rtabmap/rtabmap/frame_id: base_footprint
 * /rtabmap/rtabmap/map_negative_poses_ignored: True
 * /rtabmap/rtabmap/queue_size: 30
 * /rtabmap/rtabmap/subscribe_depth: False
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_stereo: True
 * /rtabmap/stereo_odometry/GFTT/MinDistance: 10
 * /rtabmap/stereo_odometry/GFTT/QualityLevel: 0.00001
 * /rtabmap/stereo_odometry/Odom/GuessMotion: true
 * /rtabmap/stereo_odometry/Odom/Strategy: 0
 * /rtabmap/stereo_odometry/OdomF2M/MaxSize: 1000
 * /rtabmap/stereo_odometry/Vis/EstimationType: 1
 * /rtabmap/stereo_odometry/Vis/MaxDepth: 0
 * /rtabmap/stereo_odometry/Vis/MinInliers: 10
 * /rtabmap/stereo_odometry/frame_id: base_footprint
 * /rtabmap/stereo_odometry/odom_frame_id: odom
 * /rtabmap/stereo_odometry/subscribe_rgbd: False
 * /stereo_camera/stereo_image_proc/disparity_range: 128
 * /use_sim_time: True

NODES
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)
    stereo_odometry (rtabmap_ros/stereo_odometry)
  /stereo_camera/
    stereo_image_proc (stereo_image_proc/stereo_image_proc)
  /
    republish_left (image_transport/republish)
    republish_right (image_transport/republish)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [18825]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 22a2d710-9c07-11ea-b943-aced5c308ea0
process[rosout-1]: started with pid [18836]
started core service [/rosout]
process[republish_left-2]: started with pid [18840]
process[republish_right-3]: started with pid [18844]
process[stereo_camera/stereo_image_proc-4]: started with pid [18845]
process[rtabmap/stereo_odometry-5]: started with pid [18852]
process[rtabmap/rtabmap-6]: started with pid [18857]
process[rviz-7]: started with pid [18858]
double free or corruption (out)
[ WARN] [1590136506.178513005]: The input topic '/stereo_camera/left/camera_info_throttle' is not yet advertised
[ WARN] [1590136506.179214752]: The input topic '/stereo_camera/right/camera_info_throttle' is not yet advertised
double free or corruption (out)
[rtabmap/stereo_odometry-5] process has died [pid 18852, exit code -6, cmd /home/het3/summer_ws/devel/lib/rtabmap_ros/stereo_odometry left/image_rect:=/stereo_camera/left/image_rect right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info_throttle right/camera_info:=/stereo_camera/right/camera_info_throttle rgbd_image:=/stereo_camera/rgbd_image odom:=/stereo_odometry __name:=stereo_odometry __log:=/home/het3/.ros/log/22a2d710-9c07-11ea-b943-aced5c308ea0/rtabmap-stereo_odometry-5.log].
log file: /home/het3/.ros/log/22a2d710-9c07-11ea-b943-aced5c308ea0/rtabmap-stereo_odometry-5*.log
[rtabmap/rtabmap-6] process has died [pid 18857, exit code -6, cmd /home/het3/summer_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info_throttle right/camera_info:=/stereo_camera/right/camera_info_throttle rgbd_image:=/stereo_camera/rgbd_image odom:=/stereo_odometry __name:=rtabmap __log:=/home/het3/.ros/log/22a2d710-9c07-11ea-b943-aced5c308ea0/rtabmap-rtabmap-6.log].
log file: /home/het3/.ros/log/22a2d710-9c07-11ea-b943-aced5c308ea0/rtabmap-rtabmap-6*.log
[rviz-7] process has died [pid 18858, exit code -11, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/het3/summer_ws/src/rtabmap_ros/launch/config/demo_stereo_outdoor.rviz __name:=rviz __log:=/home/het3/.ros/log/22a2d710-9c07-11ea-b943-aced5c308ea0/rviz-7.log].
log file: /home/het3/.ros/log/22a2d710-9c07-11ea-b943-aced5c308ea0/rviz-7*.log

`

matlabbe, 2020-05-25:
Is standalone `$ rtabmap` working? Can you show cmake output of rtabmap library?

Het369, 2020-05-27:
Yes it was working. 
I installed the rtabmap from source again and its working now. 

Thanks

closed 2020-06-05 by Het369