introlab/rtabmap_ros #422: Problems installing rtabmap with xfeatures2d on melodic.

kaatrasa, 2020-05-25:
Hey, I want to use BRIEF-features and those are missing. I think rtabmap is using opencv installed by ros instead of the one I built from source. What I tried:

First I installed dependencies (https://github.com/introlab/rtabmap_ros#installation):
```
sudo apt-get install ros-melodic-rtabmap ros-melodic-rtabmap-ros
sudo apt-get remove ros-melodic-rtabmap ros-melodic-rtabmap-ros
```
Then install opencv from source:
```
cd ~
git clone https://github.com/opencv/opencv.git
git clone https://github.com/opencv/opencv_contrib.git

cd opencv
mkdir build
cd build
cmake -DOPENCV_EXTRA_MODULES_PATH=~/opencv_contrib/modules -DCMAKE_BUILD_TYPE=Release ..
make -j4
sudo make install
```
Finally I installed the standalone rtabmap application from source:
```
cd ~
git clone https://github.com/introlab/rtabmap.git rtabmap
cd rtabmap/build
cmake ..  [<---double dots included]
make
sudo make install
```
Everything goes ok, but the BRIEF option is not available:
[image]

I could not get it working after many attempts, so am I doing something wrong? Thank you.

matlabbe, 2020-05-25:
What is the output of cmake? It seems it is still building against OpenCV's system binaries. You could remove `CMakeCache.txt` in `rtabmap/build`, then launch cmake with `-DOpenCV_DIR=/usr/local/lib/cmake/opencv4`. This will force using your installed `OpenCVConfig.cmake`. 

Note that if you want to use `rtabmap_ros` and OpenCV built from souce. It is recommended to uninstall `libopencv` from your computer to avoid OpenCV conflicts (seg fault when launching `rtabmap_ros`):
```
$ sudo apt remove libopencv*
```
This will remove `cv_bridge` ros package, thus you will need to clone it (and other packages depending on `cv_bridge`) in your `catkin_ws` so that `cv_bridge` is built with your OpenCV version. In this approach, as only one OpenCV would be installed, no need to explicitly set `OpenCV_DIR` as above.

kaatrasa, 2020-05-25:
Hey, thanks for the quick response. I would like to use rtabmap_ros. The output of rtabmap cmake was:
kaatrasa@kaatrasa:~/rtabmap/build$ cmake ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:64 (cmake_policy):
  The OLD behavior for policy CMP0020 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Checking for module 'eigen3'
--   Found eigen3, version 3.3.4
-- Found eigen: /usr/include/eigen3  
-- Checking for module 'libopenni'
--   Found libopenni, version 1.5.4.0
-- Found openni: /usr/lib/libOpenNI.so  
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- Found libusb-1.0: /usr/include  
-- Checking for module 'flann'
--   Found flann, version 1.9.1
-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
-- Found qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- Found PCL_COMMON: /usr/lib/x86_64-linux-gnu/libpcl_common.so  
-- Found PCL_OCTREE: /usr/lib/x86_64-linux-gnu/libpcl_octree.so  
-- Found PCL_IO: /usr/lib/x86_64-linux-gnu/libpcl_io.so  
-- Found PCL_KDTREE: /usr/lib/x86_64-linux-gnu/libpcl_kdtree.so  
-- Found PCL_SEARCH: /usr/lib/x86_64-linux-gnu/libpcl_search.so  
-- Found PCL_SURFACE: /usr/lib/x86_64-linux-gnu/libpcl_surface.so  
-- Found PCL_FILTERS: /usr/lib/x86_64-linux-gnu/libpcl_filters.so  
-- Found PCL_2D: /usr/include/pcl-1.8  
-- Found PCL_FEATURES: /usr/lib/x86_64-linux-gnu/libpcl_features.so  
-- Found PCL_REGISTRATION: /usr/lib/x86_64-linux-gnu/libpcl_registration.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/lib/x86_64-linux-gnu/libpcl_sample_consensus.so  
-- Found PCL_GEOMETRY: /usr/include/pcl-1.8  
-- Found PCL_ML: /usr/lib/x86_64-linux-gnu/libpcl_ml.so  
-- Found PCL_SEGMENTATION: /usr/lib/x86_64-linux-gnu/libpcl_segmentation.so  
-- Found PCL_VISUALIZATION: /usr/lib/x86_64-linux-gnu/libpcl_visualization.so  
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
[... 1389 lines cut ...]
    runtime library [libQt5PrintSupport.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/kaatrasa/anaconda3/lib
    runtime library [libQt5Widgets.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/kaatrasa/anaconda3/lib
    runtime library [libQt5Gui.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/kaatrasa/anaconda3/lib
    runtime library [libQt5Core.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/kaatrasa/anaconda3/lib

  Some of these libraries may not be found correctly.

CMake Warning at examples/NoEventsExample/CMakeLists.txt:63 (ADD_EXECUTABLE):
  Cannot generate a safe dependent library path for target noEventsExample
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libQt5OpenGL.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/kaatrasa/anaconda3/lib
    runtime library [libQt5Sql.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/kaatrasa/anaconda3/lib

  Some of these libraries may not be found correctly.

-- Generating done
-- Build files have been written to: /home/kaatrasa/rtabmap/build

matlabbe, 2020-05-25:
It should be building against OpenCV4. Maybe to the same rtabmap version was started. You can do "$ which rtabmap" to see which one is started and do "$ rtabmap --version" to see the version. Compare with "~/rtabmap/bin/rtabmap".

kaatrasa, 2020-05-25:
I think it should be using the same rtabmap (time is same, I just now build it):
[image]

However, when I do `rtabmap --version` I get that opencvnonfree is false. I quess this is a problem?
[image]

matlabbe, 2020-05-25:
You may have built OpenCV without nonfree module. I am looking why xfeatures2d is ignored when OpenCV is not build with nonfree. In the mean time, you can try to build/install OpenCV with OPENCV_ENABLE_NONFREE cmake option enabled.

kaatrasa, 2020-05-25:
`cmake -DOPENCV_EXTRA_MODULES_PATH=~/opencv_contrib/modules -D OPENCV_ENABLE_NONFREE=1 -DCMAKE_BUILD_TYPE=Release ..`

You were right, building with nonfree modules fixed the issue. Tomorrow I will see if I have trouble with rtabmap_ros and update here. Thanks a lot!

closed 2020-06-01 by kaatrasa

adbidwai, 2021-08-02:
@matlabbe @kaatrasa 
Hello, there's an ambiguity I am facing. 
Below is the log when I execute `rtabmap --version`. Notice `With OpenCV xfeatures2d: false` and `With OpenCV nonfree:     false`. 
```
 ⚙ adbidwai@adbidwai-g3-3579  /usr/local/lib  rtabmap --version
RTAB-Map:               0.20.9
PCL:                     1.8.1
With VTK:                6.3.0
OpenCV:                  3.2.0
With OpenCV xfeatures2d: false
With OpenCV nonfree:     false
With ORB OcTree:          true
With SuperPoint Torch:   false
With Python3:            false
With FastCV:             false
With Madgwick:            true
With TORO:                true
With g2o:                 true
With GTSAM:              false
With Vertigo:             true
With CVSBA:              false
With Ceres:              false
With OpenNI2:             true
With Freenect:            true
With Freenect2:          false
With K4W2:               false
With K4A:                false
With DC1394:              true
With FlyCapture2:        false
With ZED:                false
With ZED Open Capture:   false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:         false
With MYNT EYE S:         false
With DepthAI:            false
With libpointmatcher:    false
With CCCoreLib:          false
With octomap:             true
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM:           false
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
```
But the issue is that I have my OpenCV built from source with the flag `OPENCV_ENABLE_NONFREE=1`. Here's the proof (Notice the flag `OPENCV_ENABLE_NONFREE:BOOL=ON`) : 
```
✘ adbidwai@adbidwai-g3-3579  ~/opencv_build/opencv/build  ➦ bda89a6469  cmake -LA                          

CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error: The source directory "/home/adbidwai/opencv_build/opencv/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
-- Cache values
ANT_EXECUTABLE:FILEPATH=/usr/bin/ant
Atlas_BLAS_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libatlas.so
Atlas_CBLAS_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu
Atlas_CBLAS_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libcblas.so
Atlas_CLAPACK_INCLUDE_DIR:PATH=Atlas_CLAPACK_INCLUDE_DIR-NOTFOUND
Atlas_LAPACK_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/liblapack.so
BLAS_armpl_lp64_LIBRARY:FILEPATH=BLAS_armpl_lp64_LIBRARY-NOTFOUND
[... 421 lines cut ...]
pkgcfg_lib_GSTREAMER_pbutils_gstpbutils-1.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libgstpbutils-1.0.so
pkgcfg_lib_GSTREAMER_pbutils_gstreamer-1.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so
pkgcfg_lib_GSTREAMER_pbutils_gstvideo-1.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libgstvideo-1.0.so
pkgcfg_lib_GSTREAMER_riff_glib-2.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libglib-2.0.so
pkgcfg_lib_GSTREAMER_riff_gobject-2.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libgobject-2.0.so
pkgcfg_lib_GSTREAMER_riff_gstreamer-1.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so
pkgcfg_lib_GSTREAMER_riff_gstriff-1.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libgstriff-1.0.so
pkgcfg_lib_GTHREAD_glib-2.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libglib-2.0.so
pkgcfg_lib_GTHREAD_gthread-2.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libgthread-2.0.so
pkgcfg_lib_GTK3_atk-1.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libatk-1.0.so
pkgcfg_lib_GTK3_cairo:FILEPATH=/usr/lib/x86_64-linux-gnu/libcairo.so
pkgcfg_lib_GTK3_cairo-gobject:FILEPATH=/usr/lib/x86_64-linux-gnu/libcairo-gobject.so
pkgcfg_lib_GTK3_gdk-3:FILEPATH=/usr/lib/x86_64-linux-gnu/libgdk-3.so
pkgcfg_lib_GTK3_gdk_pixbuf-2.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so
pkgcfg_lib_GTK3_gio-2.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libgio-2.0.so
pkgcfg_lib_GTK3_glib-2.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libglib-2.0.so
pkgcfg_lib_GTK3_gobject-2.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libgobject-2.0.so
pkgcfg_lib_GTK3_gtk-3:FILEPATH=/usr/lib/x86_64-linux-gnu/libgtk-3.so
pkgcfg_lib_GTK3_pango-1.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libpango-1.0.so
pkgcfg_lib_GTK3_pangocairo-1.0:FILEPATH=/usr/lib/x86_64-linux-gnu/libpangocairo-1.0.so
pkgcfg_lib_HARFBUZZ_harfbuzz:FILEPATH=/usr/lib/x86_64-linux-gnu/libharfbuzz.so
``` 
What might be the issue?
Any help is appreciated,
TIA

matlabbe, 2021-08-18:
In your log, the rtabmap launched is built with OpenCV 3.2, not 4.2. Do `which rtabmap` to know which rtabmap is started. I guess it is the binary one, if so, uninstall `ros-melodic-rtabmap`:
```
sudo apt remove ros-melodic-rtabmap
```