introlab/rtabmap_ros #424: Python scripts likely not runnable on Noetic

sloretz, 2020-06-02:
@matlabbe FYI while reviewing https://github.com/ros/rosdistro/pull/25269/ I noticed the Python scripts are being installed using a plain CMake `install()` command. The shebangs of the installed script won't be correct and the scripts will fail to execute.

https://github.com/introlab/rtabmap_ros/blob/2fcb2898948c42f2eea437a7cde4d0ce4eb03d62/CMakeLists.txt#L511-L519

They should actually be installed with `catkin_install_python()` as described on [this ROS wiki page](wiki.ros.org/UsingPython3/SourceCodeChanges#Changing_shebangs).

matlabbe, 2020-06-02:
I didn't know about this change for Python3. I'll take a look, thx for pointing out the doc.

matlabbe, 2020-06-02:
I'll edit again, hold it.

sloretz, 2020-06-02:
Fixed as of 6f8b8b2917f0f4458c9b53f1ed329300d9c4b302

closed 2020-06-02 by sloretz