introlab/rtabmap_ros #425: rgbd_sync not subscribed to anything

blue-ring-octopus, 2020-06-03:
I'm following this example
https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_two_kinects.launch#L55
except changed the cameras to gazebo camera
when i launch it, rgbd_sync is not publishing anything. So i check with rosnode info
and found out the it is not subscribe to any topic except clock. what happened?

closed 2020-06-03 by blue-ring-octopus

lifuguan, 2020-07-22:
I've faced the same situation and it bother me so much? Is there any solution to deal with it?

blue-ring-octopus, 2020-07-22:
> I've faced the same situation and it bother me so much? Is there any solution to deal with it?

for me, it was just missing a nodelet manager, the nodelet manager was hidden in the kinect node in the example, but if you are using gazebo simulation you have to launch a separate manager

lifuguan, 2020-07-27:
Do you mean nodelet "rgbd_sync" can only subscribe nodelet and Gazebo is not a nodelet but a node?
And I don't get your point "launch a seperate manager". A little confused.

blue-ring-octopus, 2020-07-27:
> Do you mean nodelet "rgbd_sync" can only subscribe nodelet and Gazebo is not a nodelet but a node?
> And I don't get your point "launch a seperate manager". A little confused.
i'm not proficient enough to explain the technical detail, if i remember correctly rgbd_sync itself is a nodelet, you need a nodelet manager to interface a nodelet with other ros structure, something like that. if you search up"ros nodelet manager" it should be quite straight forward, assume that we have the exact same issue that is