introlab/rtabmap_ros #428: How to save the post-processed cloud in the rtabmap-ros?

armanhamzehlou, 2020-06-07:


matlabbe, 2020-06-08:
Do you means `cloud_map` topic? See [pcl_ros/pointcloud_to_pcd](http://wiki.ros.org/pcl_ros#pointcloud_to_pcd).

armanhamzehlou, 2020-06-08:
> Do you means `cloud_map` topic? See [pcl_ros/pointcloud_to_pcd](http://wiki.ros.org/pcl_ros#pointcloud_to_pcd).

When you process your point-cloud through Menu -> Tools -> Post-Processing, How do you save the new point cloud?

matlabbe, 2020-06-08:
Do File->Export 3D clouds. The optimized poses by Post-Processing will be used to assemble the clouds.

armanhamzehlou, 2020-06-08:
> Do File->Export 3D clouds. The optimized poses by Post-Processing will be used to assemble the clouds.

I don't want a ply format. I want the .db format to be actually updated. Is this possible?

pauldeee, 2020-06-08:
Open in the DB Viewer, Tools -> Edit database... Make your changes then save the changes to the databse.

armanhamzehlou, 2020-06-09:
> Open in the DB Viewer, Tools -> Edit database... Make your changes then save the changes to the database.

Yes, This seems actually promising, is there a way to modify the parameters of the loop closure detection in the database editor?

pauldeee, 2020-06-09:
Rather than exporting the database, try clicking close in the Database Viewer, and click "Save" on the dialog that appears.

The options you are looking for might be found in the Core Parameters. Inside the DB Viewer, choose View -> Core Paramters.

armanhamzehlou, 2020-06-11:
> Rather than exporting the database, try clicking close in the Database Viewer, and click "Save" on the dialog that appears.
> 
> The options you are looking for might be found in the Core Parameters. Inside the DB Viewer, choose View -> Core Parameters.

Thank you it was very helpful and time-saving.

closed 2020-06-11 by armanhamzehlou