introlab/rtabmap_ros #429: Synchronization issues

eliabntt, 2020-06-08:
Hi matlabbe. 
Currently I have this pipeline
1. Robot localization ekf takes wheel odom(20 Hz), IMU (50 Hz) and rtabmap localization_pose (aperiodic) as inputs and gives me odom->base_link transform. Probably I'll add either ICP odometry or visual odom or even both to use every sensor.
2. rtabmap is subscribed to /odometry/filtered (robot_loc_ekf, 50Hz), a 2D /robot/laser/scan (20 Hz) and /rtabmap/rgbd_image (from your rgbd_sync node, 15Hz) to navigate and ICP+Visual registration ("Reg/Strategy 2")
3. as we were discussing in previous issues the map->odom tf is given by rtabmap.

When I set
"odom_frame_id" to "odom" the rate of rtabmap is high enough to work. The issue here is that in this way I'm not using the /odometry/filtered twist covariance in the graph links but I'm using the fixed one am I right? At least in the readme is said that this is what happens.

When I do NOT set "odom_frame_id" to have the correct computed covariance I have a rate problem meaning rtabmap is receiving images every 1/2 seconds even if the processing time is <200 ms.
I've tested a lot and the problem seems related to the synchronization of the three topics rtabmap subscribes to. More in detail, the problem is in the co-existence of /odometry/filtered and /robot/laser/scan. 

When I set /subscribe_scan to false (with odom_frame_id empty) everything works flawlessly but I lose an important source of relocalization and stabilization. 

Another option is to set approx_sync to true but in that way again it does not work well. Here, I tried to look at your code but you rely on ros-specific message filtering that works directly on the topics so I can't tamper the scan messages to edit their timestamps. Moreover, I noticed that  CommonDataSubscriber::commonSingleDepthCallback is being called continuously and executed up to the last line (included) but CoreWrapper::commonDepthCallback is the one that is not being called with the right frequency.

Looking at the topics timestamps they seem more or less sync with some jumps here and there but nothing special and as you can see both (rgbd, scan, odometry, WITH odom_frame_id) and (rgbd, odometry, WITHOUT odom_frame_id) works.

Finally, I've tried also (scan, odom, WITHOUT odom_frame_id) and that also works at the right rate.

```

<group ns="rtabmap">
        <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="standalone rtabmap_ros/rgbd_sync" output="screen">
            <remap from="rgb/image" to="$(arg rgb_topic)_relay" if="$(arg compressed)"/>
            <remap from="depth/image" to="$(arg depth_topic)_relay" if="$(arg compressed)"/>
            <remap from="rgb/image" to="$(arg rgb_topic)" unless="$(arg compressed)"/>
            <remap from="depth/image" to="$(arg depth_topic)" unless="$(arg compressed)"/>
            <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
            <remap from="rgbd_image" to="rgbd_image"/> <!-- output -->
            <param name="approx_sync" value="false"/>
        </node>

        <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg delete)">

            <param name="publish_tf" value="true"/>
            <param name="database_path" type="string" value="$(arg database_path)"/>
            <param name="frame_id" type="string" value="base_link"/>
           
            <param name="odom_sensor_sync" value="true"/>

            <param name="subscribe_rgbd" type="bool" value="true"/>
            <param name="subscribe_rgb" type="bool" value="false"/>
            <param name="subscribe_depth" type="bool" value="false"/>
            <param name="subscribe_scan" type="bool" value="false"/>
            <!--param name="odom_frame_id" value="odom"/-->
           
            <param name="approx_sync" value="false"/>
            <param name="wait_for_transform_duration" type="double" value="0.1"/>

            <!-- RTAB-Map Inputs -->
            <remap from="scan" to="$(arg scanner)"/>
            <remap from="rgbd_image" to="rgbd_image"/>
            <remap from="odom" to="/odometry/filtered"/>

            <!-- RTAB-Map Output -->
            <remap from="grid_map" to="/map"/>

            <!-- Rate (Hz) at which new nodes are added to map -->
            <param name="Rtabmap/DetectionRate" type="string" value="5"/>

            <param name="map_filter_radius" type="double" value="0.5"/>
            <param name="map_filter_angle" type="double" value="20"/>
            <param name="map_always_update" type="bool" value="true"/>
            <param name="Grid/RangeMax" type="double" value="4"/>
            <param name="Grid/RangeMin" type="double" value="0.4"/>
            <param name="Grid/FootprintHeight" type="double" value="0.8"/>
            <param name="Grid/FootprintLength" type="double" value="0.4"/>
            <param name="Grid/FootprintWidth" type="double" value="0.4"/>
            <param name="Grid/FromDepth" type="string" value="true"/>
            <param name="Grid/3D" type="string" value="true"/>
            <param name="Grid/RayTracing" type="string" value="false"/>

            <param name="GridGlobal/FootprintRadius" type="double" value="0.2"/>
            <param name="GridGlobal/UpdateError" type="double" value="0.05"/> <!-- 0.01 -->

            <param name="Kp/MaxDepth" type="double" value="0"/>
            <param name="Kp/MaxDepth" type="double" value="0"/>
            <param name="Kp/MaxFeatures" type="string" value="600"/> <!-- todo try to increase -->
            <param name="Kp/SubPixIterations" type="string" value="1"/>
            <param name="Kp/DetectorStrategy" type="string" value="$(arg detector)"/>

            <param name="Mem/NotLinkedNodesKept" type="string"
                   value="false"/><!-- Set to false to avoid saving data when robot is not moving -->
            <param name="Mem/RecentWmRatio" type="double" value="0.5"/>
            <param name="Mem/ReduceGraph" type="bool" value="true"/>
            <param name="Mem/STMSize" type="string" value="15"/>

            <param name="Optimizer/Strategy" type="string" value="1"/> <!-- todo g2o=1, GTSAM=2 -->
            <param name="Optimizer/Robust" type="string"
                   value="$(arg robust)"/> <!--Optimizer/Robust=false with RGBD/OptimizeMaxError=3 -->

            <!-- Loop Closure Constraint -->
            <param name="Reg/Strategy" type="string" value="2"/>

            <param name="RGBD/SavedLocalizationIgnored" type="bool" value="true"/>
            <param name="RGBD/LocalBundleOnLoopClosure" type="bool" value="true"/>
            <param name="RGBD/LocalRadius" type="string" value="1"/>
            <param name="RGBD/LoopClosureReextractFeatures" type="bool" value="false"/> <!--todo false maybe try -->
            <param name="RGBD/MaxLocalRetrieved" type="string" value="10"/>
            <param name="RGBD/NeighborLinkRefining" type="bool" value="true"/>
            <param name="RGBD/OptimizeMaxError" type="string" value="0"
                   if="$(arg robust)"/> <!-- should be 0 if RGBD/OptimizeRobust is true -->
            <param name="RGBD/OptimizeMaxError" type="string" value="3" unless="$(arg robust)"/>
            <param name="RGBD/ProximityMaxGraphDepth" type="string" value="50"/> <!-- maybe increase -->
            <param name="RGBD/ProximityMaxPaths" type="string" value="0"/>
            <param name="RGBD/ProximityPathFilteringRadius" type="string" value="1"/>
            <param name="RGBD/ProximityPathMaxNeighbors" type="string" value="10"/>
            <param name="RGBD/AngularUpdate" type="string" value="0.01"/>
            <param name="RGBD/LinearUpdate" type="string" value="0.01"/>
            <param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/>
            <param name="RGBD/ProximityPathMaxNeighbors" type="string" value="10"/>
            <!-- <param name="Reg/Force3DoF" type="string" value="true"/>--> <!-- 2D SLAM todo check -->

            <param name="Vis/SubPixIterations" type="string" value="2"/>
            <param name="Vis/SubPixWinSize" type="string" value="5"/>
            <param name="Vis/MaxDepth" type="string" value="$(arg max_depth)"/>
            <param name="Vis/MinInliers" type="string" value="10"/>
            <param name="Vis/FeatureType" type="string" value="$(arg detector)"/> <!-- 8 -->
        </node>
    </group>

    <group if="$(arg sim)">
    </group>
</launch>

```

eliabntt, 2020-06-09:
Update:
I think the problem when I set approx_sync to true may be related to [this](https://github.com/introlab/rtabmap_ros/issues/407#issuecomment-638026089) setup. I will come back soon.

eliabntt, 2020-06-14:
I needed to set it as approx_sync true in the rtabmap space and correct the ekf :)

closed 2020-06-14 by eliabntt

matlabbe, 2020-06-30:
`approx_sync` would be `false` for `rgbd_sync` and `true` for `rtabmap` in that kind of setup because odometry messages will not have exactly the same stamp than the images.

See figure 6 of this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf):
[image]